kraj/meta-clang #9: which recipe provide llvm (dependency of mesa)?

walkthetalk, 2016-03-01:
the mesa is depends on llvm3.3, but llvm3.3 in meta-oe may has bug (I got segment fault).
so I want to use meta-clang, maybe I need to modify mesa.bb?

walkthetalk, 2016-03-01:
another problem: no recipe use llvm-common.
maybe clang_git should depends on it?

kraj, 2016-03-01:
llvm thats part of meta-clang is intentionally aimed at static compilation. with mesa you need the dynamic compilation piece, it might work but not without changes.

walkthetalk, 2016-03-01:
Can you give some clue? Thanks!

walkthetalk, 2016-03-02:
I have add some line to clang_git.bb:
```
EXTRA_OECMAKE += " \
               -DLLVM_BUILD_LLVM_DYLIB=ON \
               -DLLVM_DYLIB_EXPORT_ALL=ON \
               -DLLVM_LINK_LLVM_DYLIB=ON \
               "
```
but get that error:
```
ld: ../../lib/libLLVMSupport.a(PrettyStackTrace.cpp.o)(.text._ZN4llvm23PrettyStackTraceProgramD2Ev[_ZN4llvm23PrettyStackTraceProgramD5Ev]+0x20): R_ARM_TLS_LE32 relocation not permitted in shared object
```

I checked the command used:
```
/home/ll/kl360/build/tmp/sysroots/x86_64-linux/usr/bin/arm-jilong-linux-gnueabi/arm-jilong-linux-gnueabi-g++  -fPIC -march=armv6 -mfpu=vfp -mfloat-abi=softfp -mfpu=vfp -D__extern_always_inline=inline -L/home/ll/kl360/build/tmp/sysroots/kl360/usr/lib/libxml2 -I/home/ll/kl360/build/tmp/sysroots/kl360/usr/include/libxml2  --sysroot=/home/ll/kl360/build/tmp/sysroots/kl360  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/ll/kl360/build/tmp/work/armv6-vfp-jilong-linux-gnueabi/clang/3.8.0+gitAUTOINC+27c964e2ae_b98cd9d6a3-r0/build=/usr/src/clang -fdebug-prefix-map=/home/ll/kl360/build/tmp/work/armv6-vfp-jilong-linux-gnueabi/clang/3.8.0+gitAUTOINC+27c964e2ae_b98cd9d6a3-r0/git=/usr/src/clang -fdebug-prefix-map=/home/ll/kl360/build/tmp/sysroots/x86_64-linux= -fdebug-prefix-map=/home/ll/kl360/build/tmp/sysroots/kl360=  -fvisibility-inlines-hidden -fvisibility-inlines-hidden -Wall -W -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wno-missing-field-initializers -pedantic -Wno-long-long -Wno-maybe-uninitialized -Wdelete-non-virtual-dtor -Wno-comment -std=c++11 -ffunction-sections -fdata-sections -DNDEBUG  -Wl,-O3 -Wl,--gc-sections  -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -Wl,-z,defs -shared -Wl,-soname,libLLVM-3.8.so -o ../../lib/libLLVM-3.8.so CMakeFiles/LLVM.dir/libllvm.cpp.o -Wl,--whole-archive ../../lib/libLLVMSupport.a ../../lib/libLLVMCore.a ../../lib/libLLVMIRReader.a ../../lib/libLLVMCodeGen.a ../../lib/libLLVMSelectionDAG.a ../../lib/libLLVMAsmPrinter.a ../../lib/libLLVMMIRParser.a ../../lib/libLLVMBitReader.a ../../lib/libLLVMBitWriter.a ../../lib/libLLVMTransformUtils.a ../../lib/libLLVMInstrumentation.a ../../lib/libLLVMInstCombine.a ../../lib/libLLVMScalarOpts.a ../../lib/libLLVMipo.a ../../lib/libLLVMVectorize.a ../../lib/libLLVMObjCARCOpts.a ../../lib/libLLVMLinker.a ../../lib/libLLVMAnalysis.a ../../lib/libLLVMLTO.a ../../lib/libLLVMMC.a ../../lib/libLLVMMCParser.a ../../lib/libLLVMMCDisassembler.a ../../lib/libLLVMObject.a ../../lib/libLLVMOption.a ../../lib/libLLVMDebugInfoCodeView.a ../../lib/libLLVMDebugInfoDWARF.a ../../lib/libLLVMDebugInfoPDB.a ../../lib/libLLVMSymbolize.a ../../lib/libLLVMExecutionEngine.a ../../lib/libLLVMInterpreter.a ../../lib/libLLVMMCJIT.a ../../lib/libLLVMOrcJIT.a ../../lib/libLLVMRuntimeDyld.a ../../lib/libLLVMTarget.a ../../lib/libLLVMARMCodeGen.a ../../lib/libLLVMARMInfo.a ../../lib/libLLVMARMAsmParser.a ../../lib/libLLVMARMDisassembler.a ../../lib/libLLVMARMAsmPrinter.a ../../lib/libLLVMARMDesc.a ../../lib/libLLVMAsmParser.a ../../lib/libLLVMLineEditor.a ../../lib/libLLVMProfileData.a ../../lib/libLLVMPasses.a ../../lib/libLLVMLibDriver.a -Wl,--no-whole-archive ../../lib/libLLVMObjCARCOpts.a ../../lib/libLLVMDebugInfoDWARF.a ../../lib/libLLVMDebugInfoPDB.a ../../lib/libLLVMExecutionEngine.a ../../lib/libLLVMRuntimeDyld.a ../../lib/libLLVMSelectionDAG.a ../../lib/libLLVMAsmPrinter.a ../../lib/libLLVMCodeGen.a ../../lib/libLLVMBitWriter.a ../../lib/libLLVMInstrumentation.a ../../lib/libLLVMTarget.a ../../lib/libLLVMMCDisassembler.a ../../lib/libLLVMARMInfo.a ../../lib/libLLVMARMAsmPrinter.a ../../lib/libLLVMipo.a ../../lib/libLLVMIRReader.a ../../lib/libLLVMAsmParser.a ../../lib/libLLVMScalarOpts.a ../../lib/libLLVMInstCombine.a ../../lib/libLLVMLinker.a ../../lib/libLLVMProfileData.a ../../lib/libLLVMVectorize.a ../../lib/libLLVMTransformUtils.a ../../lib/libLLVMAnalysis.a ../../lib/libLLVMObject.a ../../lib/libLLVMBitReader.a ../../lib/libLLVMCore.a ../../lib/libLLVMMCParser.a ../../lib/libLLVMMC.a ../../lib/libLLVMOption.a ../../lib/libLLVMSupport.a -lrt -ldl -ltinfo -lpthread -lz -lm -Wl,-rpath,"\$ORIGIN/../lib"
```

it looks like the command has error mixed the '-l' and '-wl', I don't know if it is the cause. any idea?

walkthetalk, 2016-03-02:
the clang-native compile error (enabled shared library):
```
/usr/bin/g++  -fPIC   -isystem/home/ll/kl360/build/tmp/sysroots/x86_64-linux/usr/include -O2 -pipe -fPIC -fvisibility-inlines-hidden -Wall -W -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wno-missing-field-initializers -pedantic -Wno-long-long -Wno-maybe-uninitialized -Wdelete-non-virtual-dtor -Wno-comment -std=c++11 -ffunction-sections -fdata-sections -fno-common -Woverloaded-virtual -fno-strict-aliasing -DNDEBUG  -Wl,-O3 -Wl,--gc-sections  -Wl,--version-script,/home/ll/kl360/build/tmp/work/x86_64-linux/clang-native/3.8.0+gitAUTOINC+27c964e2ae_b98cd9d6a3-r0/build/tools/clang/tools/libclang/libclang.exports  -L/home/ll/kl360/build/tmp/sysroots/x86_64-linux/usr/lib -L/home/ll/kl360/build/tmp/sysroots/x86_64-linux/lib -Wl,-rpath-link,/home/ll/kl360/build/tmp/sysroots/x86_64-linux/usr/lib -Wl,-rpath-link,/home/ll/kl360/build/tmp/sysroots/x86_64-linux/lib -Wl,-rpath,/home/ll/kl360/build/tmp/sysroots/x86_64-linux/usr/lib -Wl,-rpath,/home/ll/kl360/build/tmp/sysroots/x86_64-linux/lib -Wl,-O1 -Wl,-z,defs -shared -Wl,-soname,libclang.so.3.8 -o ../../../../lib/libclang.so.3.8 CMakeFiles/libclang.dir/ARCMigrate.cpp.o CMakeFiles/libclang.dir/BuildSystem.cpp.o CMakeFiles/libclang.dir/CIndex.cpp.o CMakeFiles/libclang.dir/CIndexCXX.cpp.o CMakeFiles/libclang.dir/CIndexCodeCompletion.cpp.o CMakeFiles/libclang.dir/CIndexDiagnostic.cpp.o CMakeFiles/libclang.dir/CIndexHigh.cpp.o CMakeFiles/libclang.dir/CIndexInclusionStack.cpp.o CMakeFiles/libclang.dir/CIndexUSRs.cpp.o CMakeFiles/libclang.dir/CIndexer.cpp.o CMakeFiles/libclang.dir/CXComment.cpp.o CMakeFiles/libclang.dir/CXCursor.cpp.o CMakeFiles/libclang.dir/CXCompilationDatabase.cpp.o CMakeFiles/libclang.dir/CXLoadedDiagnostic.cpp.o CMakeFiles/libclang.dir/CXSourceLocation.cpp.o CMakeFiles/libclang.dir/CXStoredDiagnostic.cpp.o CMakeFiles/libclang.dir/CXString.cpp.o CMakeFiles/libclang.dir/CXType.cpp.o CMakeFiles/libclang.dir/IndexBody.cpp.o CMakeFiles/libclang.dir/IndexDecl.cpp.o CMakeFiles/libclang.dir/IndexTypeSourceInfo.cpp.o CMakeFiles/libclang.dir/Indexing.cpp.o CMakeFiles/libclang.dir/IndexingContext.cpp.o ../../../../lib/libclangAST.a ../../../../lib/libclangBasic.a ../../../../lib/libclangFrontend.a ../../../../lib/libclangIndex.a ../../../../lib/libclangLex.a ../../../../lib/libclangSema.a ../../../../lib/libclangTooling.a ../../../../lib/libclangARCMigrate.a ../../../../lib/libclangFormat.a ../../../../lib/libclangToolingCore.a ../../../../lib/libclangASTMatchers.a ../../../../lib/libclangFrontend.a ../../../../lib/libclangDriver.a ../../../../lib/libclangParse.a ../../../../lib/libclangSerialization.a ../../../../lib/libclangSema.a ../../../../lib/libclangEdit.a ../../../../lib/libclangStaticAnalyzerCheckers.a ../../../../lib/libclangStaticAnalyzerCore.a ../../../../lib/libclangRewrite.a ../../../../lib/libclangAnalysis.a ../../../../lib/libclangAST.a ../../../../lib/libclangLex.a ../../../../lib/libclangBasic.a ../../../../lib/libLLVM-3.8.so -Wl,-rpath,/home/ll/kl360/build/tmp/sysroots/x86_64-linux/usr/lib
| CMakeFiles/libclang.dir/CIndexer.cpp.o: In function `clang::CIndexer::getClangResourcesPath[abi:cxx11]()':
| CIndexer.cpp:(.text._ZN5clang8CIndexer21getClangResourcesPathB5cxx11Ev+0x73): undefined reference to `dladdr'
| collect2: error: ld returned 1 exit status
| tools/clang/tools/libclang/CMakeFiles/libclang.dir/build.make:694: recipe for target 'lib/libclang.so.3.8' failed
```

kraj, 2016-03-03:
does g++ on your build host support c++11 ? what version of gcc are you using ?

walkthetalk, 2016-03-03:
I'm using gcc 5.3 (ArchLinux)

kraj, 2016-03-04:
I use arch based distro too.  I think bootstrapping clang with itself is right approach that will make it consistent across all distros

walkthetalk, 2016-03-04:
I agree with you. but, is there other way to use mesa llvmpipe without dynamic libllvm? or can we have a switch? let user choose dynamic or static by themselves.

kraj, 2016-04-05:
Do we have some resolution on this ?

kraj, 2017-09-19:
llvm is also added to oe-core now. Closing

closed 2017-09-19 by kraj