kraj/meta-clang #11: qa warnning: already stripped

walkthetalk, 2016-03-03:
WARNING: clang-native-3.8.0+gitAUTOINC+27c964e2ae_b98cd9d6a3-r0 do_populate_sysroot: File '/mnt/optimus/build/tmp/sysroots/x86_64-linux/usr/bin/llc' from clang-native was already stripped, this will prevent future debugging!

kraj, 2016-03-04:
This is due to we choosing to strip this during clang make itself. So we need to ignore this qa check. Otherwise size gets out of control

walkthetalk, 2016-03-04:
If so, I think we should add INSAN_SKIP in bb file

walkthetalk, 2016-03-07:
INSANE_SKIP_${PN} += "already-stripped"
INSANE_SKIP_${PN}-native += "already-stripped"
INSANE_SKIP_nativesdk-${PN} += "already-stripped"

the first line can work, but I don't know what is the right code for native and nativesdk.

kraj, 2016-03-07:
First 2 are ok. Third one needs to usr target arch. Look into clang.bbclass for sample

walkthetalk, 2016-03-09:
pull request #15 insane skip 'already-stripped'

kraj, 2016-04-05:
should be fixed now

closed 2016-04-05 by kraj