kraj/meta-clang #20: Don't depend on clang-native

rossburton, 2016-05-16:
Currently target clang depends on clang-native for the llvm-tblgen binary.  There should be a way of either bootstrapping that in the target clang build, or a clang-tools-native recipe that just builds that instead of all of clang.

kraj, 2016-06-14:
llvm-tblgen/clang-tlbgen is whats needed precisely. However, clang-native is needed in anycase since clang only build once for one host and there is no need to rebuild it again and again for different targets unlike gcc. One can say that I would use gcc everywhere even to build clang target then there is some use of just building minimal clang native. However thats not common usecase, people prefer to bootstrap clang with clang.

kraj, 2017-11-15:
clang build sequence builds clang-native once and reuses it for cross variants and it also means it reused it across architectures if someone builds form multiple architectures they will definitely see
the benefit. This is then used to build target clang. So even if we were able to trim down the build
to just do enough during native compile, we would still require full llvm-native for other use-cases.

I am temped to not fix it.

kraj, 2019-08-15:
maybe we can create a `clang-minimal-native` which could be also provided by clang-native so in cases where clang cross compilers are not sought, users can select 

`PREFERRED_PROVIDER_virtual/clang-tools-native = "clang-minimal-native"`

others can use

`PREFERRED_PROVIDER_virtual/clang-tools-native = "clang-native"`