kraj/meta-clang #23: use meta-clang within poky environment to compile clang for cortexa9thf-neon

brs4, 2016-07-15:
Compiling clang for a cortexa9thf-neon is working, but there are 2 Problems with the binary.
1. The  DEFAULT_TARGET_TRIPLE is wrong.
This could be solved by setting it hard in the recipe. Appending
-DLLVM_DEFAULT_TARGET_TRIPLE="arm-poky-Linux-gnueabi" to EXTRA_OECMAKE_append_class-target solves the problem, but only for arm.
2. clang cannot produce machine code. It uses bad values for CXXFLAGS, CFLAGS and LDFLAGS. Tuning features form cortexa9thf-neon are not recognized correctly during the build process. Passing -mfloat-abi=hard option on clang commandline for CFLAGS CXXFLAGS and LDFLAGS solves it temporarily on the target.

kraj, 2016-07-15:
@brs4 Can you provide some more info. Are you using the clang on the target ?

brs4, 2016-07-18:
Yes. After compiling clang with the recipes from meta-clang, calling clang --version on IMX6 target shows: 
# clang --version
clang version 3.8.1 (https://github.com/llvm-mirror/clang.git 9fd77bd68130d9b2fb
c56a3138b6f981d560480a) (https://github.com/llvm-mirror/llvm.git 06cf4b238fa4563
8ed37ec7725d38930fea37ff9)
Target: x86_64-unknown-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
So I set up the target triple hard with "arm-poky-Linux-gnueabi" like posted. Then ist Shows:
root@legcom:~# clang --version
clang version 3.8.1 (https://github.com/llvm-mirror/clang.git 9fd77bd68130d9b2fb
c56a3138b6f981d560480a) (https://github.com/llvm-mirror/llvm.git 06cf4b238fa4563
8ed37ec7725d38930fea37ff9)
Target: arm-poky-linux-gnueabi
Thread model: posix
InstalledDir: /usr/bin
And now I can produce a.out, passing -mfloat-abi=hard

kraj, 2017-02-28:
2a86d67fc139cd524fcd19d3286b0d8123395947 should fix this issue

closed 2017-02-28 by kraj