kraj/meta-clang #24: TUNE_CCARGS breaks SSTATE checksums

mythi, 2016-09-01:
I'm using an SSTATE_MIRROR to get clang pre-built by our CI build system but I noticed my build target always re-builds clang locally and sstate is invalid.

With bitbake-diffsigs I get that TUNE_CCARGS changes the hash. More specifically, the ```-I${PKG_CONFIG_SYSROOT_DIR}${includedir}/libxml2``` makes it build specific.

mythi, 2016-09-01:
using krogoth / 3.8. Just saw bf0a01851fb7cd8d66a23cef0f69df519c085eff.

kraj, 2016-09-01:
@mythi that patch is good for clang 3.9, however it hasnt been tested with clang 3.8, IIRC it was needed with some older version of clang I dont remember if it was 3.8 or older. If you can test it out and report back, I would be happy to cherry-pick it to krogoth

mythi, 2016-12-01:
Fixed by #25

PR #25 (2016-09-07): backport: clang-3.9: Remove unneeded flags for target
This commit cherry-picks bf0a01851fb7cd8d66a23cef0f69df519c085eff
from master to krogoth.

TUNE_CCARGS is an SSTATE variable and it must not used so that
it becomes builder specific. Builder specific TUNE_CCARGS makes
the SSTATE checksums to change and it invalidates the SSTATE cache
if used via SSTATE_MIRROR.

I've tested it builds with Ostro OS XT.

Fixes: #24
closed 2016-12-01 by mythi