kraj/meta-clang #32: libunwind header not getting installed

sjames, 2017-05-14:
Hi,
I'm creating a layer to build the MS .NET coreclr and corefx.  The meta-clang layer is a pre-requisite to the new layer I'm creating.

I'm following the linux build instructions for coreclr at https://github.com/dotnet/coreclr/blob/master/Documentation/building/linux-instructions.md .

I've published the work done so far at https://bitbucket.org/sojan_james/meta-dotnet .

Now coming to the problem.  I've managed to got get as far as configuring the toolchain.  The cmake check scripts now fail as the libunwind header is not found. I see that the llvm-libunwind recipe does not install the header file and I suspect this might be intentional.

Before I modify the llvm-libunwind recipe to install the header,  I'd like to ask if this is the right way to approach it.

Regards,
Sojan

kraj, 2017-05-14:
@sjames llvm-libunwind can conflict with libunwind from OE-Core thats why it is not installing it. But thats not the way to go, we have to make it such that it can be an alternative to libunwind from OE-core. So please go ahead and install it and watch out for any other conflicts we might have to solve.

kraj, 2017-09-19:
With recipe specific sysroot this should work better.

kraj, 2017-11-15:
@sjames can you verify if it works ok with master for you ?

kraj, 2019-08-15:
Latest master should have this addressed, if not feel free to re-open

closed 2019-08-15 by kraj