kraj/meta-clang #33: 9649a07fcfee2bec0794ea2dd43a3f3e8c535e1c disables cross-compiling

papazof, 2017-05-18:
TARGET_ARCH and HOST_ARCH are equal for the native package, so clang-native will build only with host arch support (unless overridden by user). Furthermore, packages compiler-rt, and clang (or any other target package built by clang-native) will fail to cross-compile.

kraj, 2017-05-18:
@papazof you are right, I have been testing with LLVM_TARGETS_TO_BUILD = "AArch64;ARM;Mips;PowerPC;X86" in my local.conf so this issue was remained hidden in my builds.

kraj, 2017-05-18:
@martingkelly I will have to revert it unless there is a better solution

martingkelly, 2017-05-18:
@papazof @kraj I think I know what's wrong; I believe the commit should be using BUILD_ARCH instead of HOST_ARCH. My intention was for clang-native to have support for the native and target arches. So if you're building for ARM on an X86 machine, you should build support for ARM and X86 but not for MIPS and PowerPC.

The issue is that HOST_ARCH is actually defined to be the same as TARGET_ARCH, so we are building *only* for the targeted arch and not for the host arch (see below for details).

Assuming you agree with this conclusion, I'll issue a revised patch to use BUILD_ARCH in place of HOST_ARCH.

```
martin@willamette:~/poky/build$ cat conf/local.conf | grep 'MACHINE = '
MACHINE = "qemuarm"
martin@willamette:~/poky/build$ bitbake -e clang | grep ^TRANSLATED_TARGET_ARCH=
TRANSLATED_TARGET_ARCH="arm"
martin@willamette:~/poky/build$ bitbake -e clang | grep ^HOST_ARCH=
HOST_ARCH="arm"
martin@willamette:~/poky/build$ bitbake -e clang | grep ^LLVM_TARGETS_TO_BUILD=
LLVM_TARGETS_TO_BUILD="ARM;ARM"
martin@willamette:~/poky/build$ bitbake -e clang | grep ^BUILD_ARCH=
BUILD_ARCH="x86_64"
```

So if we use BUILD_ARCH in place of HOST_ARCH, we should get LLVM_TARGETS_TO_BUILD="X86;ARM" as desired.

kraj, 2017-05-18:
@martingkelly try your experiment with clang-native thats where the problem lies

closed 2017-05-18 by kraj

kraj, 2017-05-18:
and this would also not work in nativesdk case where HOST_ARCH != BUILD_ARCH != TARGET_ARCH and we are actually interested in HOST_ARCH and TARGET_ARCH

martingkelly, 2017-05-18:
Yep, I see what you mean: for clang-native, TRANSLATED_TARGET_ARCH=x86_64. It seems that from clang-native, the only way to know you're targeting ARM is via MACHINE_ARCH="qemuarm", which does not directly give you the actual architecture.

I don't see an easy way to fix this, so I agree with building all arches as the only quick solution. That said, it does seem like there should be a way to build only the architectures that we actually need. If in the future there are 20 possible arches and we need only 2 of them for clang-native, then building all 20 is obviously inefficient.

commit c312b19b (2017-05-18): clang: Change LLVM_TARGETS_TO_BUILD to build everything by default

As described in issue #33, LLVM_TARGETS_TO_BUILD should include
build host as well as target arches and that fails since we build
clang-native once for all due to its inherent cross compilable nature
if one needs to limit the targets please set it in local.conf or
via a bbappend in own layer e.g.

LLVM_TARGETS_TO_BUILD = "ARM"

if you only need 32bit arm or

LLVM_TARGETS_TO_BUILD = "ARM;Aarch64"

in case of both arm 32bit and 64bit

fixes #33