kraj/meta-clang #34: Using meta-clang under petalinux.

RDunkley, 2017-07-18:
I am using Petalinux from Xilinx. It basically provides Yocto Krogoth in a commercial package with support for their FPGA SOCs. When I add clang I get the following:

ERROR: Nothing PROVIDES 'binutils-cross-arm' (but /home/tester/petalinux/components/yocto/source/arm/layers/meta-clang/recipes-devtools/clang/clang-cross_git.bb DEPENDS on or otherwise requires it)
ERROR: binutils-cross-arm was skipped: PREFERRED_PROVIDER_virtual/arm-linux-gnueabihf-binutils set to binutils-cross-xilinx-arm, not binutils-cross-arm
ERROR: Nothing PROVIDES 'binutils-cross-arm' (but /home/tester/petalinux/components/yocto/source/arm/layers/meta-clang/recipes-devtools/clang/clang-cross_git.bb DEPENDS on or otherwise requires it)
ERROR: binutils-cross-arm was skipped: PREFERRED_PROVIDER_virtual/arm-linux-gnueabihf-binutils set to binutils-cross-xilinx-arm, not binutils-cross-arm
ERROR: Nothing PROVIDES 'binutils-cross-arm' (but /home/tester/petalinux/components/yocto/source/arm/layers/meta-clang/recipes-devtools/clang/clang-cross_git.bb DEPENDS on or otherwise requires it)
ERROR: binutils-cross-arm was skipped: PREFERRED_PROVIDER_virtual/arm-linux-gnueabihf-binutils set to binutils-cross-xilinx-arm, not binutils-cross-arm
ERROR: Nothing PROVIDES 'binutils-cross-arm' (but /home/tester/petalinux/components/yocto/source/arm/layers/meta-clang/recipes-devtools/clang/clang-cross_git.bb DEPENDS on or otherwise requires it)
ERROR: binutils-cross-arm was skipped: PREFERRED_PROVIDER_virtual/arm-linux-gnueabihf-binutils set to binutils-cross-xilinx-arm, not binutils-cross-arm
ERROR: Nothing PROVIDES 'binutils-cross-arm' (but /home/tester/petalinux/components/yocto/source/arm/layers/meta-clang/recipes-devtools/clang/clang-cross_git.bb DEPENDS on or otherwise requires it)
ERROR: binutils-cross-arm was skipped: PREFERRED_PROVIDER_virtual/arm-linux-gnueabihf-binutils set to binutils-cross-xilinx-arm, not binutils-cross-arm
NOTE: Preparing RunQueue

It seems to me like Xilinx is renaming their binutils-cross-arm to binutils-cross-xilinx-arm which is causing a conflict with the one clang is attempting to build. Any advice on the correct way to fix this?

kraj, 2017-07-18:
I hope you are using krogoth branch of meta-clang too

kraj, 2017-07-18:
and yes if xilinx layer is providing its own binutils then they should provide all the virtual packages that oe-core binutils does

RDunkley, 2017-07-18:
Yea I tried it with krogoth, morty, and even master. Same issue. I agree, to fix it for now I updated clang-cross_git.bb and compiler-rt_git.bb to use the Xilinx version and that seemed to work.

kraj, 2017-07-19:
@RDunkley cool. I think you can file a bug for meta-xilinx to fullfil the provides as required for oe-core. Once they do that you dont have to patch anything else.

closed 2017-07-28 by RDunkley