kraj/meta-clang #36: does not build with glibc 2.25.90

pohly, 2017-07-18:
meta-clang is e7a7590a693c42978eb160c41aeac98ab7320df6. I'm using it as part of refkit. While trying to update to the latest revisions of various layers (https://github.com/intel/intel-iot-refkit/pull/246) I noticed that glibc 2.25.90 from OE-core master breaks several recipes in meta-clang.

compile-rt needs some patches which (partially) were committed upstream. See https://github.com/pohly/intel-iot-refkit/tree/updates/meta-refkit-core/bbappends/meta-clang/recipe-devtools/clang

libcxx fails with "xlocale.h not found". I'm stuck on that one:
```
| /fast/build/refkit/pr/tmp-glibc/work/corei7-64-refkit-linux/libcxx/5.0.0+gitAUTOINC+f65227fd46_b2e9337b06_065ab9eca7-r0/git/projects/libcxx/include/__locale:39:11: fatal error: 'xlocale.h' file not found
| # include <xlocale.h>
|           ^~~~~~~~~~~
| 1 error generated.
| lib/CMakeFiles/cxx_objects.dir/build.make:329: recipe for target 'lib/CMakeFiles/cxx_objects.dir/__/src/iostream.cpp.o' failed
| make[2]: *** [lib/CMakeFiles/cxx_objects.dir/__/src/iostream.cpp.o] Error 1
...
| WARNING: exit code 2 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /fast/build/refkit/pr/tmp-glibc/work/corei7-64-refkit-linux/libcxx/5.0.0+gitAUTOINC+f65227fd46_b2e9337b06_065ab9eca7-r0/temp/log.do_compile.30669)
ERROR: Task (/fast/work/intel-iot-refkit-pr/openembedded-core/../meta-clang/recipes-devtools/clang/libcxx_git.bb:do_compile) failed with exit code '1'
```

kraj, 2017-07-18:
The fix is

1. Remove xlocal.h

if you end up with errors

replace with locale.h

pohly, 2017-07-18:
Remove from what?

projects/libcxx/include/__locale seems to be a generated file, so I cannot simply patch it in the source code.

kraj, 2017-07-18:
its not generated its a patch to libcxx component thats needed.

pohly, 2017-07-18:
projects/libcxx/include/__locale is there after "git -c unpack libcxx", but git doesn't know anything about it: "git grep xlocale.h" does not find it, and after modifying it, "git diff" also doesn't show the change.

So where does it come from?

pohly, 2017-07-18:
Now I get it. projects/libcxx is a separate git repo. I was running git in the top-level "git" directory, which is from github.com/llvm-mirror/llvm.git. How... cunning.

Is someone already working on this problem? I'm a bit busy with other problems, and this looks like an upstream issue.

kraj, 2017-07-18:
yes, its an upstream issue, I dont think anyone is working on it. I might get to this next week. Since this week I am traveling

closed 2017-07-20 by kraj

pohly, 2017-07-20:
@kraj thanks for resolving it in meta-clang. I had also submitted a fix upstream but then didn't get around to updating this issue or fixing meta-clang itself:

https://reviews.llvm.org/D35616

kraj, 2017-07-24:
your fix is ok but I think its including locale.h redundantly, I have submitted another fix which hopefully get applied soon

commit 81507c02 (2017-07-20): libcxx: Do not include xlocale.h

Thsi file is removed from glibc >= 2.26

Fixes #36