kraj/meta-clang #41: Segfault with Yocto Pyro

raymanfx, 2017-09-21:
Hi all,

I'm trying to compile Beignet (OpenCL implementation), which depends on clang/compiler-rt and friends.
The package compiler-rt fails with the following stacktrace:

|   4.	Running pass 'Live Interval Analysis' on function '@main'
| 
|   clang-4.0: error: unable to execute command: Segmentation fault (core
|   dumped)
| 
|   clang-4.0: error: clang frontend command failed due to signal (use -v to
|   see invocation)
| 
|   clang version 4.0.1 (git://github.com/llvm-mirror/clang.git
|   21fe7e8f8ab44b67238af7bf9ba9d8afdf0c0e2c)
|   (git://github.com/llvm-mirror/llvm.git
|   5ed13a6cd922be74ad12b1ba36c15906336e0e3b)
| 
|   Target: x86_64-poky-linux
| 
|   Thread model: posix

Does anyone hit the same issue?
I'm using Yocto Pyro (and the appropriate Pyro branch of this layer).

kraj, 2017-09-21:
Can you provide a pre-processed file ?

raymanfx, 2017-09-22:
Hi, I was able to narrow it down to a "broken" (?!) compiler flag.
My test call is like this (taken straight from Yocto's do_configure step for compiler-rt):

/home/redqueen/Schreibtisch/csp/yocto/build/tmp/work/corei7-64-poky-linux/compiler-rt/4.0.0+gitAUTOINC+76ab2e5c9b-r0/recipe-sysroot-native/usr/bin/x86_64-poky-linux/x86_64-poky-linux-clang \
    -m64 -march=corei7 -mtune=corei7 -mfpmath=sse -msse4.2 -mlittle-endian \
    -D__extern_always_inline=inline -no-integrated-as \
    -Wno-error=unused-command-line-argument \
    --sysroot=/home/redqueen/Schreibtisch/csp/yocto/build/tmp/work/corei7-64-poky-linux/compiler-rt/4.0.0+gitAUTOINC+76ab2e5c9b-r0/recipe-sysroot \
    -O2 -pipe -g -feliminate-unused-debug-types \
    -fdebug-prefix-map=/home/redqueen/Schreibtisch/csp/yocto/build/tmp/work/corei7-64-poky-linux/compiler-rt/4.0.0+gitAUTOINC+76ab2e5c9b-r0=/usr/src/debug/compiler-rt/4.0.0+gitAUTOINC+76ab2e5c9b-r0 \
    -fdebug-prefix-map=/home/redqueen/Schreibtisch/csp/yocto/build/tmp/work/corei7-64-poky-linux/compiler-rt/4.0.0+gitAUTOINC+76ab2e5c9b-r0/recipe-sysroot-native= \
    -fdebug-prefix-map=/home/redqueen/Schreibtisch/csp/yocto/build/tmp/work/corei7-64-poky-linux/compiler-rt/4.0.0+gitAUTOINC+76ab2e5c9b-r0/recipe-sysroot= \
    -m64 -march=corei7 -mtune=corei7 -mfpmath=sse -msse4.2 -mlittle-endian \
    -D__extern_always_inline=inline -no-integrated-as \
    -Wno-error=unused-command-line-argument \
    --sysroot=/home/redqueen/Schreibtisch/csp/yocto/build/tmp/work/corei7-64-poky-linux/compiler-rt/4.0.0+gitAUTOINC+76ab2e5c9b-r0/recipe-sysroot \
    -o CMakeFiles/cmTC_1faec.dir/testCCompiler.c.o -c \
    /home/redqueen/Schreibtisch/csp/yocto/build/tmp/work/corei7-64-poky-linux/compiler-rt/4.0.0+gitAUTOINC+76ab2e5c9b-r0/build/CMakeFiles/CMakeTmp/testCCompiler.c

This makes clang sefault.
When I remove the -O2 optimization argument, it compiles fine.
I'm not sure why this is happening and will need to further investigate.

kraj, 2017-09-22:
@raymanfx clang has been updated to 4.0.1 release on pyro can you try again ?

raymanfx, 2017-09-25:
Looking here: https://github.com/kraj/meta-clang/commits/pyro, that happened a long time ago already, I am on the latest HEAD of the Pyro branch, so that won't make a difference.

raymanfx, 2017-09-25:
After the segfault I described above, clang-native royally fails other tests as well, it's completely broken for me (when trying to compile "compiler-rt").
I am using a non-standard GCC version in my tree because of certain proprietary code. I'll try to revert to the standard Pyro GCC and try again.

raymanfx, 2017-10-10:
Okay, it works now :)
Turns out Ubuntu (17.04) is to blame here, the host toolchain it ships with seems to generate broken code in clang-native for some reason.
I switched to Fedora 26 a week ago and was able to compile beignet.

closed 2017-10-10 by raymanfx

kraj, 2017-10-10:
thanks for the update