kraj/meta-clang #43: The file /usr/include/llvm/LinkAllIR.h is installed by both rust-llvm-native and clang-native, aborting

zzeroo, 2017-10-08:
## Version(s) of meta-rust
7f80e6d68b48a0a2446a606dd35901db4505d61a (HEAD -> master, upstream/master)

## Version(s) of poky and/or oe-core
```
Build Configuration:
BB_VERSION        = "1.35.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "universal"
TARGET_SYS        = "arm-poky-linux-gnueabi"
MACHINE           = "bananapro"
DISTRO            = "eisenhardt"
DISTRO_VERSION    = "1.0"
TUNE_FEATURES     = "arm armv7ve vfp neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU        = "hard"
meta
meta-poky
meta-yocto-bsp    = "master:fcdfe09d9cc4e2977bd602dfd18f729cd9f6d73d"
meta-clang        = "master:5176136fe731f0e8e526fe7252e607dbafb3ac92"
meta-bsp_lemaker  = "development:dad114e049e7137691aa75d4c7fee77619fe75ee"
meta-xmz-mod-touch = "development:678252ce7693b4c64cdb10979e25d1e385c22e64"
meta-oe           = "master:5c9a4a08844c215ebd6f2146f50de753b8e7ecef"
meta-rust         = "master:7f80e6d68b48a0a2446a606dd35901db4505d61a"
```

## Expected result
No error

## Actual result
Error `The file /usr/include/llvm/LinkAllIR.h is installed by both rust-llvm-native and clang-native, aborting`

## Steps to reproduce
Include the `meta-rust` and `meta-clang` in your local bblayers. Then create a recipe with `DEPENDS += "clang-native"` and `bitbake` this.

I'm unsure where to report this issue. So I will here and in meta-rust

kraj, 2017-10-08:
if rust is providing its own llvm then you dont need clang I think.

zzeroo, 2017-10-08:
One of my own projects need `libclang.so`, and Rust of course, too. I've tried both, default gcc compiler and `DEPENDS += "clang-native"` hits the error. And `TOOLCHAIN ?= "clang"` hits the error, too.

kraj, 2017-10-08:
I think you can write a bbappend for rust-llvm-native in your layer and in that
```
do_install_append() {
  rm -rf  ${D}${includedir}/llvm/LinkAllIR.h
}
```

zzeroo, 2017-10-11:
@kraj Unfortunately this is not working ...

```bash
# rust-llvm_%.bbappend
do_install_append() {
  rm -rf  ${D}${includedir}/llvm/LinkAllIR.h
  rm -rf  ${D}${includedir}/llvm/PassInfo.h
}
```
 results in this error.

```bash
cargo:warning=/home/build/shared/build-bananapro/tmp/work/x86_64-linux/rust-native/1.20.0-r0/recipe-sysroot-native/usr/include/llvm/PassSupport.h:26:27: fatal error: llvm/PassInfo.h: No such file or directory
```

Without rm `PassInfo.h` I tigger the original error from above.

kraj, 2017-10-11:
rust-native seems to be needing development headers and libraries from llvm, so deleting just one file is not going to be enough, there are two plausible approaches.

1. Install rust-llvm's headers into /usr/include/rust-llvm
2. See if you can live with llvm headers/libs from clang them delete then completely from rust-llvm and add a depends on clang-native to get them for building rust-native
3. Ensure that rust-llvm-native and clang-native are not in DEPENDS of same recipe

Explore if rust can be build with llvm from meta-clang then just create dependency on meta-clang and remove llvm recipes from meta-rust.

In short you need one llvm,

cardoe, 2017-10-18:
Just wondering why the clang package is providing LLVM when it would seem like you'd have another recipe to build LLVM and then one for clang. Granted its been ages since I've had to build clang and llvm but I recall them being separate tarballs in the past. I'm adding `PROVIDES = "llvm"` to rust-llvm to help the situation.

kraj, 2017-10-18:
ideally, every llvm based project could use a common libllvm.so but thats not the case. For any tools that you want to develop on top of say clang you pretty much need the development headers and libraries from llvm.

cardoe, 2017-10-18:
Well I see you're adding llvm to openembedded-core so I could certainly try to take advantage of that in the future. Looks like its a git only recipe however. Rust can switch between using its bundled llvm and an upstream version. That's how I package it up in Gentoo.

kraj, 2017-10-18:
That seems right, if rust can use upstream llvm then may be depend on the one in oe-core

zzeroo, 2017-11-09:
Hi @kraj I hope to bump this issue again. As a reminder and if that was not clear yet a project of mine needs just a libclang.so (version 3.9 or greater). 
Unfortunately I can't use your meta-clang recipes direct because of the error (above). Can I extract libclang.so before installing, or skip all other installing steps?

Do you have any idea what else I could do? Looks like meta-rust is ignoring this issue.

kraj, 2017-11-09:
you can write a bbappend in your layer to delete in do_install_append

kraj, 2017-11-15:
@zzeroo I think there is nothing we should fix in meta-clang for this. and tackling it either ways as described below

1. making meta-rust depend on meta-clang 
2. Write a bbappend in own layer to apply needed tweaks

closed 2017-11-29 by zzeroo