kraj/meta-clang #45: It's not possible to use LTO

zavorka, 2017-11-12:
Adding `-flto=thin -fuse-ld=gold` to the `CCARGS` variable causes linking errors. Have anyone even tried using LTO with yocto?

kraj, 2017-11-12:
@zavorka I havent used it distro wide, but on some packages I have used thin-lto. lto works ok on yocto with gcc and I thought it works well with clang as well. Can you describe the issue you are seeing.

zavorka, 2017-11-14:
@kraj Well, the first problem is that the `clang-cross-arm` recipe won't produce `LLVMgold.so` but that's easily fixed by appending `-DLLVM_BINUTILS_INCDIR=${includedir}` to the `EXTRA_OECMAKE` variable.

Then I put the `flto=thin -fuse-ld=gold` parameters into `TUNE_CCARGS` and substitute `ar`, `nm` and `ranlib` for their `llvm-` prefixed counterparts. Unfortunately, the `gold` linker produced by `binutils-cross-arm` can't handle the `armelf_linux_eabi` emulation (according to the error messages I get flooded with). Funnily enough, if I try copying the `ld.gold` binary from `binutils-native` into the `recipe-sysroot`, unsupported emulation problem is gone, though immediately replaced by `gold`'s inability to read the bitcode (which doesn't make sense -- it loads the `LLVMgold.so` plugin just ok).

Finally, I've got `musl` built with ThinLTO once I replaced `gold` for `lld`. Sadly, there's no `lld-cross-arm` recipe yet, so there's a non-trivial amount of work yet to be done.

kraj, 2017-11-15:
@zavorka thanks for trying it out. There are bunch of issues you are seeing

please submit a patch for clang-cross-arm to this layer
Also add a new issue to add lld-cross 

binutils-cross gold issue needs investigation this looks like OE bug to me.

kraj, 2019-01-28:
@zavorka now lld is available in cross toolchain adding `-fuse-ld=lld` should be able to use it. Can you try your experiment with latest meta-clang ?

kraj, 2019-02-08:
https://github.com/kraj/meta-clang/commit/00b9cd93d591099071e7087b9a354da6338dc1b7

lets clang itself build with Thin LTO, so I am interested to know if it will work for other recipes

kraj, 2019-02-10:
https://github.com/kraj/meta-clang/pull/96

should help with this issue can you verify with latest master ?

kraj, 2019-02-11:
@zavorka can you try latest master, -fuse-ld=lld should behave better hopefully.

kraj, 2019-02-13:
LTO work and -fuse-ld=lld works too in SDK environment as well as internal builds.

closed 2019-02-13 by kraj

anilavakundu, 2020-08-11:
@zavorka  I was hoping if you could tell me which EXTRA_OECMAKE did you change LLVMgold.so isn't built for me either