kraj/meta-clang #46: Trying to build for raspberry pi

Radhika-aquarius, 2017-11-27:
I am trying to use morty branch of meta-clang.I am receiving the following error when i try to bitbake clang
ERROR: clang-native-3.9.1+gitAUTOINC+46079999e9_c3709e72d2-r0 do_compile: oe_runmake failed
ERROR: clang-native-3.9.1+gitAUTOINC+46079999e9_c3709e72d2-r0 do_compile: Function failed: do_compile (log file is located at build/tmp/work/x86_64-linux/clang-native 3.9.1+gitAUTOINC+46079999e9_c3709e72d2-r0/temp/log.do_compile.67346)

ERROR: clang-3.9.1+gitAUTOINC+46079999e9_c3709e72d2-r0 do_fetch: Fetcher failure for URL: 'git://github.com/llvm-mirror/llvm.git;protocol=git;branch=release_39;name=llvm'. Unable to fetch URL from any source.
ERROR: clang-3.9.1+gitAUTOINC+46079999e9_c3709e72d2-r0 do_fetch: Function failed: base_do_fetch
ERROR: Task (sources/meta-clang/recipes-devtools/clang/clang_git.bb:do_fetch) failed with exit code '1'

kraj, 2017-11-29:
@Radhika-aquarius Can you try to do a manual clone of the repo and see if you are able to do it

```
git clone git://github.com/llvm-mirror/llvm.git
```
look into any network firewall issues you might have w.r.t. git protocol.

Radhika-aquarius, 2017-11-30:
Yes,I am able to clone it but where should I place it in the directory structure?

kraj, 2017-11-30:
OK then it should work even with bitbake fetch.  I wonder if it was a network glitch, are you able to to reproduce it always. Please update meta-clang to latest and use morty branch

Radhika-aquarius, 2017-11-30:
I am using morty 3.9.1 and bitbake clang is giving me the error repeatedly.I don't feel its a network glitch because error occurs repeatedly.Please provide detailed solution as I am new to this environment.

kraj, 2017-12-01:
this works fine here,  
```
git clone git://github.com/openembedded/openembedded-core -b morty
cd openembedded-core
git clone git://github.com/openembedded/bitbake -b 1.32
git clone git:/github.com/kraj/meta-clang -b morty

. ./oe-init-build-env

bitbake -cpatch clang
```

please retry and see if it helps

Radhika-aquarius, 2017-12-01:
On giving the following command:
git clone git:/github.com/kraj/meta-clang -b morty
I get this issue:
Cloning into 'meta-clang'...
ssh: Could not resolve hostname git: Name or service not known
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

kraj, 2017-12-01:
there was a missing '/' there after git:/ retry

Radhika-aquarius, 2017-12-01:
bitbake -cpatch clang
Loading cache: 100% |############################################| Time: 0:00:00
Loaded 1316 entries from dependency cache.
ERROR: Nothing PROVIDES 'clang'. Close matches:
  slang

Summary: There was 1 ERROR message shown, returning a non-zero exit code.

kraj, 2017-12-01:
i missed specifying 
```
bitbake-layers add-layer ../meta-clang
```
retry after running this

Radhika-aquarius, 2017-12-05:
I am able to build openembedded-core individually with the way explained by you.I am trying to build for raspberry pi which still gives an error.I have an environment provided by the supplier and I have to build on top of that.I am able to build for 5.0.1 but not morty. Can you please tell me how can I look into the error oe_runmakae which I get for morty.I want to know what is changed in 5.0.1 that it builds for it but fails for morty.

kraj, 2017-12-05:
try to use pyro branch for meta-clang

kraj, 2018-05-13:
@Radhika-aquarius  Can we close this issue or do you still need help ?