kraj/meta-clang #47: Building coreclr fails when building with clang 5.0.1

sam0302, 2017-12-11:
There seems to be a bug with clang 5.0.1 version as mentioned in https://reviews.llvm.org/D33852
As a result when i try to build coreclr, it gives me error. This has been mentioned in : https://github.com/dotnet/coreclr/issues/15197

Is it possible to create a patch with the changes mentioned in https://reviews.llvm.org/D33852  and update it when clang 5.1 is officially released?

kraj, 2017-12-13:
@sam0302 seems a good patch.  I will test it out and push it out by this weekend unless someone else provides a patch earlier.

kraj, 2017-12-13:
@sam0302 can you test it now by pulling latest master and let me know if issue is gone. If so we will backport it to rocko as well

kraj, 2018-01-10:
should be fixed please reopen if still not fixed.

closed 2018-01-10 by kraj