kraj/meta-clang #48: Unable to build clang for IntelCE i686

romanek-adam, 2018-01-09:
Hi,

recently I've started the work on #44. Unfortunately I have some trouble building clang when the target is i686 (I'm doing this in context of IntelCE builds). I'm building it as part of Yocto 2.1 using meta-clang from master.

```
ERROR: clang-5.0.1+gitAUTOINC+9bc59d4c9b_232230afd3-r0 do_compile: Function failed: do_compile (log file is located at /opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/work/core2-32-rdk-linux/clang/5.0.1+gitAUTOINC+9bc59d4c9b_232230afd3-r0/temp/log.do_compile.23668)
ERROR: Logfile of failure stored in: /opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/work/core2-32-rdk-linux/clang/5.0.1+gitAUTOINC+9bc59d4c9b_232230afd3-r0/temp/log.do_compile.23668
Log data follows:
| DEBUG: Executing shell function do_compile
| [  0%] cd /opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/work/core2-32-rdk-linux/clang/5.0.1+gitAUTOINC+9bc59d4c9b_232230afd3-r0/build && /opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/sysroots/x86_64-linux/usr/bin/cmake -E make_directory /opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/work/core2-32-rdk-linux/clang/5.0.1+gitAUTOINC+9bc59d4c9b_232230afd3-r0/build/NATIVE
| [  0%] cd /opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/work/core2-32-rdk-linux/clang/5.0.1+gitAUTOINC+9bc59d4c9b_232230afd3-r0/build/NATIVE && /opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/sysroots/x86_64-linux/usr/bin/cmake -G Ninja /opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/work/core2-32-rdk-linux/clang/5.0.1+gitAUTOINC+9bc59d4c9b_232230afd3-r0/git -DLLVM_TARGET_IS_CROSSCOMPILE_HOST=TRUE
| FAILED: NATIVE/CMakeCache.txt
| cd /opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/work/core2-32-rdk-linux/clang/5.0.1+gitAUTOINC+9bc59d4c9b_232230afd3-r0/build/NATIVE && /opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/sysroots/x86_64-linux/usr/bin/cmake -G Ninja /opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/work/core2-32-rdk-linux/clang/5.0.1+gitAUTOINC+9bc59d4c9b_232230afd3-r0/git -DLLVM_TARGET_IS_CROSSCOMPILE_HOST=TRUE
| -- The C compiler identification is GNU 5.4.0
| -- The CXX compiler identification is GNU 5.4.0
| -- The ASM compiler identification is GNU
| -- Found assembler: /opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/sysroots/x86_64-linux/usr/bin/i686-rdk-linux/i686-rdk-linux-gcc
| -- Check for working C compiler using: Ninja
| -- Check for working C compiler using: Ninja -- broken
| CMake Error at /opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/sysroots/x86_64-linux/usr/share/cmake-3.4/Modules/CMakeTestCCompiler.cmake:61 (message):
|   The C compiler
|   "/opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/sysroots/x86_64-linux/usr/bin/i686-rdk-linux/i686-rdk-linux-gcc"
|   is not able to compile a simple test program.
| 
|   It fails with the following output:
| 
|    Change Dir: /opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/work/core2-32-rdk-linux/clang/5.0.1+gitAUTOINC+9bc59d4c9b_232230afd3-r0/build/NATIVE/CMakeFiles/CMakeTmp
| 
| 
| 
|   Run Build
|   Command:"/opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/sysroots/x86_64-linux/usr/bin/ninja"
|   "cmTC_37a9f"
| 
|   [ 50%] Building C object CMakeFiles/cmTC_37a9f.dir/testCCompiler.c.o
| 
|   [100%] Linking C executable cmTC_37a9f
| 
|   FAILED: cmTC_37a9f
| 
|   : &&
|   /opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/sysroots/x86_64-linux/usr/bin/i686-rdk-linux/i686-rdk-linux-gcc
|   -Os -pipe -g -feliminate-unused-debug-types
|   -fdebug-prefix-map=/opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/work/core2-32-rdk-linux/clang/5.0.1+gitAUTOINC+9bc59d4c9b_232230afd3-r0=/usr/src/debug/clang/5.0.1+gitAUTOINC+9bc59d4c9b_232230afd3-r0
|   -fdebug-prefix-map=/opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/sysroots/x86_64-linux=
|   -fdebug-prefix-map=/opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/sysroots/smtg7401-debug=
|   -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong
|   -Wl,-z,relro,-z,now -Wl,-rpath,/lib,-rpath,/usr/lib
|   CMakeFiles/cmTC_37a9f.dir/testCCompiler.c.o -o cmTC_37a9f && :
| 
| 
|   /opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/sysroots/x86_64-linux/usr/libexec/i686-rdk-linux/gcc/i686-rdk-linux/5.4.0/ld:
|   error: cannot open crt1.o: No such file or directory
| 
| 
|   /opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/sysroots/x86_64-linux/usr/libexec/i686-rdk-linux/gcc/i686-rdk-linux/5.4.0/ld:
|   error: cannot open crti.o: No such file or directory
| 
| 
|   /opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/sysroots/x86_64-linux/usr/libexec/i686-rdk-linux/gcc/i686-rdk-linux/5.4.0/ld:
|   error: cannot open crtbegin.o: No such file or directory
| 
| 
|   /opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/sysroots/x86_64-linux/usr/libexec/i686-rdk-linux/gcc/i686-rdk-linux/5.4.0/ld:
|   error: cannot open crtend.o: No such file or directory
| 
| 
|   /opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/sysroots/x86_64-linux/usr/libexec/i686-rdk-linux/gcc/i686-rdk-linux/5.4.0/ld:
|   error: cannot open crtn.o: No such file or directory
| 
| 
|   /opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/sysroots/x86_64-linux/usr/libexec/i686-rdk-linux/gcc/i686-rdk-linux/5.4.0/ld:
|   error: cannot find -lssp_nonshared
| 
| 
|   /opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/sysroots/x86_64-linux/usr/libexec/i686-rdk-linux/gcc/i686-rdk-linux/5.4.0/ld:
|   error: cannot find -lgcc
| 
| 
|   /opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/sysroots/x86_64-linux/usr/libexec/i686-rdk-linux/gcc/i686-rdk-linux/5.4.0/ld:
|   error: cannot find -lgcc_s
| 
| 
|   /opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/sysroots/x86_64-linux/usr/libexec/i686-rdk-linux/gcc/i686-rdk-linux/5.4.0/ld:
|   error: cannot find -lc
| 
| 
|   /opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/sysroots/x86_64-linux/usr/libexec/i686-rdk-linux/gcc/i686-rdk-linux/5.4.0/ld:
|   error: cannot find -lgcc
| 
| 
|   /opt/onemw/oe-builds/build-intel-2017-11-09-csa/onemw/build-smtg7401/tmp/sysroots/x86_64-linux/usr/libexec/i686-rdk-linux/gcc/i686-rdk-linux/5.4.0/ld:
|   error: cannot find -lgcc_s
| 
|   collect2: error: ld returned 1 exit status
| 
|   ninja: build stopped: subcommand failed.

```

It looks like clang build system uses i686-rdk-linux-gcc for linking whereas it should probably use i686-rdk-linux-ld directly or at least provide --sysroot=... param to i686-rdk-linux-gcc. When I use i686-rdk-linux-ld manually to build the test code everything works as expected.

Please advise on how to fix this.

kraj, 2018-01-10:
@romanek-adam good that you took on #44, I am very interested in it. It seems you are using krogoth branch for oe-core but different branch for meta-clang. Usually you should be using krogoth branch for meta-clang too. Although I would suggest to use master of all repos for such a work.

Lorac, 2018-03-08:
@romanek-adam I don't know if you fixed it by now, but for our case, we patched the 

https://github.com/Kitware/CMake/blob/master/Modules/CMakeDetermineSystem.cmake#L89

by hardcoding the CMAKE_TOOLCHAIN_FILE path in a script that our developers run.

You might wanna check the resulted [OEToolchainConfig.cmake](https://github.com/openembedded/openembedded-core/blob/master/meta/recipes-devtools/cmake/cmake/OEToolchainConfig.cmake) file. We hardcode paths there too.

In our case we don't want to source the environment, we just hardcode paths in cmake configuration files and it works very well when switching between cross-compiling and normal environment.

kraj, 2018-03-10:
@romanek-adam I think combining meta-clang/master with yocto 2.1 release is not a tested combo. I think you should be using master of every layer for this to fix.

romanek-adam, 2019-01-11:
For anyone interested - recently I forked this repository and created a Yocto 2.1 compatible branch based on this repo's master - https://github.com/romanek-adam/meta-clang/commits/master-yocto-2.1. It works perfectly, Clang 7 is compiled without any issues and Clang Static Analyzer can be used via scan-build.bbclass.

At this point I assume this issue can be closed.

closed 2019-01-11 by romanek-adam