kraj/meta-clang #51: Missing compiler-rt-staticdev in SDK

Lorac, 2018-03-08:
With a recipe using `TOOLCHAIN = "clang"` and its main package installed in an image, `bitbake myImage -c populate_sdk` contains clang compiler, but missing `libclang_rt.builtins-aarch64.a`

When cross-compiling with `cmake` tries to compile a test program it fails.

```
  MYSDKPATH/sysroots/x86_64-pikosdk-linux/usr/bin/aarch64-piko-linux/aarch64-piko-linux-ld:
  cannot find
MYSDKPATH/sysroots/aarch64-piko-linux/usr/lib/clang/5.0.1/lib/linux/libclang_rt.builtins-aarch64.a:
  No such file or directory
```

kraj, 2018-03-10:
Can you add 
```
TOOLCHAIN_TARGET_TASK_append = " compiler-rt-staticdev compiler-rt-dev"
```
in local.conf and see if it helps.

Lorac, 2018-03-10:
Yes, it works by adding that line, maybe I should've mentioned it... actually, we are doing that in the `myImage`,

But I don't think it should be the image or local.conf that adds it, but it should come by itself when the image is installing something that declares the `TOOLCHAIN = "clang"`, am I wrong?

kraj, 2018-03-10:
It is a bug nevertheless in sdk generation it should have included the dev packages for compiler-rt

Lorac, 2018-03-10:
To get the "libclang_rt.builtins-aarch64.a" I need to add the `compiler-rt-staticdev`, I didn't check for the `compiler-rt-dev` package

kraj, 2018-03-10:
thats ok, I think if you add compiler-rt-staticdev that should be enough, or alternatively move the .a to main package itself but that might not be best.

commit 29f52010 (2018-05-16): compiler-rt: Let dev package rdep on staticdev package

Let dev package be empty, this should solve the
dependency issues when generating SDKs

Fixes #51
commit bfd6ac76 (2018-07-06): compiler-rt: Let dev package rdep on staticdev package

Let dev package be empty, this should solve the
dependency issues when generating SDKs

Fixes #51
closed 2018-05-16 by kraj