kraj/meta-clang #52: fresh clone of poky/meta-oe/meta-clang. clang fails to build

Lorac, 2018-03-10:
```
| [  6%] Building CXX object lib/scudo/CMakeFiles/clang_rt.scudo-dynamic-x86_64.dir/scudo_crc32.cpp.o
| FAILED: lib/scudo/CMakeFiles/clang_rt.scudo-dynamic-x86_64.dir/scudo_crc32.cpp.o
| /root/yocto/build/tmp/work/core2-64-poky-linux/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0/recipe-sysroot-native/usr/bin/x86_64-poky-linux/x86_64-poky-linux-clang++  -Dclang_rt_scudo_dynamic_x86_64_EXPORTS -I/root/yocto/build/tmp/work/core2-64-poky-linux/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0/git/lib/scudo/.. -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse -mlittle-endian -D__extern_always_inline=inline -no-integrated-as -Wno-error=unused-command-line-argument -Qunused-arguments  --sysroot=/root/yocto/build/tmp/work/core2-64-poky-linux/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0/recipe-sysroot -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/root/yocto/build/tmp/work/core2-64-poky-linux/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0=/usr/src/debug/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0 -fdebug-prefix-map=/root/yocto/build/tmp/work/core2-64-poky-linux/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0/recipe-sysroot-native= -fdebug-prefix-map=/root/yocto/build/tmp/work/core2-64-poky-linux/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0/recipe-sysroot= -fvisibility-inlines-hidden -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse -mlittle-endian -D__extern_always_inline=inline -no-integrated-as -Wno-error=unused-command-line-argument -Qunused-arguments  --sysroot=/root/yocto/build/tmp/work/core2-64-poky-linux/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0/recipe-sysroot -Wall -std=c++11 -Wno-unused-parameter -fPIC    -m64 -fno-lto -fPIC -fno-builtin -fno-exceptions -fomit-frame-pointer -funwind-tables -fno-stack-protector -fno-sanitize=safe-stack -fvisibility=hidden -fno-lto -O3 -gline-tables-only -Wno-gnu -Wno-variadic-macros -Wno-c99-extensions -Wno-non-virtual-dtor -fbuiltin -fno-rtti -march=armv8.1-a -MD -MT lib/scudo/CMakeFiles/clang_rt.scudo-dynamic-x86_64.dir/scudo_crc32.cpp.o -MF lib/scudo/CMakeFiles/clang_rt.scudo-dynamic-x86_64.dir/scudo_crc32.cpp.o.d -o lib/scudo/CMakeFiles/clang_rt.scudo-dynamic-x86_64.dir/scudo_crc32.cpp.o -c /root/yocto/build/tmp/work/core2-64-poky-linux/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0/git/lib/scudo/scudo_crc32.cpp
| error: unknown target CPU 'armv8.1-a'
| [  8%] Building CXX object lib/scudo/CMakeFiles/clang_rt.scudo-x86_64.dir/scudo_crc32.cpp.o
| FAILED: lib/scudo/CMakeFiles/clang_rt.scudo-x86_64.dir/scudo_crc32.cpp.o
| /root/yocto/build/tmp/work/core2-64-poky-linux/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0/recipe-sysroot-native/usr/bin/x86_64-poky-linux/x86_64-poky-linux-clang++   -I/root/yocto/build/tmp/work/core2-64-poky-linux/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0/git/lib/scudo/.. -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse -mlittle-endian -D__extern_always_inline=inline -no-integrated-as -Wno-error=unused-command-line-argument -Qunused-arguments
--sysroot=/root/yocto/build/tmp/work/core2-64-poky-linux/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0/recipe-sysroot -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/root/yocto/build/tmp/work/core2-64-poky-linux/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0=/usr/src/debug/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0 -fdebug-prefix-map=/root/yocto/build/tmp/work/core2-64-poky-linux/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0/recipe-sysroot-native= -fdebug-prefix-map=/root/yocto/build/tmp/work/core2-64-poky-linux/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0/recipe-sysroot= -fvisibility-inlines-hidden -m64 -march=core2 -mtune=core2 -msse3 -mfpmath=sse -mlittle-endian -D__extern_always_inline=inline -no-integrated-as -Wno-error=unused-command-line-argument -Qunused-arguments  --sysroot=/root/yocto/build/tmp/work/core2-64-poky-linux/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0/recipe-sysroot -Wall -std=c++11 -Wno-unused-parameter    -m64 -fno-lto -fPIC -fno-builtin -fno-exceptions -fomit-frame-pointer -funwind-tables -fno-stack-protector -fno-sanitize=safe-stack -fvisibility=hidden -fno-lto -O3 -gline-tables-only -Wno-gnu -Wno-variadic-macros -Wno-c99-extensions -Wno-non-virtual-dtor -fbuiltin -fno-rtti -march=armv8.1-a -MD -MT lib/scudo/CMakeFiles/clang_rt.scudo-x86_64.dir/scudo_crc32.cpp.o -MF lib/scudo/CMakeFiles/clang_rt.scudo-x86_64.dir/scudo_crc32.cpp.o.d -o lib/scudo/CMakeFiles/clang_rt.scudo-x86_64.dir/scudo_crc32.cpp.o -c /root/yocto/build/tmp/work/core2-64-poky-linux/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0/git/lib/scudo/scudo_crc32.cpp
| error: unknown target CPU 'armv8.1-a'
```

I cloned meta-oe, poky, meta-clang, and my machine is `genericx86-64`

When I invoke `bitbake clang` it fails. Did I misconfigured something? I am not sure why I get `unknown target CPU 'armv8.1-a'`

kraj, 2018-03-10:
@Lorac it seems to be a bug in local patch we are carrying 

https://github.com/kraj/meta-clang/blob/master/recipes-devtools/clang/compiler-rt/0005-Pass-march-armv8.1-a-instead-of-mcrc.patch

Can you try to disable that patch and see if it helps

Lorac, 2018-03-11:
I disabled the patch and it works without a problem!

Thanks to the `PARALLEL_MAKE` you have in the `do_compile` section, I had to tune it down, I was running out of ram... 😞

kraj, 2018-03-11:
Glad to know it worked would it be possible to check if it still works on aarch64 

While you are here 

MACHINE = “qemuarm64” bitbake compiler-rt 

Should do it

Lorac, 2018-03-11:
That's fails because 
```
| [ 89%] Building CXX object lib/scudo/CMakeFiles/clang_rt.scudo-aarch64.dir/scudo_crc32.cpp.o
| FAILED: lib/scudo/CMakeFiles/clang_rt.scudo-aarch64.dir/scudo_crc32.cpp.o
| /root/yocto/build/tmp/work/aarch64-poky-linux/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-clang++   -I/root/yocto/build/tmp/work/aarch64-poky-linux/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0/git/lib/scudo/.. -mlittle-endian -D__extern_always_inline=inline -no-integrated-as -Wno-error=unused-command-line-argument -Qunused-arguments  --sysroot=/root/yocto/build/tmp/work/aarch64-poky-linux/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0/recipe-sysroot -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/root/yocto/build/tmp/work/aarch64-poky-linux/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0=/usr/src/debug/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0 -fdebug-prefix-map=/root/yocto/build/tmp/work/aarch64-poky-linux/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0/recipe-sysroot-native= -fdebug-prefix-map=/root/yocto/build/tmp/work/aarch64-poky-linux/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0/recipe-sysroot= -fvisibility-inlines-hidden -mlittle-endian -D__extern_always_inline=inline -no-integrated-as -Wno-error=unused-command-line-argument -Qunused-arguments  --sysroot=/root/yocto/build/tmp/work/aarch64-poky-linux/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0/recipe-sysroot -Wall -std=c++11 -Wno-unused-parameter    -march=armv8-a -fno-lto -fPIC -fno-builtin -fno-exceptions -fomit-frame-pointer -funwind-tables -fno-stack-protector -fno-sanitize=safe-stack -fvisibility=hidden -fno-lto -O3 -gline-tables-only -Wno-gnu -Wno-variadic-macros -Wno-c99-extensions -Wno-non-virtual-dtor -fbuiltin -fno-rtti -mcrc -MD -MT lib/scudo/CMakeFiles/clang_rt.scudo-aarch64.dir/scudo_crc32.cpp.o -MF lib/scudo/CMakeFiles/clang_rt.scudo-aarch64.dir/scudo_crc32.cpp.o.d -o lib/scudo/CMakeFiles/clang_rt.scudo-aarch64.dir/scudo_crc32.cpp.o -c /root/yocto/build/tmp/work/aarch64-poky-linux/compiler-rt/6.0.0+gitAUTOINC+9d61c78bce-r0/git/lib/scudo/scudo_crc32.cpp
| /tmp/scudo_crc32-3b366f.s: Assembler messages:
| /tmp/scudo_crc32-3b366f.s:14: Error: selected processor does not support `crc32cx w0,w0,x1'
| clang-6.0: error: assembler command failed with exit code 1 (use -v to see invocation)
```

kraj, 2018-03-11:
yeah so we still need this patch, just need to make it conditional on TARGET_ARCH being aarch64

Lorac, 2018-03-11:
Something like that?

```
SRC_URI_append_aarch64 = " \                                 
    file://0005-Pass-march-armv8.1-a-instead-of-mcrc.patch \ 
"
```

kraj, 2018-03-11:
thats a solution we  can do yes. although I Wanted to see if we could fix the CMakefile itself.

Lorac, 2018-03-11:
maybe with `CMAKE_HOST_SYSTEM_PROCESSOR`

kraj, 2018-03-11:
yes that seems a good approach

Lorac, 2018-03-11:
I would like to understand what are the drawbacks of using the `SRC_URI` way? I think it might be because it feels like two patches but done in two different ways?

kraj, 2018-03-11:
sometimes  conditional patches are not well tested and secondly, we want to have exact source
and fixing CMakefile method is upstreamable.

closed 2018-05-13 by kraj