kraj/meta-clang #56: cxxabi.h not found

sronsse, 2018-04-26:
Hi @kraj,

Managed to get 99% of my code base to compile, so thank you. I am currently running through a last issue: ```#include <cxxabi.h>``` fails as the file cannot be found. In my SDK, it located here:
```<sdk-root>/usr/include/c++/6.4.0/cxxabi.h```

It looks like this path is not searched by clang when building my program - adding the include path manually brings another error with bits/c++config.h not found (it is located in ```<sdk-root>/usr/include/c++/6.4.0/cxxabi.h/aarch64-sronsse-linux/bits/c++config.h```).

Reproducing the issue is straight-forward, any .c file including cxxabi.h will fail under the SDK environment (but not in a native host environment).

sronsse, 2018-04-26:
FYI - manually copying ```build/tmp/work/aarch64-sronsse-linux/libcxx/5.0.1+gitAUTOINC+1368f4044e_598ae4fa2e_5df6b5da0d_29d1624468-r0/build/include/c++/v1/cxxabi.h``` (and ```__cxxabi_config.h```) into my SDK sysroot ```under usr/include/c++/v1/``` "works". I am not sure what triggers the proper copy of libcxxabi's headers or how to fix it nicely.

kraj, 2018-04-27:
@sronsse do you see same issue on master branch as well ?

sronsse, 2018-04-27:
I tried it on my end and confirmed that fixed the issue! Do you have any potential idea/solution for fixing the cxxabi.h problem on rocko?

sronsse, 2018-04-28:
I am using another meta for tegra support which requires GCC6 because of CUDA9 compatibility issues with GCC7+.

kraj, 2018-04-28:
There has been changes how C++ headers are included between gcc6 and gcc7 in OE I think thats whats the issue is here. You might be able to pin point it if you look into gcc6 and gcc7 recipe common changes.

sronsse, 2018-04-28:
I'll check it out on Monday and let you know, but I'm not sure this is related - these headers are supposed to be brought by meta-clang and libcxx in particular, aren't they?

sronsse, 2018-04-30:
I confirm switching to gcc7 does not change anything. The libcxx abi from meta-clang headers are still not copied unfortunately.

kraj, 2018-05-13:
So it seems the gxx-headers from oe-core is in effect here. We have to see what we changes in oe-ocre with respect to system headers

closed 2019-08-15 by kraj