kraj/meta-clang #57: compiler-rt's lib/scudo/CMakeLists.txt enables -march=armv8.1-a by default

bernardoaraujor, 2018-05-01:
I included meta-clang into my genericx86-64 build.

However, compiler-rt presented the following error:
```
bernardo@yocto:~/dev/yocto/poky/build-qemux86-64$ bitbake compiler-rt
WARNING: Layer openmining should set LAYERSERIES_COMPAT_openmining in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer crypto-layer should set LAYERSERIES_COMPAT_crypto-layer in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer meta-phoronix should set LAYERSERIES_COMPAT_meta-phoronix in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer openmining should set LAYERSERIES_COMPAT_openmining in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer crypto-layer should set LAYERSERIES_COMPAT_crypto-layer in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer meta-phoronix should set LAYERSERIES_COMPAT_meta-phoronix in its conf/layer.conf file to list the core layer names it is compatible with.
Loading cache: 100% |####################################################################| Time: 0:00:00
Loaded 2128 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.37.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "x86_64-poky-linux"
MACHINE              = "qemux86-64"
DISTRO               = "poky"
DISTRO_VERSION       = "2.5"
TUNE_FEATURES        = "m64 core2"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "master:b6bc5b2840246ac0f0e5d8cebcdffc1a0ca2518f"
meta-oe              = "master:663bc6f0160adb570c3a7d0b64d1f7f60cb2269e"
meta-openmining      = "master:b5d0e95b872fdf0660c9415f7bcafd2df41c2333"
meta-cryptocurrency  = "master:13065cbcb2f6a4e218841ecb6bf642840c07ee80"
meta-phoronix        = "master:1bb933aecd583645624b802a7d885cdf40b57d37"
meta-clang           = "master:7a61e5a0240b87b9c1c30cc53fabf2084cecd2e3"

Initialising tasks: 100% |###############################################################| Time: 0:00:00
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: compiler-rt-6.0.1+gitAUTOINC+9d61c78bce-r0 do_compile: Function failed: do_compile (log file is located at /home/bernardo/dev/yocto/poky/build-qemux86-64/tmp/work/core2-64-poky-linux/compiler-rt/6.0.1+gitAUTOINC+9d61c78bce-r0/temp/log.do_compile.5924)
ERROR: Logfile of failure stored in: /home/bernardo/dev/yocto/poky/build-qemux86-64/tmp/work/core2-64-poky-linux/compiler-rt/6.0.1+gitAUTOINC+9d61c78bce-r0/temp/log.do_compile.5924
Log data follows:
| DEBUG: Executing shell function do_compile
| [1/524] Copying compiler-rt's sanitizer/asan_interface.h...
| [2/524] Copying compiler-rt's sanitizer/allocator_interface.h...
| [3/524] Copying compiler-rt's sanitizer/common_interface_defs.h...
| [4/524] Copying compiler-rt's sanitizer/coverage_interface.h...
| [5/524] Copying compiler-rt's sanitizer/dfsan_interface.h...
| [6/524] Copying compiler-rt's sanitizer/esan_interface.h...
| [7/524] Copying compiler-rt's sanitizer/hwasan_interface.h...
| [8/524] Copying compiler-rt's sanitizer/linux_syscall_hooks.h...
| [9/524] Copying compiler-rt's sanitizer/msan_interface.h...
| [10/524] Copying compiler-rt's sanitizer/lsan_interface.h...
| [11/524] Copying compiler-rt's sanitizer/scudo_interface.h...
| [12/524] Copying compiler-rt's sanitizer/tsan_interface.h...
| [13/524] Copying compiler-rt's sanitizer/tsan_interface_atomic.h...
| [14/524] Building CXX object lib/sanitizer_common/CMakeFiles/RTSanitizerCommonNoTermination.x86_64.dir/sanitizer_errno.cc.o
| [15/524] Building CXX object lib/sanitizer_common/CMakeFiles/RTSanitizerCommonNoTermination.x86_64.dir/sanitizer_deadlock_detector2.cc.o
| [16/524] Building CXX object lib/sanitizer_common/CMakeFiles/RTSanitizerCommonNoTermination.x86_64.dir/sanitizer_symbolizer_libbacktrace.cc.o
| [17/524] Building CXX object lib/sanitizer_common/CMakeFiles/RTSanitizerCommonNoTermination.x86_64.dir/sanitizer_file.cc.o
| [18/524] Building C object lib/builtins/CMakeFiles/clang_rt.builtins-x86_64.dir/addvsi3.c.o
| [19/524] Building C object lib/builtins/CMakeFiles/clang_rt.builtins-x86_64.dir/addtf3.c.o
| [20/524] Building CXX object lib/sanitizer_common/CMakeFiles/RTSanitizerCommonNoTermination.x86_64.dir/sanitizer_fuchsia.cc.o
| [21/524] Building CXX object lib/sanitizer_common/CMakeFiles/RTSanitizerCommonNoTermination.x86_64.dir/sanitizer_flag_parser.cc.o
| [22/524] Building CXX object lib/sanitizer_common/CMakeFiles/RTSanitizerCommonNoTermination.x86_64.dir/sanitizer_flags.cc.o
| [23/524] Building CXX object lib/sanitizer_common/CMakeFiles/RTSanitizerCommonNoTermination.x86_64.dir/sanitizer_linux_s390.cc.o
| [24/524] Building CXX object lib/sanitizer_common/CMakeFiles/RTSanitizerCommonNoTermination.x86_64.dir/sanitizer_mac.cc.o
| [25/524] Building CXX object lib/sanitizer_common/CMakeFiles/RTSanitizerCommonNoTermination.x86_64.dir/sanitizer_common.cc.o
| [26/524] Building CXX object lib/sanitizer_common/CMakeFiles/RTSanitizerCommonNoTermination.x86_64.dir/sanitizer_persistent_allocator.cc.o
| [27/524] Building CXX object lib/sanitizer_common/CMakeFiles/RTSanitizerCommonNoTermination.x86_64.dir/sanitizer_platform_limits_netbsd.cc.o
| [28/524] Building CXX object lib/sanitizer_common/CMakeFiles/RTSanitizerCommonNoTermination.x86_64.dir/sanitizer_platform_limits_linux.cc.o
| [29/524] Building CXX object lib/sanitizer_common/CMakeFiles/RTSanitizerCommonNoTermination.x86_64.dir/sanitizer_libignore.cc.o
[... 445 lines cut ...]
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

My temporary solution was to pop up a devshell and do the following modification:
```
root@yocto:~/dev/yocto/poky/build-qemux86-64/tmp/work/core2-64-poky-linux/compiler-rt/6.0.1+gitAUTOINC+9d61c78bce-r0/git# vim lib/scudo/CMakeLists.txt
root@yocto:~/dev/yocto/poky/build-qemux86-64/tmp/work/core2-64-poky-linux/compiler-rt/6.0.1+gitAUTOINC+9d61c78bce-r0/git# git diff lib/scudo/CMakeLists.txt
diff --git a/lib/scudo/CMakeLists.txt b/lib/scudo/CMakeLists.txt
index 4d26a34..f6c1a53 100644
--- a/lib/scudo/CMakeLists.txt
+++ b/lib/scudo/CMakeLists.txt
@@ -27,9 +27,9 @@ endif()
 
 # Enable the AArch64 CRC32 feature for scudo_crc32.cpp, if available.
 # Note that it is enabled by default starting with armv8.1-a.
-if (COMPILER_RT_HAS_MCRC_FLAG)
-  set_source_files_properties(scudo_crc32.cpp PROPERTIES COMPILE_FLAGS -march=armv8.1-a)
-endif()
+#if (COMPILER_RT_HAS_MCRC_FLAG)
+#  set_source_files_properties(scudo_crc32.cpp PROPERTIES COMPILE_FLAGS -march=armv8.1-a)
+#endif()
 
```

Not 100% how to permanently solve this. Maybe add PACKAGECONFIG[scudo] with some cmake tweaks?

kraj, 2018-05-02:
This is same problem as reported in issue #52 and its due to the patch as said in comment below

https://github.com/kraj/meta-clang/issues/52#issuecomment-372063146

kraj, 2018-05-02:
Can you try the patchlet I proposed here in review comment

https://reviews.llvm.org/D35696#820019

closed 2018-05-13 by kraj