kraj/meta-clang #62: ld.lld is not deployed to the recipe-sysroot when depending on it.

Lorac, 2018-06-11:
I have a recipe that `DEPENDS` on `lld`, but all the lld targets (`lld-link ld.lld ld64.lld wasm-ld`) are never deployed to the recipe's sysroot. Then when trying to use `ld.lld`, it can't find it.

kraj, 2018-06-11:
@Lorac This seems a bug to me if you are seeing this in target lld recipe.
We do not have cross recipes for building lld yet. May be we should.

Lorac, 2018-06-11:
I forgot to mention my recipe is an `AArch64` target.

Lorac, 2018-06-11:
To make it a bit more precise I think I should've of said 

> deployed to MY recipe's sysroot.

Sorry for the confusion

kraj, 2019-01-10:
Does it still happen with latest lld work on master ?

Lorac, 2019-01-29:
With `poky` at master and `meta-oe` at master I get this weird error

```
fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).

ERROR: llvm-project-source-8.0.0-8.0.0-r0 do_fetch: Fetcher failure for URL: 'git://github.com/llvm/llvm-project;protocol=https;branch=release/8.x'. Unable to fetch URL from any source.
ERROR: llvm-project-source-8.0.0-8.0.0-r0 do_fetch: Function failed: base_do_fetch
```

I can fetch other repository fine...what could cause something like this?

kraj, 2019-01-29:
@Lorac can you try `git clone https://github.com/llvm/llvm-project -b release/8.x` on same machine and see if that works ?

Lorac, 2019-01-29:
Yes that works.

kraj, 2019-01-29:
> Yes that works.

OK, it fetches fine for me here. May be try to change `protocol=git` ?

Lorac, 2019-01-29:
It fetches from a mirror that we set up because the filename matches `git2_github.com.llvm.llvm-project.tar.gz` and it downloads it? But the file is only 6532 bytes... 

I'm wondering why the version or a sha isn't attached to those .tar.gz fileo?

> OK, it fetches fine for me here. May be try to change protocol=git ?
It seems to work with the `git` protocol.

Lorac, 2019-01-29:
But I'm getting the warning `WARNING: llvm-project-source-8.0.0-8.0.0-r0 do_fetch: Failed to fetch URL git://github.com/llvm/llvm-project;protocol=git;branch=release/8.x, attempting MIRRORS if available`

But it can fetch it from another mirror but don't know which.

kraj, 2019-01-29:
Hmm try to do `bitbake -ccleanall llvm-project-source-8.0.0`

Lorac, 2019-01-29:
I think the warning is because it still trying to fetch from our mirror and : 
This is the first attempt
```
2019-01-29 02:41:54 (307 MB/s) - /builds/downloads/git2_github.com.llvm.llvm-project.tar.gz saved [6532]
...
gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now
...
```
Then tries multiple other mirrors.

Maybe I should clear that remote mirror cache.

The last attempt that it does succeced with `LANG=C git -c core.fsyncobjectfiles=0 clone --bare --mirror https://github.com/llvm/llvm-project`

Lorac, 2019-01-29:
Ok sooo... when it tries to fetch from our mirror which isn't https we get `URL transformed to HTTPS due to an HSTS policy` from `wget`. Then gets redirected to some HTML page, download that then obviously fails to untar it.

What I don't understand is why it isn't falling back to the github url...

kraj, 2019-01-29:
You can disable PREMIRRORS for this recipe or alternatively delete the bad tar from your mirror

Lorac, 2019-01-29:
The mirror doesn't have the tar.gz in the cache it only downloads because `wget` gets redirected. 

So it works without the mirror, but I'm wondering should our mirror support https? Or the bitbake fetcher has a bug?

kraj, 2019-01-29:
may be try `BB_GENERATE_MIRROR_TARBALLS = "0"` but it could be a fetcher problem too. but somewhere this file is hosted I dont know where

Lorac, 2019-01-29:
I think it's this : https://bugzilla.yoctoproject.org/show_bug.cgi?id=9061

kraj, 2019-01-29:
yes possible I guess do you have a way to go around the fetcher issue and see if you can get to lld

Lorac, 2019-01-29:
I have a way around it, I'll do a test tonight and update if I can find the lld tool.

Lorac, 2019-01-30:
It is inside the recipe-sysroot-native so all is good!

kraj, 2019-01-30:
Can you check if ```-fuse-ld=lld``` works and uses lld ?

Lorac, 2019-01-30:
`-fuse-ld=/builds/yocto/dublin/tmp/work/aarch64-piko-linux/MYRECIPE/1.74.58+gitAUTOINC+cd2114bd5c-r0/recipe-sysroot-native/usr/bin/ld.lld -flto`

I had to use `ld.lld` otherwise I get this error message : https://github.com/llvm-mirror/lld/blob/master/tools/lld/lld.cpp#L139

kraj, 2019-01-30:
hmm that seems to be a bug to me. it seems to be failing to detect `-flavor`

kraj, 2019-02-13:
I have validated this with SDK that lld is working ok

```
% $CLANGCXX hello.cpp -fuse-ld=lld -v
clang version 8.0.0
Target: aarch64-yoe-linux
Thread model: posix
InstalledDir: /opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/x86_64-yoesdk-linux/usr/bin/aarch64-yoe-linux
Found candidate GCC installation: /opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux/usr/lib//aarch64-yoe-linux/9.0.1
Selected GCC installation: /opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux/usr/lib//aarch64-yoe-linux/9.0.1
Candidate multilib: .;@m64
Selected multilib: .;@m64
 "/opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/x86_64-yoesdk-linux/usr/bin/clang-8" -cc1 -triple aarch64-yoe-linux -emit-obj -mrelax-all -disable-free -disable-llvm-verifier -discard-value-names -main-file-name hello.cpp -mrelocation-model pic -pic-level 2 -pic-is-pie -mthread-model posix -mdisable-fp-elim -fmath-errno -masm-verbose -mconstructor-aliases -fuse-init-array -target-cpu cortex-a53 -target-feature +fp-armv8 -target-feature +crypto -target-feature +neon -target-feature +crc -target-feature +sha2 -target-feature +aes -target-abi
 aapcs -fallow-half-arguments-and-returns -dwarf-column-info -debugger-tuning=gdb -v -resource-dir /opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/x86_64-yoesdk-linux/usr/lib/clang/8.0.0 -isysroot /opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux -internal-isystem /opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux/usr/lib//aarch64-yoe-linux/9.0.1/../../../include/c++/9.0.1 -internal-isystem /opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux/usr/lib//aarch64-yoe-linux/9.0.1/../../../include/c++/9.0.1/aarch64-yoe-linux -internal-isystem /opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux/usr/lib//aarch64-yoe-linux/9.0.1/../../../include/c++/9.0.1/backward -internal-isystem /opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux/usr/local/include -internal-isystem /opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/x86_64-yoesdk-linux/usr/lib/clang/8.0.0/include -internal-externc-isystem /opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux/include -internal-externc-isystem /opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux/usr/include -fdeprecated-macro -fdebug-compilation-dir /home/kraj/tests -ferror-limit 19 -fmessage-length 272 -stack-protector 2 -fno-signed-char -fobjc-runtime=gcc -fcxx-exceptions -fexceptions -fdiagnostics-show-option -fcolor-diagnostics -o /tmp/hello-2ad893.o -x c++ hello.cpp -faddrsig
clang -cc1 version 8.0.0 based upon LLVM 8.0.0 default target x86_64-unknown-linux-gnu
ignoring nonexistent directory "/opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux/usr/local/include"
ignoring nonexistent directory "/opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux/include"
#include "..." search starts here:
#include <...> search starts here:
 /opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux/usr/lib//aarch64-yoe-linux/9.0.1/../../../include/c++/9.0.1
 /opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux/usr/lib//aarch64-yoe-linux/9.0.1/../../../include/c++/9.0.1/aarch64-yoe-linux
 /opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux/usr/lib//aarch64-yoe-linux/9.0.1/../../../include/c++/9.0.1/backward
 /opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/x86_64-yoesdk-linux/usr/lib/clang/8.0.0/include
 /opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux/usr/include
End of search list.
 "/opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/x86_64-yoesdk-linux/usr/bin/aarch64-yoe-linux/aarch64-yoe-linux-ld.lld" --sysroot=/opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux -pie -EL --eh-frame-hdr -m aarch64linux -dynamic-linker /lib/ld-linux-aarch64.so.1 -o a.out /opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux/usr/lib//aarch64-yoe-linux/9.0.1/../../Scrt1.o /opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux/usr/lib//aarch64-yoe-linux/9.0.1/../../crti.o /opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux/usr/lib//aarch64-yoe-linux/9.0.1/crtbeginS.o -L/opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux/usr/lib//aarch64-yoe-linux/9.0.1 -L/opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux/usr/lib//aarch64-yoe-linux/9.0.1/../../aarch64-yoe-linux -L/opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux//usr/lib/aarch64-yoe-linux -L/opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux/usr/lib//aarch64-yoe-linux/9.0.1/../.. -L/opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux//lib -L/opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux//usr/lib /tmp/hello-2ad893.o -lstdc++ -lm -lgcc_s -lgcc -lc -lgcc_s -lgcc /opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux/usr/lib//aarch64-yoe-linux/9.0.1/crtendS.o /opt/yoe-glibc-sysvinit-wayland/2.6/sysroots/aarch64-yoe-linux/usr/lib//aarch64-yoe-linux/9.0.1/../../crtn.o
```

closed 2019-02-13 by kraj