kraj/meta-clang #63: CMake / clang may fail with generated SDK

voltamund, 2018-07-01:
I use the yocto rocko release and wanted to build an SDK that includes the clang compilers:
MACHINE="raspberrypi3" bitbake core-image-minimal -c populate_sdk 
I only added the meta-clang layer but did not add TOOLCHAIN ?= "clang" to local.conf.

When I use CMake with the generated SDK I get errors during the configuration of the C++ compiler:
Linking fails with:  cannot find -lc++

To use the clang compiler with cmake added this to the environment-setup
    export CC=${CLANGCC}
    export CXX=${CLANGCXX}
    export CPP=${CLANGCPP}

If I add TOOLCHAIN ?= "clang" to local.conf and create the SDK again, CMake works. Is this expected?
I thought that TOOLCHAIN ?= "clang" is optional and only needed if the default compiler should be clang instead of gcc.

kraj, 2018-07-04:
@voltamund your steps are correct. it seems libc++-dev is not added to SDK somehow.  Can you try adding 
```
TOOLCHAIN_TARGET_TASK_append = " libcxx-dev" 
```
in local.conf

Please confirm if you see same issue with master/sumo, if you are on older releases you might need to backport

https://github.com/kraj/meta-clang/commit/29f52010a2591282b0b3c03bc88a7a7582d809f4
https://github.com/kraj/meta-clang/commit/82d3e0ec66bd63a348785d321366242c902ed051

voltamund, 2018-07-06:
Thank for the help, Kraj.
Using `TOOLCHAIN_TARGET_TASK_append = " libcxx-dev"` works 

I also tried this with the sumo branch (meta-clang = "sumo:707eeaeffb94fbd561eacb0fc8fb1194b531a6de"). There was another issue with this branch. The SDK installs some files into /sysroots/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/usr/lib/clang/6.0.**0**/ but the linker looks into 6.0.**1** I could fix this by changing the PATCH_VER in clang.inc from "0" to "1".  The master branch of meta-clang has the correct PATCH_VER.

For both the sumo and the master branch I need
`TOOLCHAIN_TARGET_TASK_append = " compiler-rt-staticdev libcxx-dev"`
in local.conf

Without this I get linker errors:
arm-poky-linux-gnueabi-ld: cannot find -lc++
arm-poky-linux-gnueabi-ld: cannot find libclang_rt.builtins-armhf.a

Is there no way to get this automatically into the SDK?

kraj, 2018-07-06:
@voltamund These issues are already fixed on master. I have cherry-picked some commits to sumo and pushed. Please update and re-validate with sumo and let me know how it goes.

voltamund, 2018-07-08:
I tested again with the updated sumo branch (meta-clang = "sumo:fa4592d574b039d5c4a14045902125478d0e09b3"). To me it looks like the fixes are not enough.

If I understand correctly the RDEPENDS_${PN}**_toolchain-clang** in packagegroup-core-standalone-sdk-target.bbappend should put these files into the SDK. It looks like this only works if local.conf contains TOOLCHAIN ?= "clang" or at least one recipe sets TOOLCHAIN = "clang".

I added another packagegroup-core-standalone-sdk-target.bbappend which contains:
```
RDEPENDS_${PN} += "\
    libcxx-dev \
    libcxx-staticdev \
    compiler-rt-dev \
    compiler-rt-staticdev \
"
```
RDEPENDS does not have the _toolchain-clang suffix. With this bbappend the generated SDK has the required files. There is no need to have TOOLCHAIN ?= "clang" in local.conf or build at least one recipe with clang.

mdavis777, 2018-07-09:
This pull breaks building images with libstdc++ and libunwind due to the conflict between libunwind and libcxx.  Perhaps instead of a RDEPENDS it could be included with RRECOMMENDS so that it can be listed under BAD_RECOMMENDATIONS?

kraj, 2018-07-09:
I see thanks for report I guess we could just make libcxx provide everything that libunwind provides and there by replace it with libc++ third option is to not conflict at all with gcc runtime and stop proving libunwind from libc++

Can you try these options and see if it helps

kraj, 2019-02-13:
works ok with latest meta-clang/master

I built a CMake project successfully using clang and gcc and clang+clang-runtime

https://github.com/kigster/cmake-project-template

```
CC="$CLANGCC -stdlib=libc++ -rtlib=compiler-rt" CXX="${CLANGCXX} -stdlib=libc++ -rtlib=compiler-rt" cmake ../..
make
Scanning dependencies of target division
[  6%] Building CXX object src/division/CMakeFiles/division.dir/division.cpp.o
clang-8: warning: argument unused during compilation: '-rtlib=compiler-rt' [-Wunused-command-line-argument]
[ 13%] Linking CXX static library libdivision.a
[ 13%] Built target division
Scanning dependencies of target divider
[ 20%] Building CXX object src/CMakeFiles/divider.dir/main.cpp.o
clang-8: warning: argument unused during compilation: '-rtlib=compiler-rt' [-Wunused-command-line-argument]
[ 26%] Linking CXX executable divider
[ 26%] Built target divider
Scanning dependencies of target gtest
[ 33%] Building CXX object test/lib/googletest/googlemock/gtest/CMakeFiles/gtest.dir/src/gtest-all.cc.o
clang-8: warning: argument unused during compilation: '-rtlib=compiler-rt' [-Wunused-command-line-argument]
[ 40%] Linking CXX static library ../../../../../lib/libgtest.a
[ 40%] Built target gtest
Scanning dependencies of target divider_tests
[ 46%] Building CXX object test/CMakeFiles/divider_tests.dir/main.cpp.o
clang-8: warning: argument unused during compilation: '-rtlib=compiler-rt' [-Wunused-command-line-argument]
[ 53%] Building CXX object test/CMakeFiles/divider_tests.dir/src/divider_tests.cpp.o
clang-8: warning: argument unused during compilation: '-rtlib=compiler-rt' [-Wunused-command-line-argument]
[ 60%] Linking CXX executable divider_tests
[ 60%] Built target divider_tests
Scanning dependencies of target gmock
[ 66%] Building CXX object test/lib/googletest/googlemock/CMakeFiles/gmock.dir/src/gmock-all.cc.o
clang-8: warning: argument unused during compilation: '-rtlib=compiler-rt' [-Wunused-command-line-argument]
[ 73%] Linking CXX static library ../../../../lib/libgmock.a
[ 73%] Built target gmock
Scanning dependencies of target gmock_main
[ 80%] Building CXX object test/lib/googletest/googlemock/CMakeFiles/gmock_main.dir/src/gmock_main.cc.o
clang-8: warning: argument unused during compilation: '-rtlib=compiler-rt' [-Wunused-command-line-argument]
[ 86%] Linking CXX static library ../../../../lib/libgmock_main.a
[ 86%] Built target gmock_main
Scanning dependencies of target gtest_main
[ 93%] Building CXX object test/lib/googletest/googlemock/gtest/CMakeFiles/gtest_main.dir/src/gtest_main.cc.o
clang-8: warning: argument unused during compilation: '-rtlib=compiler-rt' [-Wunused-command-line-argument]
[100%] Linking CXX static library ../../../../../lib/libgtest_main.a
[100%] Built target gtest_main
```

closed 2019-02-13 by kraj