kraj/meta-clang #64: SDK does not contain libclang_rt.builtins-i586.a but libclang_rt.builtins-i386.a

voltamund, 2018-07-01:
I tried to use clang compilers with a Yocto SDK for MACHINE qemux86 (rocko)

The build fails with:
.../sysroots/x86_64-pokysdk-linux/usr/bin/i586-poky-linux/i586-poky-linux-ld: cannot find .../sysroots/i586-poky-linux/usr/lib/clang/5.0.1/lib/linux/libclang_rt.builtins-i586.a: No such file or directory

The directory contains libclang_rt.builtins-i**386**.a There are a lot of other files in this directory that have 386 in their names. If I rename libclang_rt.builtins-i386.a to libclang_rt.builtins-i586.a, the build succeeds.
Is it possible to fix the SDK generation?

kraj, 2018-07-04:
@voltamund can you check if you see same issue with master or sumo ?

voltamund, 2018-07-08:
I do not see this issue with the sumo branch (meta-clang = "sumo:fa4592d574b039d5c4a14045902125478d0e09b3"). The SDK still contains libclang_rt.builtins-i**386**.a but the linker now also looks for this file and not for libclang_rt.builtins-i**586**.a.

kraj, 2018-07-08:
its possible that it was a bug that got fixed over time. I think its fine if you send a patch for creating a symlink for i586 case for rocko

kraj, 2019-02-13:
@voltamund can you retry with latest master which has clang8 now ?