kraj/meta-clang #66: Different default stdlib for native and nativesdk clang compiler

voltamund, 2018-07-11:
I noticed that simple C++ program does not have a dependency on libc++ if it is compiled by a bitbake recipe, it will depend on libstdc++. If the same program is built inside the SDK environment it depends on libc++. 
In clang_git.bb there is:
```
PACKAGECONFIG_class-native = ""
PACKAGECONFIG_class-nativesdk = "compiler-rt libcplusplus"
```
I think this has the effect that by default a bitbake recipe with TOOLCHAIN = "clang" uses libstdc++, however the clang compiler from the SDK by default will use libc++ (and compiler-rt).

Was there a reason for having different defaults for the native and nativesdk clang compiler?

kraj, 2018-07-11:
this is right. SDKs could produce static binaries which then does not depend upon the libraries in image. So clang runtime was added as default since its static link friendly. However, you bring up a good point here. Probably, we should not enable these packageconfigs just for nativesdk alone, may be it should be enabled for both cross compiler and SDK compiler, since we generate SDKs based upon images, its better to be consistent. 

So I am fine if we were to switch to one of other as default and document the way to switch the runtime.

kraj, 2019-01-27:
@voltamund  Can you try with latest master ? I have fixed building target packages with compiler-rt/libcxx, I would be interested to hear if this helps your case as well.

kraj, 2019-02-10:
Fixed

closed 2019-02-10 by kraj