kraj/meta-clang #72: LLVM's lld within SDK

sronsse, 2018-09-20:
Hi @kraj,

Thanks again for the development of this layer - it's been very helpful. I am reaching out as I'd like to experiment with lld along with clang. I've tried the latest master, but only clang is present when installing my SDK - is there a way to "cross-link" for my target using lld?

Thanks!

kraj, 2018-09-20:
@sronsse lld-native is now built along. So if you try it with -fuse-ld=lld it should try to use lld for linker
try it out.

sronsse, 2018-09-20:
@kraj No luck - clang cannot find the lld linker.
lld doesn't look to be built as part of my toolchain. Here is what I have after installing the SDK, I see only the gold and bfd version:
/opt/distro/0.1.0/sysroots/x86_64-sronssesdk-linux/usr/bin/x86_64-sronsse-linux/x86_64-sronsse-linux-ld.bfd
/opt/distro/0.1.0/sysroots/x86_64-sronssesdk-linux/usr/bin/x86_64-sronsse-linux/x86_64-sronsse-linux-ld.gold

kraj, 2018-09-20:
ah you are trying it in SDK. I think so far I have it in cross compile sysroot for building with yocto, for SDK we will need lld-crosssdk recipe, now that lld-native is working it is next step. I have not yet got around to that

sronsse, 2018-09-20:
Gotcha! Yep this is exactly what we need - thanks!

kraj, 2019-01-10:
closing please re-open if needed

closed 2019-01-10 by kraj