kraj/meta-clang #76: lld-native failing install step?

ericblade, 2018-12-26:
Hi! I'm attempting to add clang to a working bitbake system right now, so I simply added the meta layer, and then attempted to run bitbake clang

After chugging around for a while, I got an error in lld-native do_install:

````
Log data follows:
| DEBUG: Executing shell function do_install
| ninja: error: unknown target 'tools/lld/install', did you mean 'tools/lld/test/all'?
| WARNING: exit code 1 from a shell command.
````

Am I doing something incorrectly?

kraj, 2018-12-26:
@ericblade are you working off master ? if so, which architecture are you building for

ericblade, 2018-12-26:
Hi, thanks for the quick response.  I'm not at all familiar with this toolset, so forgive me if I don't really know what I'm doing. :-)  I've been using bitbake for years, but I pretty much just write fairly simple recipes with it.

Building for an ARM system, using master. I didn't see any releases noted in github, so I'm not sure if there's a good spot to pin it at or not.

kraj, 2018-12-26:
@ericblade I think morty definitely is a problem for lld. Latest master would work with master or at max with thud ( yocto 2.6 ) release. ARM is well supported so no issues there. Please update your core layer to use 2.6 release and that will work better.

ericblade, 2018-12-26:
way easier said than done, working with a commercial product that moves at a bit of a slow pace to adjust to most recent. :-) Im not entirely certain what all I need out of this, only that my task is to figure out how to build a piece of software in bitbake, and a big catch on it, is that it only builds with clang. :-|

ericblade, 2018-12-27:
@kraj thanks for the help, looks like it all built successfully when i switched the branch for meta-clang to the morty branch, so I'm going to close this out as "yes, I was in fact doing something wrong" because i didn't see the branches for the different releases :-)

Thanks again, happy holidays!

closed 2018-12-27 by ericblade

kraj, 2018-12-27:
@ericblade I understand, I am trying to reproduce the problem here(with master), If I can, I will see what can be done. Maybe lld should be merged into clang recipe itself, since standalone it does not add a lot of value.

kraj, 2018-12-27:
Glad it worked for you ! happy holidays to you too