kraj/meta-clang #78: No recipes available for: meta-clang/recipes-kernel/kernel-selftest/kernel-selftest.bbappend

d-wang03, 2018-12-27:
Hello Kraj, 
I'm new to yocto project. I just followed the steps in ReadMe, and when I came to the building step, an error message jumped out. I don't know how to fix it. 

wangding@wangding-VirtualBox:~/git/openembedded-core/build$ MACHINE=qemux86 bitbake core-image-minimal
Parsing recipes: 100% |########################################################################################################################################################| Time: 0:01:18
Parsing of 787 .bb files complete (0 cached, 787 parsed). 1274 targets, 66 skipped, 0 masked, 0 errors.
ERROR: No recipes available for:
  /home/wangding/git/openembedded-core/meta-clang/recipes-kernel/kernel-selftest/kernel-selftest.bbappend

Summary: There was 1 ERROR message shown, returning a non-zero exit code.

Best Regards,

kraj, 2018-12-27:
you need meta-oe layer, from meta-openembedded. Soon we will have dependency on that layer to build clang anyway.

kraj, 2018-12-28:
please test it with latest master and confirm that it is fixed

d-wang03, 2018-12-28:
It works! Thank you!

closed 2018-12-28 by d-wang03