kraj/meta-clang #103: FEATURE REQUEST: Clang-tools-extra

tomasz3dk, 2019-02-25:
Is any chance to add clang-tools-extra (or only clang-tidy) to layer?

kraj, 2019-02-25:
Yes sure that will be a good  addition  but I dont know when I can get to that, but if someone beats me tto the patch that will be awesome.

tomasz3dk, 2019-02-26:
Well. I can try. Can you provide some hints from which recipes should i start looking at?

tomasz3dk, 2019-03-05:
I've tested latest commit on master branch and it seems that clang-tools-extra are already included. Did you provide some patch or maybe changing revision do the job or maybe I missed something ;)?

kraj, 2019-03-05:
build has been overhauled recently to use monorepo, which has enabled building additional tools however, I have not tested extra tools. if it works for you then close this ticket.

tomasz3dk, 2019-03-05:
Ok. I'm closing it.

closed 2019-03-05 by tomasz3dk