kraj/meta-clang #115: Clang doesn't work when installed on target

nrclark, 2019-03-31:
I'm working on an aarch64 Yocto build, and I'd like to include Clang in the image (we have some devs that like it better, and choose to work on-target when they can)

I checked out a recent meta-clang (98c3afc5293) and built it for 'thud' from meta-openembedded (4cd3a39f22a27) and 'thud-20.0.1' from poky (1cab405d88149).

The build finishes OK, and I can install the package without issue. Installing clang on-target also pulls in libcxx and clang-libllvm.

But when I try to use it on-target, I get some errors:

```
localhost:/mnt/usb$ vim test.c
localhost:/mnt/usb$ gcc test.c
localhost:/mnt/usb$ clang test.c
/usr/bin/aarch64-poky-linux-ld: cannot find /usr/lib/clang/8.0.0/lib/linux/libclang_rt.builtins-aarch64.a: No such file or directory
/usr/bin/aarch64-poky-linux-ld: cannot find -l:libunwind.a
clang-8: error: linker command failed with exit code 1 (use -v to see invocation)
```

Got any thoughts on how I can debug this? GCC works fine on-target, but I'd love to be able to provide my devs with a working Clang as well.

kraj, 2019-04-03:
I see that we need to fix two things

1. Add `compiler-rt` to rdeps for clang
2. Add `libcxx-dev` to rdeps
3. Use `-mfloat-abi=hard` on commandline

Ideally for 3 I would like to configure clang to do that automatically, so I will look into it, I think this is due to the fact that we do not use `gnueabihf` in target tuple but `-gnueabi`

So you can meanwhile workaround this by adding

IMAGE_INSTALL += "compiler-rt libcxx-dev"

and when you compile on target then invoke clang with `-mfloat-abi=hard`

closed 2019-04-04 by kraj

nrclark, 2019-04-04:
Awesome! Thanks for looking into that. I'm excited to try it out over the weekend.

nrclark, 2019-04-17:
Just a follow-up - the new fixes work great for on-target use. I'm working on aarch64, so I didn't need to worry about the `-mfloat` flags. Thanks for looking into it!