kraj/meta-clang #123: Fail to add clang to SDK for sumo intel-core-i7 with x86-64-x32 tuning.

mabnhdev, 2019-04-19:
I'm working with Sumo branches all around.  I'm trying to add clang to my SDK for an x86_64 bit machine running an x32 user space.

I get the following error from the compiler-rt build.

|   /sbox/mabnhdev/clang-repro/build/tmp/work/x86_64_x32-poky-linux-gnux32/compiler-rt/6.0.1+gitAUTOINC+9d61c78bce-r0/recipe-sysroot/usr/include/bits/long-double.h:59:10:
|   fatal error: 'bits/long-double-64.h' file not found
| 
|   #include <bits/long-double-64.h>
| 
|            ^~~~~~~~~~~~~~~~~~~~~~~
| 
|   1 error generated.
| 
|   ninja: build stopped: subcommand failed.
| 
| Call Stack (most recent call first):
|   cmake/base-config-ix.cmake:142 (test_target_arch)
|   cmake/config-ix.cmake:138 (test_targets)
|   CMakeLists.txt:124 (include)

To reproduce on sumo or thud:

```
BASEDIR=$HOME
PROJDIR=$BASEDIR/clang-sdk
BRANCH=thud
cd $BASEDIR
git clone git://git.yoctoproject.org/poky $PROJDIR -b $BRANCH
cd $PROJDIR
git clone git://git.yoctoproject.org/meta-inel -b $BRANCH
git clone git://github.com/kraj/meta-clang.git -b $BRANCH
. oe-init-build-env
bitbake-layers add-layer $PROJDIR/meta-intel
bitbake-layers add-layer $PROJDIR/meta-clang
echo 'MACHINE = "intel-corei7-64"' >> conf/local.conf
echo 'DEFAULTTUNE = "x86-64-x32"' >> conf/local.conf
bitbake core-image-minimal -c populate_sdk
```

tomasz3dk, 2019-04-19:
I had similar issue, don't remember that it was related to meta-clang or something else but for me installing gcc-multilib and g++-multilib on my computer solved the problem

mabnhdev, 2019-04-22:
I wonder if installing gcc-multilib and g++-multilib was coincidental to solving the problem.

I'm reproducing this issue on a Ubuntu Bionic with both multilibs installed.

```
mabnhdev@mabnhdev-Ubuntu:~/clang-repro/build$ dpkg -l | grep multilib
ii  g++-7-multilib                             7.3.0-27ubuntu1~18.04                        amd64        GNU C++ compiler (multilib support)
ii  g++-multilib                               4:7.3.0-3ubuntu2.1                           amd64        GNU C++ compiler (multilib files)
ii  gcc-7-multilib                             7.3.0-27ubuntu1~18.04                        amd64        GNU C compiler (multilib support)
ii  gcc-multilib                               4:7.3.0-3ubuntu2.1                           amd64        GNU C compiler (multilib files)
mabnhdev@mabnhdev-Ubuntu:~/clang-repro/build$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.2 LTS
Release:	18.04
Codename:	bionic

...

|   /home/mabnhdev/clang-repro/build/tmp/work/x86_64_x32-poky-linux-gnux32/compiler-rt/7.0.1-r0/recipe-sysroot/usr/include/bits/long-double.h:60:10:
|   fatal error: 'bits/long-double-64.h' file not found
| 
|   #include <bits/long-double-64.h>
| 
|            ^~~~~~~~~~~~~~~~~~~~~~~
| 
|   1 error generated.
| 
|   ninja: build stopped: subcommand failed.
| 
| Call Stack (most recent call first):
|   cmake/base-config-ix.cmake:163 (test_target_arch)
|   cmake/config-ix.cmake:140 (test_targets)
|   CMakeLists.txt:177 (include)
| 
| 
| -- Configuring incomplete, errors occurred!

```

mabnhdev, 2019-04-26:
In the interest of moving forward, I added a hack to define stubs for the missing header files that was tripping up the compiler_rt configuration.

I was able to get a little farther, but ran into a bunch of linker problems - for example.

  /home/mabnhdev/clang-issue-123-repro/build/tmp/work/x86_64_x32-poky-linux-gnux32/compiler-rt/7.0.1-r0/recipe-sysroot-native/usr/bin/x86_64-poky-linux-gnux32/x86_64-poky-linux-gnux32-ld:
  skipping incompatible
  /home/mabnhdev/clang-issue-123-repro/build/tmp/work/x86_64_x32-poky-linux-gnux32/compiler-rt/7.0.1-r0/recipe-sysroot/lib/libgcc_s.so.1
  when searching for libgcc_s.so.1

I've enclosed a zip file containing the full log file and a bash script to apply my workaround and reproduce this latest failure.

Regards, Mike

[clang-issue-123.zip](https://github.com/kraj/meta-clang/files/3121405/clang-issue-123.zip)

kraj, 2019-05-20:
@mabnhdev can you reproduce this on master too ?

mabnhdev, 2019-05-29:
I tried running my test against master and hit an error pretty quickly.  Looks like Clang is not ready for master.

ERROR: Layer clang-hack is not compatible with the core layer which only supports these series: warrior (layer is compatible with thud sumo)

kraj, 2019-06-13:
hmm are you trying master older than b6f8b068 ?

mabnhdev, 2019-06-24:
Same linker failures using the latest master for all repos.

I've enclosed the log files and a reproduction shell script that uses the master branch.

Regards, Mike

[clang-issue-123-master.zip](https://github.com/kraj/meta-clang/files/3321191/clang-issue-123-master.zip)

kraj, 2019-06-24:
in compiler-rt recipe can you try adding

```bitbake
do_configure_prepend_linux-gnux32() {
        cp ${STAGING_INCDIR}/gnu/stubs-x32.h ${STAGING_INCDIR}/gnu/stubs-64.h
        cp ${STAGING_INCDIR}/bits/long-double-32.h ${STAGING_INCDIR}/bits/long-double-64.h
}
```

mabnhdev, 2019-06-25:
I also needed floatn-64.h which I added to the do_configure_prepend_linux-gnux32() you suggested.

I'm now getting configure errors for compiler-rt.

I've enclosed the error log and repoduction script updated with your most recent suggestion.

Regards, Mike
[clang-issue-123-20190625.zip](https://github.com/kraj/meta-clang/files/3325718/clang-issue-123-20190625.zip)

mabnhdev, 2019-08-16:
Master is not quite ready for prime time yet?

Parsing recipes: 100% |##############################################################################################| Time: 0:00:38
Parsing of 838 .bb files complete (0 cached, 838 parsed). 1374 targets, 70 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.43.1"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "x86_64-poky-linux-gnux32"
MACHINE              = "intel-corei7-64"
DISTRO               = "poky"
DISTRO_VERSION       = "2.7+snapshot-20190816"
TUNE_FEATURES        = "mx32"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "master:20946c63c272b2128e6dc76490313b174ee97d8e"
meta-intel           = "master:14415d0d3ce280ce20eafb40cb3008a05f8e2003"
meta-clang           = "master:4e05dac9af2b9f000e31f799511899870663b482"
meta-hack            = "master:20946c63c272b2128e6dc76490313b174ee97d8e"

NOTE: Fetching uninative binary shim from http://downloads.yoctoproject.org/releases/uninative/2.6/x86_64-nativesdk-libc.tar.xz;sha256sum=133387753a9acf3e1b788103c59fac91e968e2ee331d7a4b9498e926ada7be57
Initialising tasks: 100% |###########################################################################################| Time: 0:00:02
Sstate summary: Wanted 884 Found 0 Missed 884 Current 0 (0% match, 0% complete)
NOTE: Executing Tasks
NOTE: Setscene tasks completed
ERROR: llvm-project-source-9.0.0-9.0.0-r0 do_patch: Command Error: 'quilt --quiltrc /home/mabnhdev/clang-issue-123-repro/build/tmp/work-shared/llvm-project-source-9.0.0-r0/recipe-sysroot-native/etc/quiltrc push' exited with 0  Output:
Applying patch 0001-OpenCL-Change-type-of-block-pointer-for-OpenCL.patch
patching file lib/CodeGen/CodeGenTypes.cpp
Hunk #1 FAILED at 637.
1 out of 1 hunk FAILED -- rejects in file lib/CodeGen/CodeGenTypes.cpp
patching file test/CodeGenOpenCL/blocks.cl
Hunk #1 FAILED at 35.
Hunk #2 FAILED at 50.
2 out of 2 hunks FAILED -- rejects in file test/CodeGenOpenCL/blocks.cl
patching file test/CodeGenOpenCL/cl20-device-side-enqueue.cl
Hunk #1 FAILED at 11.
Hunk #2 FAILED at 77.
Hunk #3 FAILED at 95.
Hunk #4 FAILED at 300.
Hunk #5 FAILED at 346.
5 out of 5 hunks FAILED -- rejects in file test/CodeGenOpenCL/cl20-device-side-enqueue.cl
Patch 0001-OpenCL-Change-type-of-block-pointer-for-OpenCL.patch does not apply (enforce with -f)
ERROR: Logfile of failure stored in: /home/mabnhdev/clang-issue-123-repro/build/tmp/work-shared/llvm-project-source-9.0.0-r0/temp/log.do_patch.12649
ERROR: Task (/home/mabnhdev/clang-issue-123-repro/meta-clang/recipes-devtools/clang/llvm-project-source.bb:do_patch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 924 tasks of which 0 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/mabnhdev/clang-issue-123-repro/meta-clang/recipes-devtools/clang/llvm-project-source.bb:do_patch
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

kraj, 2019-08-16:
`0001-OpenCL-Change-type-of-block-pointer-for-OpenCL.patch` is not a patch from meta-clang where is it coming from ?

mabnhdev, 2019-08-19:
It comes from Yocto's meta-intel layer - https://git.yoctoproject.org/cgit/cgit.cgi/meta-intel/tree/dynamic-layers/clang-layer/recipes-devtools/clang/files/0001-OpenCL-Change-type-of-block-pointer-for-OpenCL.patch

kraj, 2019-08-19:
I do not maintain that layer so please report or to yocto mailing lists and meta-intel maintainers

mabnhdev, 2019-08-20:
FYI - from the meta-intel mailing list...

> Those patches are backports and specific to clang8 and aren't required
> for 9.0. Unfortunately, the components in the layer aren't yet
> completely ready to be built with llvm/clang9. I'll update these as
> soon as they are.
> 
> If the aim is just to add support for x32 in clang, you could perhaps
> just remove this bbappend altogether? This would only impact the OpenCL
> components.
> 
> Thanks,
> 
> Anuj