kraj/meta-clang #127: Building with multilib fails

saiarcot895, 2019-06-01:
On Yocto Thud 2.6.0, with the meta-clang layer on the thud branch, if multilib is enabled, then the `compiler-rt` packaging step fails, since the files are installed to `/usr/lib`, but `do_install` is searching in `/usr/lib64`. I was able to get past this by having my own `do_install_append` that is hardcoded to search in `/usr/lib`, but the `libcxx` hits a similar issue.

What's the expected behavior when multilib is enabled? Should clang libraries be installed into `/usr/lib64` or `/usr/lib`?

`conf/local.conf` customizations:
```
TOOLCHAIN ?= "clang"

require conf/multilib.conf
MULTILIBS = "multilib:lib32"
DEFAULTTUNE_virtclass-multilib-lib32 = "core2-32"
DEFAULTTUNE = "core2-64"
```

Error log:
```
ERROR: compiler-rt-7.0.1-r0 do_package: QA Issue: compiler-rt: Files/directories were installed but not shipped in any package:
  /usr/lib/linux/libclang_rt.ubsan_minimal-x86_64.so
  /usr/lib/linux/libclang_rt.asan_cxx-x86_64.a
  /usr/lib/linux/libclang_rt.safestack-x86_64.a
  /usr/lib/linux/libclang_rt.ubsan_standalone-x86_64.so
  /usr/lib/linux/libclang_rt.ubsan_minimal-x86_64.a
  /usr/lib/linux/libclang_rt.scudo_minimal-x86_64.a
  /usr/lib/linux/libclang_rt.hwasan_cxx-x86_64.a
  /usr/lib/linux/libclang_rt.hwasan-x86_64.a.syms
  /usr/lib/linux/libclang_rt.hwasan_cxx-x86_64.a.syms
  /usr/lib/linux/libclang_rt.dfsan-x86_64.a
  /usr/lib/linux/libclang_rt.scudo-x86_64.a
  /usr/lib/linux/libclang_rt.asan-x86_64.a
  /usr/lib/linux/libclang_rt.fuzzer_no_main-x86_64.a
  /usr/lib/linux/libclang_rt.esan-x86_64.a
  /usr/lib/linux/libclang_rt.ubsan_minimal-x86_64.a.syms
  /usr/lib/linux/libclang_rt.msan_cxx-x86_64.a.syms
  /usr/lib/linux/libclang_rt.scudo_minimal-x86_64.so
  /usr/lib/linux/libclang_rt.stats-x86_64.a
  /usr/lib/linux/libclang_rt.ubsan_standalone_cxx-x86_64.a.syms
  /usr/lib/linux/libclang_rt.ubsan_standalone-x86_64.a
  /usr/lib/linux/libclang_rt.hwasan-x86_64.so
  /usr/lib/linux/libclang_rt.scudo_cxx-x86_64.a
  /usr/lib/linux/libclang_rt.builtins-x86_64.a
  /usr/lib/linux/libclang_rt.scudo-x86_64.so
  /usr/lib/linux/libclang_rt.msan_cxx-x86_64.a
  /usr/lib/linux/libclang_rt.lsan-x86_64.a
  /usr/lib/linux/libclang_rt.asan-preinit-x86_64.a
  /usr/lib/linux/libclang_rt.cfi-x86_64.a
  /usr/lib/linux/libclang_rt.dfsan-x86_64.a.syms
  /usr/lib/linux/libclang_rt.asan_cxx-x86_64.a.syms
  /usr/lib/linux/libclang_rt.cfi_diag-x86_64.a
  /usr/lib/linux/libclang_rt.stats_client-x86_64.a
  /usr/lib/linux/libclang_rt.msan-x86_64.a.syms
  /usr/lib/linux/libclang_rt.ubsan_standalone_cxx-x86_64.a
  /usr/lib/linux/libclang_rt.profile-x86_64.a
  /usr/lib/linux/libclang_rt.scudo_cxx_minimal-x86_64.a
  /usr/lib/linux/libclang_rt.ubsan_standalone-x86_64.a.syms
  /usr/lib/linux/libclang_rt.hwasan-x86_64.a
  /usr/lib/linux/libclang_rt.fuzzer-x86_64.a
  /usr/lib/linux/libclang_rt.esan-x86_64.a.syms
  /usr/lib/linux/libclang_rt.asan-x86_64.so
  /usr/lib/linux/libclang_rt.asan-x86_64.a.syms
  /usr/lib/linux/libclang_rt.msan-x86_64.a
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
compiler-rt: 43 installed and not shipped files. [installed-vs-shipped]
ERROR: compiler-rt-7.0.1-r0 do_package: Fatal QA errors found, failing task.
```

kraj, 2019-06-13:
@saiarcot895 Can you try moving them to `/usr/lib64` in multilib env

RichardSun1116, 2019-06-14:
@ kraj 
I encountered the same issue and fixed as below patch:
### 
In compiler-rt_git.bb:
do_install_append_aarch64 () {
	mv ${D}/usr/lib ${D}/usr/lib64
}
### 

by the way , I also encountered the compiler-rt-8.0.1 build issue for multilibs
In llvm-ar.cpp [1] line 1133 ~ 1137 , it judges the keyword "lib" before "ar",
In multilib case , it will cause the fail because the compiler prefix is including aarch64-xxxlib64 or arm-xxxlib32.
the solution may patch llvm-ar.cpp and judge llvm-ar keyword before line 1133,
or creating a wrapper to avoid this case.
i.e 
### 
clang-cross_git.bb
do_install_append() {

	rm -f ${D}${bindir}/${TARGET_PREFIX}llvm-ar
	cat > ${D}${bindir}/${TARGET_PREFIX}llvm-ar << 'EOF'
#!/bin/sh
basedir=$(dirname "$0")
"${basedir}/../llvm-ar" "$@"
exit "$?"
EOF
	chmod +x ${D}${bindir}/${TARGET_PREFIX}llvm-ar

}
### 

[1] : https://github.com/llvm-mirror/llvm/blob/master/tools/llvm-ar/llvm-ar.cpp

Thanks.
Richard

jyong2, 2019-09-12:
Hi @kraj,

I am also seeing this issue when dealing with opencl-clang from meta-intel with multilib enabled. Fixes all seem pretty messy, llvm-config still does not work.

What I saw gentoo did might be a cleaner approach, they had this their ebuild:
                -DCMAKE_INSTALL_PREFIX="${EPREFIX}/usr/lib/llvm/${SLOT%/*}"

And then just update PATH to the appropriate versions. Presumably clang/llvm can find its own libraries. I have yet to actually test this out.

jyong2, 2019-09-15:
I found another alternative by using " -DLLVM_LIBDIR_SUFFIX=${LIBSUFFIX} -DCLANG_LIBDIR_SUFFIX=${LIBSUFFIX} -DLIBCXX_LIBDIR_SUFFIX=${LIBSUFFIX}" in EXTRA_OECMAKE.

The suffix is computed with:
def get_libdir_suffix(d):
    import re
    libdir = d.getVar("libdir")
    prefix = d.getVar("prefix")
    return re.sub("^" + prefix + "/lib", "", libdir)

Unfortunately llvm-config --ldflags is still broken, so you may need something like:
do_configure_prepend(){
        if [ -n "${LIBSUFFIX}" -a ! -d "${STAGING_DIR_HOST}/${prefix}/lib" ]; then
                ln -s "${STAGING_LIBDIR}" "${STAGING_DIR_HOST}/${prefix}/lib"
        fi
}

kraj, 2019-09-15:
llvm-config could be patched for this logic see `0007-llvm-allow-env-override-of-exe-path.patch`

jpuhlman, 2019-09-24:
So the patch to use llvm-ar in thud, tripped over this. compiler-rt was building before, but now it stopped complaining that  qc does not exist. When running the util it prints out nothing. Our multilib is labeled as mllib32.

kraj, 2019-09-25:
@jpuhlman on thud we really dont need llvm-ar to be default AR if its causing problems lets remove it. compiler-ar was really tested in warrior so we can leave thud alone

JeffyCN, 2019-11-22:
Hi @kraj 

I hit the same issue(https://github.com/JeffyCN/meta-rockchip/issues/3) as @RichardSun1116  described above:

In multilib(lib32) case, the arm-pokymllib32-linux-gnueabi-llvm-ar (${TARGET_PREFIX}llvm-ar) gives:
qc: no such file or directory

Which is because when the llvm-ar symbol link's name contains "lib", it would be considered as llvm-lib:
https://github.com/llvm-mirror/llvm/blob/master/tools/llvm-ar/llvm-ar.cpp#L1178

/usr/bin# ln -s llvm-ar lib-ar
/usr/bin# lib-ar qc ls
qc: no such file or directory

/usr/bin# llvm-lib qc ls
qc: no such file or directory

kraj, 2019-12-04:
https://github.com/kraj/meta-clang/pull/237 should fix this.

closed 2019-12-05 by kraj