kraj/meta-clang #128: clang sdk build fails for raspberrypi2 - /tmp/hosttools/ld: cannot find -lstdc++

shrkamat, 2019-06-17:
```
Build Configuration:
BB_VERSION           = "1.43.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-16.04"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi2"
DISTRO               = "poky"
DISTRO_VERSION       = "2.7+snapshot-20190616"
TUNE_FEATURES        = "arm vfp cortexa7 neon vfpv4 thumb callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       = "HEAD:27d60c5a812774f0e5c43161e5b514c4aebdf301"
meta-raspberrypi     = "HEAD:40283f583b57d67798135ed4a94cb8b5288ea965"
meta-oe              
meta-networking      
meta-python          = "HEAD:866daa14e30e2141a025fd9e71ba5406a8b0329a"
meta-clang           = "HEAD:ba3e396e7b20e7d56f08f3540e780ec1ad51ecfd"
```

**Steps to recreate problem** 

> git clone https://github.com/shrkamat/yocto
> cd yocto
> git submodule update
> source poky/oe-init-build-env pi2
> bitbake meta-toolchain -c populate_sdk

Build logs are shared here -> https://github.com/shrkamat/yocto/blob/master/pi2/nohup.out

kraj, 2019-06-17:
Can you check if libstdc++ is installed on your build machine ? If not then you should install it

shrkamat, 2019-06-17:
Yes, it's available.   

```
gcc --version
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

> file ./gcc/x86_64-linux-gnu/5/libstdc++.so
./gcc/x86_64-linux-gnu/5/libstdc++.so: symbolic link to ../../../x86_64-linux-gnu/libstdc++.so.6
> readlink -f ./gcc/x86_64-linux-gnu/5/libstdc++.so
/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21
> file /usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21
/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.21: ELF 64-bit LSB shared object, x86-64, version 1 (GNU/Linux), dynamically linked, BuildID[sha1]=c5ca582a8e1ee50f59caaaf84a1464c8cb3c2f3d, stripped
```

shrkamat, 2019-06-19:
This seems to me link CMAKE cache related issue. If I manually delete the cmake generated files and run cmake manually, I am able to proceed with the build

```
sudo apt-get install ninja-build
cd /home/shrkamat/yocto/pi2/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/8.0.1-r0/build/NATIVE/NATIVE
rm -rfv *
/home/shrkamat/yocto/pi2/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/8.0.1-r0/recipe-sysroot-native/usr/bin/cmake -G Ninja -DCMAKE_MAKE_PROGRAM="ninja" -DCMAKE_C_COMPILER=/home/shrkamat/yocto/pi2/tmp/hosttools/gcc -DCMAKE_CXX_COMPILER=/home/shrkamat/yocto/pi2/tmp/hosttools/g++ /home/shrkamat/yocto/pi2/tmp/work-shared/llvm-project-source-8.0.1-r0/git/llvm -DLLVM_TARGET_IS_CROSSCOMPILE_HOST=TRUE -DLLVM_TARGETS_TO_BUILD="AArch64;ARM;BPF;Mips;PowerPC;X86;RISCV" -DLLVM_EXPERIMENTAL_TARGETS_TO_BUILD="RISCV;" -DLLVM_DEFAULT_TARGET_TRIPLE="x86_64-unknown-linux-gnu" -DLLVM_TARGET_ARCH="host" -DLLVM_TEMPORARILY_ALLOW_OLD_TOOLCHAIN="ON" -DCMAKE_BUILD_TYPE=Release -DLLVM_EXTERNAL_CLANG_SOURCE_DIR=/home/shrkamat/yocto/pi2/tmp/work-shared/llvm-project-source-8.0.1-r0/git/llvm/../clang
```

Quick question, can I limit LLVM_TARGETS_TO_BUILD="ARM" and retrigger build without performing cleansstate ?

kraj, 2019-06-19:
Yes you can

shrkamat, 2019-06-24:
I see in the yocto gererated SDK toolchain is configured as gcc in OE_CMAKE_TOOLCHAIN_FILE. I this expected? I have configured TOOLCHAIN="clang".

I see there are a couple of exports related to clang, 

```
CLANGCC=arm-poky-linux-gnueabi-clang  -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -mcpu=cortex-a7 -mlittle-endian --sysroot=/opt/poky/2.7+snapshot/sysroots/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi
CLANGCPP=arm-poky-linux-gnueabi-clang -E  -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -mcpu=cortex-a7 -mlittle-endian --sysroot=/opt/poky/2.7+snapshot/sysroots/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi
CLANGCXX=arm-poky-linux-gnueabi-clang++  -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -mcpu=cortex-a7 -mlittle-endian --sysroot=/opt/poky/2.7+snapshot/sysroots/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi
CLANG_TIDY_EXE=arm-poky-linux-gnueabi-clang-tidy  -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -mcpu=cortex-a7 -mlittle-endian --sysroot=/opt/poky/2.7+snapshot/sysroots/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi
```

Now I could probably try overriding the cmake vaiables, but I was wondering  if there is a TOOLCHAIN cmake file with clag configured as default compiler.

kraj, 2019-06-24:
@shrkamat yes this is expected. default compiler is still gcc in SDK always regardless of TOOLCHAIN variable, TOOLCHAIN var is primarily for internal build. May be time now is to change it based on TOOLCHAIN settings.However If you try setting

CC=${CLANGCC}
CPP=${CLANGCPP}
CXX=${CLANGCXX}

in your environment before running cmake then it might work. But I guess if you cook up a patch either for

1. Use CC etc as clang when TOOLCHAIN = "clang" and gcc when TOOLCHAIN = "gcc"
2. Generate a clang defaulting Toolchain file for CMake.

might be interesting.

madscientist42, 2019-08-15:
Known to work if you do the first suggestion.  Copying CLANGCC, etc. over works **_flawlessly_** with anything where autotools or CMake is your build system.  The needed minimal magic to make it use the SDK sandbox is in the macros and copying them over to CC, etc. does the trick (In a production build system in place for the last 6+ months...)

madscientist42, 2019-08-15:
The other's intriguing, but we kind-of, sort-of needed both toolchains so we left that alone.  Might be worth a try to see if it ends up working out more simply so we can document the first and maybe the second so people know how to make this work.

kraj, 2019-08-16:
this would be awesome. Can you make amends to README documenting all this and send a patch ?

madscientist42, 2019-08-16:
Yes.

madscientist42, 2019-08-29:
Got bandwidth now to actually go through and make the requested changes to README.  Do you want it in just master or do you want it dropped in what other branches?

kraj, 2019-08-29:
this kind of change can go to all branches where its applicable

madscientist42, 2019-08-29:
Will do.  Will update accordingly and should have a PR in a day or several as I get it finished up.

carlocaione, 2019-09-13:
Is there a proper solution for the `...hosttools/ld: cannot find -lstdc++ ` problem? It's currently biting me on one of the build servers I'm using.

kraj, 2019-09-14:
can you describe your build set up and steps to reproduce it ?

kraj, 2019-09-15:
@carlocaione @shrkamat  we have disabled stage2 builds for clang which should fix this issue. Please try with latest master and see if it helps your case.

carlocaione, 2019-09-18:
@kraj it solves the compilation problem but now chromium-ozone is segfaulting at boot. Not sure if related to this change yet.

kraj, 2019-09-18:
@carlocaione thats because of llvm libunwind being in use. Perhaps a workaround by trying to set
```
COMPIILER_RT = ""
LIBCPLUSPLUS = ""
```
in chromium recipe will help

sohamm17, 2020-11-17:
I mistakenly called `bitbake <image name> -c populate_sdk` and now I am getting the same above error:
```
tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-clang/9.0.1-r0/recipe-sysroot-native/usr/bin/clang++ -- broken
...
build/tmp/hosttools/ld.gold: error: cannot find -lstdc++
|     clang-9: error: linker command failed with exit code 1 (use -v to see invocation)
|     ninja: build stopped: subcommand failed.
```

Is there any way to revert `populate_sdk` or other ways to get it right? I have intsalled g++-multilib on my machine, so do not know what should be causing this issue.

kraj, 2020-11-17:
@sohamm17 does it work if you remove g++-multilib from your build host ?

sohamm17, 2020-11-17:
No, it does not. From the above discussion, I thought that multilib would fix the issue.

kraj, 2020-11-17:
@sohamm17 which branch are you using ?
does it have 2bbb0477e5578591daeba0d1bb9dcd500b31fdf3

sohamm17, 2020-11-17:
I am using the zeus branch. It the clang_git.bb has the following line, not the one that is mentioned in that commit:
```
BINPATHPREFIX = "${@bb.utils.contains('PACKAGECONFIG', 'bootstrap', '/tools/clang/stage2-bins/NATIVE', '', d)}"
```

sohamm17, 2020-11-17:
> I am using the zeus branch. It the clang_git.bb has the following line, not the one that is mentioned in that commit:
> 
> ```
> BINPATHPREFIX = "${@bb.utils.contains('PACKAGECONFIG', 'bootstrap', '/tools/clang/stage2-bins/NATIVE', '', d)}"
> ```

@kraj do you think I need to change the above lines with:
```
BINPATHPREFIX ?= ""
BINPATHPREFIX_class-native = "/tools/clang/stage2-bins/NATIVE"
```
As in your mentioned commit?

kraj, 2020-11-17:
@sohamm17 perhaps try yes. I dont have zeus build readily available here so cant see more. Usually right now we only support master/dunfell

sohamm17, 2020-11-18:
This breaks even the clang-native builds. Now, nothing gets compiled.

kraj, 2021-03-25:
zeus is EOL, but if you still see this issue with dunfell and newer please open a new issue.

closed 2021-03-25 by kraj

rathanil, 2024-11-18:
@sohamm17 

you can install below packages on host system 
 sudo apt-get install clang cmake ninja-build pkg-config libgtk-3-dev liblzma-dev libstdc++-12-dev