kraj/meta-clang #134: clang (warrior) sdk build fails at ubuntu 16.04 and 18.04

danieldittmann, 2019-07-30:
When building (warrior) nativesdk-clang at ubuntu 16.04 or 18.04:
gcc: error: unrecognized command line option ‘-fmacro-prefix-map=

Ubuntus host compiler is gcc 5.4.0 (at 16.04) or gcc 7.4.0 (at 18.04).
Both compiler does not support "-fmacro-prefix-map=" option.

kraj, 2019-07-30:
In `clang_git.bb` Can you try setting

`TOOLCHAIN_class-nativesdk = "clang"

that will try to use clang-native to build nativesdk recipe.
`

danieldittmann, 2019-07-31:
`TOOLCHAIN_class-nativesdk = "clang"` is already set.
Setting `TOOLCHAIN_class-native = "clang"` does also not work.
When I set
`DEBUG_PREFIX_MAP_class-nativesdk = "-fdebug-prefix-map=${WORKDIR}=/usr/src/debug/${PN/${EXTENDPE}${PV}-${PR} \`
`                     -fdebug-prefix-map=${STAGING_DIR_HOST}= \`
`                     -fdebug-prefix-map=${STAGING_DIR_NATIVE}= \`
`"`
in clang_git.bb than it works.

kraj, 2019-07-31:
I think this is fine as a fix. Can you prepare a patch for this ?

danieldittmann, 2019-08-01:
Further analysis shows, that some cmake based steps when compiling clang uses the gcc instead of clang (even if TOOLCHAIN is set to clang). When compiling the nativesdk no gcc is in native toolchain sysroot, so that the host gcc is used.

With this patch native-clang is used to build the nativesdk-clang.
#135

closed 2019-08-05 by danieldittmann