kraj/meta-clang #136: Failure build sdk with clang tools included

reid-p, 2019-08-05:
I'm getting the following linker error while trying to build an arm sdk with the clang tools included. Any ideas?

```
| projects/libcxxabi/src/CMakeFiles/cxxabi_objects.dir/cxa_guard.cpp.o(.ARM.exidx+0x0): error: undefined reference to '__aeabi_unwind_cpp_pr1'
| projects/libcxxabi/src/CMakeFiles/cxxabi_objects.dir/cxa_guard.cpp.o(.ARM.exidx+0x8): error: undefined reference to '__aeabi_unwind_cpp_pr1'
| projects/libcxxabi/src/CMakeFiles/cxxabi_objects.dir/cxa_guard.cpp.o(.ARM.exidx+0x10): error: undefined reference to '__aeabi_unwind_cpp_pr1'
| /builddir/v2019.07/tmp-glibc/work-shared/llvm-project-source-8.0.1-r0/git/libcxxabi/src/cxa_vector.cpp:0: error: undefined reference to '_Unwind_Resume'
| /builddir/v2019.07/tmp-glibc/work-shared/llvm-project-source-8.0.1-r0/git/libcxxabi/src/cxa_vector.cpp:0: error: undefined reference to '_Unwind_Resume'
| /builddir/v2019.07/tmp-glibc/work-shared/llvm-project-source-8.0.1-r0/git/libcxxabi/src/cxa_vector.cpp:0: error: undefined reference to '_Unwind_Resume'
| /builddir/v2019.07/tmp-glibc/work-shared/llvm-project-source-8.0.1-r0/git/libcxxabi/src/cxa_vector.cpp:0: error: undefined reference to '_Unwind_Resume'
| projects/libcxxabi/src/CMakeFiles/cxxabi_objects.dir/stdlib_stdexcept.cpp.o(.ARM.exidx+0x0): error: undefined reference to '__aeabi_unwind_cpp_pr1'
| /usr/lib/clang/8.0.1/include/unwind.h:235: error: undefined reference to '_Unwind_VRS_Set'
| /usr/lib/clang/8.0.1/include/unwind.h:228: error: undefined reference to '_Unwind_VRS_Get'
| /usr/lib/clang/8.0.1/include/unwind.h:228: error: undefined reference to '_Unwind_VRS_Get'
| /builddir/v2019.07/tmp-glibc/work-shared/llvm-project-source-8.0.1-r0/git/libcxxabi/src/cxa_personality.cpp:0: error: undefined reference to '__gnu_unwind_frame'
| /builddir/v2019.07/tmp-glibc/work-shared/llvm-project-source-8.0.1-r0/git/libcxxabi/src/cxa_personality.cpp:0: error: undefined reference to '_Unwind_VRS_Set'
| /builddir/v2019.07/tmp-glibc/work-shared/llvm-project-source-8.0.1-r0/git/libcxxabi/src/cxa_personality.cpp:0: error: undefined reference to '_Unwind_VRS_Set'
| /builddir/v2019.07/tmp-glibc/work-shared/llvm-project-source-8.0.1-r0/git/libcxxabi/src/cxa_personality.cpp:0: error: undefined reference to '_Unwind_VRS_Get'
| /builddir/v2019.07/tmp-glibc/work-shared/llvm-project-source-8.0.1-r0/git/libcxxabi/src/cxa_personality.cpp:0: error: undefined reference to '_Unwind_VRS_Set'
| /builddir/v2019.07/tmp-glibc/work-shared/llvm-project-source-8.0.1-r0/git/libcxxabi/src/cxa_personality.cpp:606: error: undefined reference to '_Unwind_GetLanguageSpecificData'
| /usr/lib/clang/8.0.1/include/unwind.h:228: error: undefined reference to '_Unwind_VRS_Get'
| /builddir/v2019.07/tmp-glibc/work-shared/llvm-project-source-8.0.1-r0/git/libcxxabi/src/cxa_personality.cpp:619: error: undefined reference to '_Unwind_GetRegionStart'
| projects/libcxxabi/src/CMakeFiles/cxxabi_objects.dir/cxa_aux_runtime.cpp.o(.ARM.exidx+0x0): error: undefined reference to '__aeabi_unwind_cpp_pr0'
| projects/libcxxabi/src/CMakeFiles/cxxabi_objects.dir/cxa_default_handlers.cpp.o(.ARM.exidx+0x0): error: undefined reference to '__aeabi_unwind_cpp_pr0'
| projects/libcxxabi/src/CMakeFiles/cxxabi_objects.dir/cxa_demangle.cpp.o(.ARM.exidx+0x98): error: undefined reference to '__aeabi_unwind_cpp_pr0'
| projects/libcxxabi/src/CMakeFiles/cxxabi_objects.dir/cxa_demangle.cpp.o(.ARM.exidx+0x1d0): error: undefined reference to '__aeabi_unwind_cpp_pr0'
| /builddir/v2019.07/tmp-glibc/work-shared/llvm-project-source-8.0.1-r0/git/libcxxabi/src/cxa_exception.cpp:281: error: undefined reference to '_Unwind_RaiseException'
| /builddir/v2019.07/tmp-glibc/work-shared/llvm-project-source-8.0.1-r0/git/libcxxabi/src/cxa_exception.cpp:551: error: undefined reference to '_Unwind_DeleteException'
| /builddir/v2019.07/tmp-glibc/work-shared/llvm-project-source-8.0.1-r0/git/libcxxabi/src/cxa_exception.cpp:609: error: undefined reference to '_Unwind_RaiseException'
| /builddir/v2019.07/tmp-glibc/work-shared/llvm-project-source-8.0.1-r0/git/libcxxabi/src/cxa_exception.cpp:734: error: undefined reference to '_Unwind_RaiseException'
| clang-8: error: linker command failed with exit code 1 (use -v to see invocation)
| ninja: build stopped: subcommand failed.
| WARNING: /builddir/v2019.07/tmp-glibc/work/cortexa9t2hf-neon-emit-linux-gnueabi/libcxx/8.0.1-r0/temp/run.do_compile.69709:1 exit 1 from 'ninja -v -j 12 cxxabi'
```

kraj, 2019-08-06:
I think we need to add -lgcc or -lunwind when building libcxx

kraj, 2019-08-13:
recipes are upgraded to use 9.x RC, please try with the latest master and report back. Also, post your build configuration so we can reproduce your case.

reid-p, 2019-08-13:
Just did a quick try and failing in the configure stage.
`
-- Found PythonLibs: /build/tmp-glibc/work/x86_64-linux/clang-native/9.0.0-r0/recipe-sysroot-native/usr/lib/libpython3.7m.so (found version "3.7.2") 
CMake Error at /build/tmp-glibc/work-shared/llvm-project-source-9.0.0-r0/git/lldb/cmake/modules/LLDBConfig.cmake:204 (message):
  Found incompatible Python interpreter (3.6.8) and Python libraries (3.7.2)
Call Stack (most recent call first):
  /build/tmp-glibc/work-shared/llvm-project-source-9.0.0-r0/git/lldb/CMakeLists.txt:20 (include)
`

3.6.8 corresponds to my native install. 3.7.2 looks to be the version built by bitbake.
I'll 

What files do you need to replicate my build config?

reid-p, 2019-08-13:
I add python3native to clang_git.bb

ie:
inherit cmake cmake-native
changed to
inherit cmake cmake-native python3native

and have got past the configure step.

Waiting for the do_compile

kraj, 2019-08-13:
@reid-p thanks, can you create a pull?

kraj, 2019-08-15:
should be fixed with 1746d053273024037e3ef227130a6582f191993e

closed 2019-08-15 by kraj

reid-p, 2019-08-16:
@kraj 
Took me a while to get a uild going, had to get a new host tool chain installed as it was failing in stage2 due to a warning about the age of the gcc compiler. Which looks to have been recently addressed.

Unfortunately still getting the failure.

`02:47:08 | /home/cleanbuild/build/workspace/soc-image-v2019.07/tmp-glibc/work-shared/llvm-project-source-9.0.0-r0/git/libcxxabi/src/cxa_guard_impl.h:0: error: undefined reference to '_Unwind_Resume'
02:47:08 | /home/cleanbuild/build/workspace/soc-image-v2019.07/tmp-glibc/work-shared/llvm-project-source-9.0.0-r0/git/libcxxabi/src/cxa_vector.cpp:57: error: undefined reference to '_Unwind_Resume'
02:47:08 | /home/cleanbuild/build/workspace/soc-image-v2019.07/tmp-glibc/work-shared/llvm-project-source-9.0.0-r0/git/libcxxabi/src/cxa_vector.cpp:0: error: undefined reference to '_Unwind_Resume'
02:47:08 | /home/cleanbuild/build/workspace/soc-image-v2019.07/tmp-glibc/work-shared/llvm-project-source-9.0.0-r0/git/libcxxabi/src/cxa_vector.cpp:0: error: undefined reference to '_Unwind_Resume'
02:47:08 | projects/libcxxabi/src/CMakeFiles/cxxabi_shared.dir/cxa_vector.cpp.o(.ARM.exidx+0x50): error: undefined reference to '__aeabi_unwind_cpp_pr0'
02:47:08 | projects/libcxxabi/src/CMakeFiles/cxxabi_shared.dir/stdlib_stdexcept.cpp.o(.ARM.exidx+0x0): error: undefined reference to '__aeabi_unwind_cpp_pr1'
02:47:08 | projects/libcxxabi/src/CMakeFiles/cxxabi_shared.dir/stdlib_stdexcept.cpp.o(.ARM.exidx+0x8): error: undefined reference to '__aeabi_unwind_cpp_pr1'
02:47:08 | projects/libcxxabi/src/CMakeFiles/cxxabi_shared.dir/stdlib_stdexcept.cpp.o(.ARM.exidx+0x18): error: undefined reference to '__aeabi_unwind_cpp_pr1'
02:47:08 | projects/libcxxabi/src/CMakeFiles/cxxabi_shared.dir/stdlib_stdexcept.cpp.o(.ARM.exidx+0x20): error: undefined reference to '__aeabi_unwind_cpp_pr1'
02:47:08 | /usr/lib/clang/9.0.0/include/unwind.h:221: error: undefined reference to '_Unwind_VRS_Set'
02:47:08 | /usr/lib/clang/9.0.0/include/unwind.h:214: error: undefined reference to '_Unwind_VRS_Get'
02:47:08 | /usr/lib/clang/9.0.0/include/unwind.h:214: error: undefined reference to '_Unwind_VRS_Get'
02:47:08 | /home/cleanbuild/build/workspace/soc-image-v2019.07/tmp-glibc/work-shared/llvm-project-source-9.0.0-r0/git/libcxxabi/src/cxa_personality.cpp:0: error: undefined reference to '__gnu_unwind_frame'
02:47:08 | /home/cleanbuild/build/workspace/soc-image-v2019.07/tmp-glibc/work-shared/llvm-project-source-9.0.0-r0/git/libcxxabi/src/cxa_personality.cpp:0: error: undefined reference to '_Unwind_VRS_Set'
02:47:08 | /home/cleanbuild/build/workspace/soc-image-v2019.07/tmp-glibc/work-shared/llvm-project-source-9.0.0-r0/git/libcxxabi/src/cxa_personality.cpp:0: error: undefined reference to '_Unwind_VRS_Set'
02:47:08 | /home/cleanbuild/build/workspace/soc-image-v2019.07/tmp-glibc/work-shared/llvm-project-source-9.0.0-r0/git/libcxxabi/src/cxa_personality.cpp:0: error: undefined reference to '_Unwind_VRS_Get'
02:47:08 | /home/cleanbuild/build/workspace/soc-image-v2019.07/tmp-glibc/work-shared/llvm-project-source-9.0.0-r0/git/libcxxabi/src/cxa_personality.cpp:0: error: undefined reference to '_Unwind_VRS_Set'
02:47:08 | /home/cleanbuild/build/workspace/soc-image-v2019.07/tmp-glibc/work-shared/llvm-project-source-9.0.0-r0/git/libcxxabi/src/cxa_personality.cpp:605: error: undefined reference to '_Unwind_GetLanguageSpecificData'
02:47:08 | /usr/lib/clang/9.0.0/include/unwind.h:214: error: undefined reference to '_Unwind_VRS_Get'
02:47:08 | /home/cleanbuild/build/workspace/soc-image-v2019.07/tmp-glibc/work-shared/llvm-project-source-9.0.0-r0/git/libcxxabi/src/cxa_personality.cpp:618: error: undefined reference to '_Unwind_GetRegionStart'
02:47:08 | projects/libcxxabi/src/CMakeFiles/cxxabi_shared.dir/cxa_aux_runtime.cpp.o(.ARM.exidx+0x0): error: undefined reference to '__aeabi_unwind_cpp_pr0'
02:47:08 | projects/libcxxabi/src/CMakeFiles/cxxabi_shared.dir/cxa_default_handlers.cpp.o(.ARM.exidx+0x0): error: undefined reference to '__aeabi_unwind_cpp_pr0'
02:47:08 | projects/libcxxabi/src/CMakeFiles/cxxabi_shared.dir/cxa_demangle.cpp.o(.ARM.exidx+0x98): error: undefined reference to '__aeabi_unwind_cpp_pr0'
02:47:08 | /home/cleanbuild/build/workspace/soc-image-v2019.07/tmp-glibc/work-shared/llvm-project-source-9.0.0-r0/git/libcxxabi/src/cxa_exception.cpp:280: error: undefined reference to '_Unwind_RaiseException'
02:47:08 | /home/cleanbuild/build/workspace/soc-image-v2019.07/tmp-glibc/work-shared/llvm-project-source-9.0.0-r0/git/libcxxabi/src/cxa_exception.cpp:550: error: undefined reference to '_Unwind_DeleteException'
02:47:08 | /home/cleanbuild/build/workspace/soc-image-v2019.07/tmp-glibc/work-shared/llvm-project-source-9.0.0-r0/git/libcxxabi/src/cxa_exception.cpp:608: error: undefined reference to '_Unwind_RaiseException'
02:47:08 | /home/cleanbuild/build/workspace/soc-image-v2019.07/tmp-glibc/work-shared/llvm-project-source-9.0.0-r0/git/libcxxabi/src/cxa_exception.cpp:733: error: undefined reference to '_Unwind_RaiseException'
02:47:08 | clang-9: error: linker command failed with exit code 1 (use -v to see invocation)
02:47:08 | ninja: build stopped: subcommand failed.
02:47:08 | WARNING: /home/cleanbuild/build/workspace/soc-image-v2019.07/tmp-glibc/work/cortexa9t2hf-neon-emit-linux-gnueabi/libcxx/9.0.0-r0/temp/run.do_compile.29522:1 exit 1 from 'ninja -v -j 16 cxxabi'
[... cut ...] p.o utils/TableGen/CMakeFiles/llvm-tblgen.dir/WebAssemblyDisassemblerEmitter.cpp.o utils/TableGen/CMakeFiles/llvm-tblgen.dir/CTagsEmitter.cpp.o  -o bin/llvm-tblgen  -Wl,-rpath,"\$ORIGIN/../lib" lib/libLLVMSupport.a lib/libLLVMTableGen.a -lpthread lib/libLLVMSupport.a -lz -lrt -ldl -lpthread -lm lib/libLLVMDemangle.a && :
02:48:23 | clang-9: error: unable to execute command: Segmentation fault (core dumped)
02:48:23 | clang-9: error: linker command failed due to signal (use -v to see invocation)`