kraj/meta-clang #137: I would like to use lldb in the SDK

Lorac, 2019-08-06:
Using my IDE, I would like to use lldb as my debugger to remote debug on my device.

I have patched the lldb recipe with 

```
BBCLASSEXTEND = "native nativesdk"
```

And I added `lldb` inside a `packagegroup-core-tools-debug.bbappend`

Does the LLDB recipe is missing `BBCLASSEXTEND`?

kraj, 2019-08-06:
@Lorac yes it is missing and it was not straightforward to build lldb without rebuilding clang and llvm since it needs some intermediate build artifacts. On `kraj/master` branch I am working on 9.0  upgrade where it works now since lldb is now being built along with clang in one recipe.

Lorac, 2019-08-06:
For the 9.0 release will I have to add/modify anything to get the `lldb` inside my SDK? Or if I have the `TOOLCHAIN = "clang"` inside my recipe it will get deployed automatically (lldb)?

kraj, 2019-08-06:
there is no change needed from your side it will already be bundled with clang and nativesdk-clang

kraj, 2019-08-13:
9.x is committed. Please try latest master and let me know if all is okay

Lorac, 2019-08-15:
I think that the layer is no more compatible with `thud` and `sumo` because of the libedit change. We are still stuck on thud at the moment.

libedit does exists in thud (meta-oe, not oe-core) but missing `BBCLASSEXTEND = "native nativesdk"`. I can patch that locally pretty easily. Worth creating a patch for it?

Another thing is that my system is using Debian Buster, and it's using python 3.7.3, I get this error

```
Found PythonInterp: /usr/local/bin/python3.7 (found version "3.7") Why my own sysroot?
....
....
| -- Found PythonLibs: /builds/yocto/MACHINE/tmp/work/x86_64-linux/clang-native/9.0.0-r0/recipe-sysroot-native/usr/lib/libpython3.5m.so (found version "3.5.6")
| CMake Error at /builds/yocto/MACHINE/tmp/work-shared/llvm-project-source-9.0.0-r0/git/lldb/cmake/modules/LLDBConfig.cmake:204 (message):
|   Found incompatible Python interpreter (3.7) and Python libraries (3.5.6)
| Call Stack (most recent call first):
|   /builds/yocto/MACHINE/tmp/work-shared/llvm-project-source-9.0.0-r0/git/lldb/CMakeLists.txt:20 (include)
|
```

I'm using pyenv to deal with multiple python versions for various projects.

Lorac, 2019-08-15:
I do get new warnings now too

```
WARNING: clang-cross-aarch64-9.0.0-r0 do_populate_lic: clang-cross-aarch64: No generic license file exists for: Apache-2.0-with-LLVM-exception in any provider
WARNING: libcxx-9.0.0-r0 do_populate_lic: libcxx: No generic license file exists for: Apache-2.0-with-LLVM-exception in any provider
WARNING: compiler-rt-9.0.0-r0 do_populate_lic: compiler-rt: No generic license file exists for: Apache-2.0-with-LLVM-exception in any provider
WARNING: clang-native-9.0.0-r0 do_populate_lic: clang-native: No generic license file exists for: Apache-2.0-with-LLVM-exception in any provider
```

Is this due that I'm not on `warrior` yet?

kraj, 2019-08-15:
@Lorac yes it is due to other layers being warrior, you need to backport the following oe-core commit specifically
https://git.openembedded.org/openembedded-core/commit/?id=fc634c41e4b3fbaf29dc0104ae6b15757e77f60a

kraj, 2019-08-15:
> I think that the layer is not compatible anymore with `thud` and `sumo` because of the libedit change. We are still stuck on thud at the moment.
> 
> libedit does exists in thud (meta-oe, not oe-core) but missing `BBCLASSEXTEND = "native nativesdk"`. I can patch that locally pretty easily. Worth creating a patch for it?

perhaps yes

> Another thing is that my system is using Debian Buster, and it's using python 3.7.3, I get this error
>
> ```
> Found PythonInterp: /usr/local/bin/python3.7 (found version "3.7") Why my own sysroot?
> .... [...]

I think this was fixed by 1746d05

Lorac, 2019-08-15:
I am on latest master so I do have that commit.

Lorac, 2019-08-15:
pyenv isn't the issue here. I just checked and it's an old manual install of python 3.7.0 that I did back of couple months.

I guess https://github.com/kraj/meta-clang/commit/1746d053273024037e3ef227130a6582f191993e doesn't fix all of it?

kraj, 2019-08-15:
hmmm it should direct it to use the python from native sysroot and nothing else, so it should fix mix and match thats happening here.

Lorac, 2019-09-03:
Having this error during stage-2: 

```
-- Found PythonInterp: /builds/yocto/MACHINE/tmp/hosttools/python (found version "2.7.16")
...
-- Found PythonLibs: /usr/local/lib/libpython3.7m.a
| CMake Error at /builds/yocto/dublin/tmp/work-shared/llvm-project-source-9.0.0-r0/git/lldb/cmake/modules/LLDBConfig.cmake:229 (message):
|   Found incompatible Python interpreter (2.7) and Python libraries (3.7)
```

kraj, 2019-09-03:
hmm so its still there if you have local python install

Lorac, 2019-09-03:
I feel that : https://github.com/kraj/meta-clang/blob/9fd82b1647461de153c8da5eaa606f93cb933e34/recipes-devtools/clang/clang_git.bb#L22

Shouldn't be set?

kraj, 2019-09-03:
> I feel that :
> 
> https://github.com/kraj/meta-clang/blob/9fd82b1647461de153c8da5eaa606f93cb933e34/recipes-devtools/clang/clang_git.bb#L22
> 
> Shouldn't be set?

does this work without it ?

Lorac, 2019-09-03:
So I removed it and the `-DPYTHON_EXECUTABLE=${PYTHON} \` and now I get:

```
| -- Found PythonInterp: /builds/yocto/MACHINE/tmp/hosttools/python (found version "2.7.16")
...
| -- Found PythonLibs: /builds/yocto/MACHINE/tmp/work/x86_64-linux/clang-native/9.0.0-r0/recipe-sysroot-native/usr/lib/libpython3.5m.so (found version "3.5.6")
| CMake Error at /builds/yocto/MACHINE/tmp/work-shared/llvm-project-source-9.0.0-r0/git/lldb/cmake/modules/LLDBConfig.cmake:229 (message):
|   Found incompatible Python interpreter (2.7) and Python libraries (3.5)
```

kraj, 2019-09-04:
can you try this patch and report
https://github.com/YoeDistro/meta-clang/commit/6c6db3db7e88deed8b28f98466960d9ca5b375fb

kraj, 2019-09-05:
Please cherry-pick https://github.com/kraj/meta-clang/pull/163 to meta-clang and https://patchwork.openembedded.org/patch/164648/ to oe-core and report back

Lorac, 2019-09-05:
That's with your patch and without `OECMAKE_FIND_ROOT_PATH_MODE_PROGRAM = "BOTH"` during stage-2 build

```
| -- Found PythonInterp: /builds/yocto/MACHINE/tmp/hosttools/python (found version "2.7.16")
...
| -- Found PythonLibs: /usr/local/lib/libpython3.7m.a (found version "3.7.0")
| CMake Error at /builds/yocto/MACHINE/tmp/work-shared/llvm-project-source-9.0.0-r0/git/lldb/cmake/modules/LLDBConfig.cmake:229 (message):
|   Found incompatible Python interpreter (2.7) and Python libraries (3.7)
| Call Stack (most recent call first):
```

That's with your patch and with `OECMAKE_FIND_ROOT_PATH_MODE_PROGRAM = "BOTH"` during stage-2 build

```
| -- Found PythonInterp: /builds/yocto/MACHINE/tmp/hosttools/python (found version "2.7.16")
...
| -- Found PythonLibs: /usr/local/lib/libpython3.7m.a (found version "3.7.0")
| CMake Error at /builds/yocto/MACHINE/tmp/work-shared/llvm-project-source-9.0.0-r0/git/lldb/cmake/modules/LLDBConfig.cmake:229 (message):
|   Found incompatible Python interpreter (2.7) and Python libraries (3.7)
```

looks like `OECMAKE_FIND_ROOT_PATH_MODE_PROGRAM` has no effect.

kraj, 2019-09-07:
@Lorac can you try with latest master of oe-core/poky and meta-clang and see if it helps ?
I am also adding a packageconfig to disable python support in lldb which should also bypass
this error

Lorac, 2019-09-07:
```
| -- Found PythonInterp: /builds/yocto/MACHINE/tmp/hosttools/python (found version "2.7.16")
...
| -- Found PythonLibs: /usr/local/lib/libpython3.7m.a (found version "3.7.0")
| CMake Error at /builds/yocto/MACHINE/tmp/work-shared/llvm-project-source-9.0.0-r0/git/lldb/cmake/modules/LLDBConfig.cmake:228 (message):
|   Found incompatible Python interpreter (2.7) and Python libraries (3.7)
```
 
Same error with all master.

> I am also adding a packageconfig to disable python support in lldb which should also bypass
this error

I need Python support for my case, but for those who don't it make sense to create a package config!

kraj, 2019-09-07:
strange, whats your build host distro ? which recipe is failing ( native or target )
libpython is now provided on cmake cmdline so it should not be looking into searching for them at all. there is a patch to disable check for pythonInterp, your logs show nothing is effective. Can you post your Build Configuration as printed by bitbake ?

Lorac, 2019-09-07:
> whats your build host distro

Using latest Debian : `Linux mbedesk 4.19.0-5-amd64 #1 SMP Debian 4.19.37-5+deb10u1 (2019-07-19) x86_64 GNU/Linux`

> which recipe is failing

clang-native

```
BB_VERSION           = "1.43.1"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "qemuarm64"
DISTRO               = "poky"
DISTRO_VERSION       = "2.7+snapshot-20190907"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "HEAD:f388dd48283d4cb0c9e82e140703295e67e659a1"
meta-oe              
meta-networking      
meta-multimedia      
meta-python          = "HEAD:8d4d608b4e937bb3b8e3b260bd75338c3ff7e8fd"
meta-clang           = "HEAD:3533df716f1a98293d7e42413cb85964ca041898"
```

kraj, 2019-09-07:
In my case I am seeing
```
-- Found PythonInterp: /mnt/b/yoe/build/tmp/work/x86_64-linux/clang-native/9.0.0-r0/recipe-sysroot-native/usr/bin/python3-native/python3 (found version "3.7.4")
...
-- Found PythonLibs: /mnt/b/yoe/build/tmp/work/x86_64-linux/clang-native/9.0.0-r0/recipe-sysroot-native/usr/lib/libpython3.7m.so (found version "3.7.4")
```

Lorac, 2019-09-08:
Do you have a manual installation of python on your system?

kraj, 2019-09-08:
I dont but I wonder why its looking into `/builds/yocto/MACHINE/tmp/hosttools/python`

Lorac, 2019-09-09:
I looked in the logs and during [cmake configuration](https://pastebin.com/2U1kKpvz) the paths are correct, but during the second stage during lldb [compilation logs](https://pastebin.com/x913gm99)

kraj, 2019-09-09:
Can you try https://github.com/kraj/meta-clang/pull/165 on top of latest master and report back

Lorac, 2019-09-10:
During `| [2609/2613] Performing configure step for 'stage2'`

`| -- Found PythonInterp: /builds/yocto/MACHINE/tmp/work/x86_64-linux/clang-native/9.0.0-r0/recipe-sysroot-native/usr/bin/python3-native/python3 (found version "3.7.4")
`

`| -- Found PythonLibs: /builds/yocto/MACHINE/tmp/work/x86_64-linux/clang-native/9.0.0-r0/recipe-sysroot-native/usr/lib/libpython3.7m.so (found version "3.7.4")
`

And then 

```
| -- Build files have been written to: /builds/yocto/MACHINE/tmp/work/x86_64-linux/clang-native/9.0.0-r0/build/tools/clang/stage2-bins
| [2610/2613] Performing build step for 'stage2'
```

```
| [176/4196] Building CXX object utils/TableGen/CMakeFiles/llvm-tblgen.dir/RegisterBankEmitter.cpp.o
| [177/4196] Configuring NATIVE LLVM...
| FAILED: NATIVE/CMakeCache.txt
...
| -- Found PythonInterp: /builds/yocto/MACHINE/tmp/hosttools/python (found version "2.7.16")
...
| -- Found PythonLibs: /usr/local/lib/libpython3.7m.a (found version "3.7.0")
...
| FAILED: tools/clang/stage2-stamps/stage2-build
```

kraj, 2019-09-10:
Disable stage2 

from `clang_git.bb` Remove
```
BOOTSTRAPSTAGE_class-native = "stage2"
INSTALLTARGET_class-native = "stage2-install"
```
does that help ?

Lorac, 2019-09-10:
fails on `install: cannot stat '/builds/yocto/MACHINE/tmp/work/x86_64-linux/clang-native/9.0.0-r0/build/tools/clang/stage2-bins/bin/lldb-tblgen': No such file or directory
WARNING: exit code 1 from a shell command.
`

kraj, 2019-09-10:
yeah no more stage2 where is it now in you ${B} find it and then change
```
install -Dm 0755 ${B}/tools/clang/stage2-bins/bin/lldb-tblgen ${D}${bindir}/lldb-tblgen
```
to
```
install -Dm 0755 ${B}/tools/clang/bin/lldb-tblgen ${D}${bindir}/lldb-tblgen
```

Lorac, 2019-09-10:
I had to use 
```
install -Dm 0755 ${B}/bin/lldb-tblgen ${D}${bindir}/lldb-tblgen
```
but now it builds

kraj, 2019-09-10:
Okay so now we need to figure out why stage2 is not respecting python settings from stage1

Lorac, 2019-09-13:
Is it possible to split `do_compile` in 2 tasks? Let's say that we could do : `do_compile` for stage1 and then have `do_compile_stage2`, it might be easier to debug, because having to build the first part all the time takes a lot of time (at least on my system).

kraj, 2019-09-14:
@Lorac can you check latest master and see if lldb w/ python works now

Lorac, 2019-09-15:
Yes without Stage-2 it works

kraj, 2019-09-15:
Thanks. I think I will leave defaults with stage2 disabled until the python detection mechanism becomes sound in future. So we can close this ticket.

Lorac, 2019-09-15:
> until the python detection mechanism becomes sound in future

You mean until llvm fixes it?

kraj, 2019-09-15:
> > until the python detection mechanism becomes sound in future
> 
> You mean until llvm fixes it?

right.

kraj, 2019-09-18:
I will mark this done based on @Lorac reports on building without stage2

closed 2019-09-18 by kraj