kraj/meta-clang #147: Does not build with Warrior for 8.0.1 or 9.0.0

madscientist42, 2019-08-15:
ERROR: clang-native-9.0.0-r0 do_compile: Function failed: do_compile (log file is located at /home/fearl/g3incar_build/build/tmp/work/x86_64-linux/clang-native/9.0.0-r0/temp/log.do_compile.2401097)
ERROR: Logfile of failure stored in: /home/fearl/g3incar_build/build/tmp/work/x86_64-linux/clang-native/9.0.0-r0/temp/log.do_compile.2401097
Log data follows:
| DEBUG: Executing shell function do_compile
| ninja: error: unknown target 'stage2'
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/fearl/g3incar_build/build/tmp/work/x86_64-linux/clang-native/9.0.0-r0/temp/log.do_compile.2401097)
ERROR: Task (virtual:native:/home/fearl/g3incar_build/build/../meta-clang/recipes-devtools/clang/clang_git.bb:do_compile) failed with exit code '1'

Same thing with differing versions on Warrior.  Simply put, metadata is asking for things that flatly DON'T exist in the build system.

kraj, 2019-08-16:
hmm are you using `warrior` branch of meta-clang as well

madscientist42, 2019-08-16:
Yes.

kraj, 2019-08-16:
Can you confirm if you use master
Of all layers then you can build master of meta-clang 9.0 on same build host ?

Secondly what is your build host os ?

I will see if I can reproduce it here

madscientist42, 2019-08-16:
I can, but this looks more like a missed patch.  Building with Warrior or
Master wouldn't change a state where your second stage call-out is missing
from the build targets...

kraj, 2019-08-16:
> I can, but this looks more like a missed patch. Building with Warrior or Master wouldn't change a state where your second stage call-out is missing from the build targets...
> […](#)

this should have been fixed with https://github.com/kraj/meta-clang/commit/4e05dac9af2b9f000e31f799511899870663b482

kraj, 2019-08-16:
I have backported the above said fix to warrior try it again please.

madscientist42, 2019-08-16:
Will do.  Will be later today.  Will keep  you posted.

madscientist42, 2019-08-23:
Sorry for the delay in response.  Got pulled off onto a fire-drill for this stuff...  X-D

Back onto it and got this as a result to this point:

ERROR: Nothing PROVIDES 'virtual/nativesdk-x86_64-pokysdk-linux-libc-for-gcc' (but /home/fearl/g3incar_build/build/../meta-clang/recipes-devtools/clang/clang-cross-canadian_git.bb DEPENDS on or otherwise requires it). Close matches:
  virtual/nativesdk-x86_64-pokysdk-linux-compilerlibs
  virtual/nativesdk-x86_64-pokysdk-linux-go-runtime
NOTE: Runtime target 'clang-cross-canadian-aarch64' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['clang-cross-canadian-aarch64', 'virtual/nativesdk-x86_64-pokysdk-linux-libc-for-gcc']
NOTE: Runtime target 'packagegroup-cross-canadian-g2wearevm' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-cross-canadian-g2wearevm', 'clang-cross-canadian-aarch64', 'virtual/nativesdk-x86_64-pokysdk-linux-libc-for-gcc']
ERROR: Required build target 'g2wear-minimal-image' has no buildable providers.
Missing or unbuildable dependency chain was: ['g2wear-minimal-image', 'packagegroup-cross-canadian-g2wearevm', 'clang-cross-canadian-aarch64', 'virtual/nativesdk-x86_64-pokysdk-linux-libc-for-gcc']

Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
fearl@dev1:~/g3incar_build/build$ 

I'm needing to use Linaro 7.2 for this specific target for several reasons (CUDA being one of the main ones...) so is this something where we've got an assumption of GCC 8.X and it's recipe behaviors and I need to try to re-work GCC 7.3 from Sumo/Thud to act right for us here?

madscientist42, 2019-08-28:
> probably backporting this commit openembedded/openembedded-core@122217b
> from oe-core will help

That was kind of my thought on things with 7.3  Should be able to get to
something in that vein tomorrow...now that I sorted out other "more
pressing" matters for the day job.  :D

madscientist42, 2019-08-29:
Got a build going right now.  Looks more like being pulled off on the fire drill was the culprit.  Found out that my stuff in the metadata for PR purposes on some of this was inconsistent.  Will keep updated, but it looks like maybe this WAS the only fix needed for that part on this metadata layer.  :D

madscientist42, 2019-08-30:
Nope.  Still busted with Warrior.  Something's off in that it just simply doesn't stage up the stage2 part of the build for native right at the moment.  Doesn't matter if I'm building for my GCC 7.x specific targets or if I'm building for the 8.x ones...it does the same fail. 

Rather annoying, really.  Guess I'll be doing digging to see if I can help find this one.

kraj, 2019-08-30:
Maybe try building with pure warrior oe-core/bitbake/meta-clang and see if you can reproduce it

madscientist42, 2019-08-30:
It's beginning to look like the multi-stage compilation is not working in my case.  Upon turning it off fully in the recipe (Using the older recipes for guidance, it's getting MUCH further along...

madscientist42, 2019-08-30:
Will do on the other, it'll take me staging up my personal setup to do it- so it may be this evening before I can try there for you.

madscientist42, 2019-09-03:
It's looking like it's something to do with the GCC 7.X stuff...  That's going to be "fun" and a problem- there's a lot of SoCs that need Linaro GCC 7.2 or GCC  7.3 to build with.

I'll dig further and keep everyone updated.

madscientist42, 2019-09-05:
Looks like it was self-inflicted (as in one of my team did it to me...  X-D) there was a .bbappend I didn't know much of the details about that was a Kludge for the senior build engineer's needs/purposes that broke the build process for the native step.  Things are now building.  Will close this one and work on the request in the other ticket to document that it's a simple thing to get the Classic SDK to build for Autotools and CMake builds.

closed 2019-09-05 by madscientist42