kraj/meta-clang #157: libcxx_git fails: undefined reference to `_Unwind_Resume'

saininav, 2019-08-26:
Build failing with latest master meta-clang and poky. I am using default configuration (gcc toolchain). 

| /data/saininav/work/workspace/poky/build-opencl/tmp/work/core2-64-poky-linux/libcxx/9.0.0-r0/recipe-sysroot-native/usr/bin/x86_64-poky-linux/../../libexec/x86_64-poky-linux/gcc/x86_64-poky-linux/9.2.0/ld: projects/libcxxabi/src/CMakeFiles/cxxabi_shared.dir/cxa_personality.cpp.o: in function `__cxa_call_unexpected':
| /data/saininav/work/workspace/poky/build-opencl/tmp/work-shared/llvm-project-source-9.0.0-r0/git/libcxxabi/src/cxa_personality.cpp:1234: undefined reference to `_Unwind_Resume'
| /data/saininav/work/workspace/poky/build-opencl/tmp/work/core2-64-poky-linux/libcxx/9.0.0-r0/recipe-sysroot-native/usr/bin/x86_64-poky-linux/../../libexec/x86_64-poky-linux/gcc/x86_64-poky-linux/9.2.0/ld: projects/libcxxabi/src/CMakeFiles/cxxabi_shared.dir/cxa_thread_atexit.cpp.o: in function `__cxa_thread_atexit':
| /data/saininav/work/workspace/poky/build-opencl/tmp/work-shared/llvm-project-source-9.0.0-r0/git/libcxxabi/src/cxa_thread_atexit.cpp:111: undefined reference to `_Unwind_Resume'
| collect2: error: ld returned 1 exit status
| ninja: build stopped: subcommand failed.
| WARNING: exit code 1 from a shell command.

kraj, 2019-08-26:
Can you try adding 
```
TOOLCHAIN = "clang" 
```
in the libcxx recipe and see if that helps.

reid-p, 2019-08-26:
Still compiling but so far looks to be the furthest the build has progressed for me.
Seems to address my failures reported on #136

saininav, 2019-08-26:
> Can you try adding
> 
> ```
> TOOLCHAIN = "clang" 
> ```
> 
> in the libcxx recipe and see if that helps.

Build gets passed. Thanks Khem.
Should I expect a patch soon !