kraj/meta-clang #162: Build failure: Found incompatible Python interpreter (2.7) and Python libraries (3.7)

saininav, 2019-09-04:
I am building with latest master branch. [9fd82b1647461de153c8da5eaa606f93cb933e34]

Looks clang-native get pass but not non-native. Issue is with lldb.

Error log:
-- Found PythonLibs: /data/saininav/work/workspace/poky/build-opencl/tmp/work/corei7-64-poky-linux/clang/9.0.0-r0/recipe-sysroot-native/usr/lib/libpython3.7m.so (found version "3.7.4") 
CMake Error at /data/saininav/work/workspace/poky/build-opencl/tmp/work-shared/llvm-project-source-9.0.0-r0/git/lldb/cmake/modules/LLDBConfig.cmake:229 (message):
  Found incompatible Python interpreter (2.7) and Python libraries (3.7)
Call Stack (most recent call first):
  /data/saininav/work/workspace/poky/build-opencl/tmp/work-shared/llvm-project-source-9.0.0-r0/git/lldb/CMakeLists.txt:20 (include)

kraj, 2019-09-04:
Can you change in `clang_git.bb`

```
EXTRA_OECMAKE_append_class-target = "\
                  -DCMAKE_CROSSCOMPILING:BOOL=ON \
```
to
```
EXTRA_OECMAKE_append_class-target = "\
                  -DCMAKE_CROSSCOMPILING=TRUE \
```

and see if that helps

kraj, 2019-09-04:
and can you try this patch and report back

https://github.com/YoeDistro/meta-clang/commit/6c6db3db7e88deed8b28f98466960d9ca5b375fb

kraj, 2019-09-05:
Please cherry-pick https://github.com/kraj/meta-clang/pull/163 to meta-clang and https://patchwork.openembedded.org/patch/164648/ to oe-core and report back

saininav, 2019-09-05:
> Please cherry-pick #163 to meta-clang and https://patchwork.openembedded.org/patch/164648/ to oe-core and report back

Build still fails with above patches. Same error.

| -- Looking for os_signpost_interval_begin
| -- Looking for os_signpost_interval_begin - not found
| -- Found PythonInterp: /data/saininav/work/workspace/poky/build-opencl/tmp/hosttools/python (found version "2.7.12")
| -- Constructing LLVMBuild project information
| -- Found Git: /data/saininav/work/workspace/poky/build-opencl/tmp/hosttools/git (found version "2.7.4")
| -- Linker detection: GNU ld
| -- Targeting X86
| -- Looking for sys/resource.h
| -- Looking for sys/resource.h - found
| -- Clang version: 9.0.0
| -- Performing Test CXX_SUPPORTS_NO_NESTED_ANON_TYPES_FLAG
| -- Performing Test CXX_SUPPORTS_NO_NESTED_ANON_TYPES_FLAG - Success
| -- Looking for include file sys/inotify.h
| -- Looking for include file sys/inotify.h - found
| -- LLD version: 9.0.0
| -- Found LibEdit: /data/saininav/work/workspace/poky/build-opencl/tmp/work/corei7-64-poky-linux/clang/9.0.0-r0/recipe-sysroot-native/usr/include (found version ".")
| -- Looking for el_winsertstr
| -- Looking for el_winsertstr - found
| -- Looking for stdint.h
| -- Looking for stdint.h - found
| -- Looking for stddef.h
| -- Looking for stddef.h - found
| -- Check size of el_rfunc_t
| -- Check size of el_rfunc_t - done
| -- Found PythonLibs: /data/saininav/work/workspace/poky/build-opencl/tmp/work/corei7-64-poky-linux/clang/9.0.0-r0/recipe-sysroot-native/usr/lib/libpython3.7m.so (found version "3.7.4")
| CMake Error at /data/saininav/work/workspace/poky/build-opencl/tmp/work-shared/llvm-project-source-9.0.0-r0/git/lldb/cmake/modules/LLDBConfig.cmake:229 (message):
|   Found incompatible Python interpreter (2.7) and Python libraries (3.7)
| Call Stack (most recent call first):
|   /data/saininav/work/workspace/poky/build-opencl/tmp/work-shared/llvm-project-source-9.0.0-r0/git/lldb/CMakeLists.txt:20 (include)

kraj, 2019-09-05:
@saininav  can you drop the last patch and replace it with https://github.com/YoeDistro/meta-clang/commit/ff644955ec94fa93a6c196696b1eaeac29bb8a08 and give it a shot again

saininav, 2019-09-05:
> @saininav can you drop the last patch and replace it with [YoeDistro@ff64495](https://github.com/YoeDistro/meta-clang/commit/ff644955ec94fa93a6c196696b1eaeac29bb8a08) and give it a shot again

Ok

saininav, 2019-09-05:
> > @saininav can you drop the last patch and replace it with [YoeDistro@ff64495](https://github.com/YoeDistro/meta-clang/commit/ff644955ec94fa93a6c196696b1eaeac29bb8a08) and give it a shot again
> 
> Ok

> @saininav can you drop the last patch and replace it with [YoeDistro@ff64495](https://github.com/YoeDistro/meta-clang/commit/ff644955ec94fa93a6c196696b1eaeac29bb8a08) and give it a shot again

Still build fails. Same error.

| -- Found PythonInterp: /data/saininav/work/workspace/poky/build-opencl/tmp/hosttools/python (found version "2.7.12")

saininav, 2019-09-05:
[log.do_compile_75102.txt](https://github.com/kraj/meta-clang/files/3577910/log.do_compile_75102.txt)

kraj, 2019-09-07:
@saininav I tried setting python to 2.x and python 3.x and it built fine here for me.  The oe-core change is in master now and the relevant pull for meta-clang is also merged into master, can you try with latest master of both layers and report back

saininav, 2019-09-10:
> @saininav I tried setting python to 2.x and python 3.x and it built fine here for me. The oe-core change is in master now and the relevant pull for meta-clang is also merged into master, can you try with latest master of both layers and report back
I still face the issue. Let me try with other build machine.

kraj, 2019-09-14:
@anujm1 @saininav  can you guys try latest master and see if the python issue is settled ?

oxygene, 2019-09-17:
I don't mean to hijack this issue, but I came here via google because of the very same error message.

This happened for me while being 788b680, and updating to most recent master fd43459 fixed the issue for me.

kraj, 2019-09-18:
based on @oxygene report I am closing it as fixed. Please reopen if not so

closed 2019-09-18 by kraj

rchard2scout, 2019-12-05:
I had this same issue while building nativesdk-clang, I fixed it by adding `-DPYTHON_EXECUTABLE='${PYTHON}' \` to `EXTRA_OECMAKE_append_class-nativesdk` in clang_git.bb. @kraj is that something that you want to take? Or should I keep it as a local bbappend?

kraj, 2019-12-05:
@rchard2scout please send a patch for nativesdk

reid-p, 2019-12-06:
> 
> 
> I had this same issue while building nativesdk-clang, I fixed it by adding `-DPYTHON_EXECUTABLE='${PYTHON}' \` to `EXTRA_OECMAKE_append_class-nativesdk` in clang_git.bb. @kraj is that something that you want to take? Or should I keep it as a local bbappend?

This fixed the build for me also.

PR #242 (2019-12-06): clang: ensure correct python for nativesdk builds
This fixes #162 for nativesdk-clang.
commit 17dd87b3 (2019-12-06): clang: ensure correct python for nativesdk builds

This fixes #162 for nativesdk-clang.
commit 4c17cbfc (2019-12-07): clang: ensure correct python for nativesdk builds

This fixes #162 for nativesdk-clang.