kraj/meta-clang #185: DEBUG_PREFIX_MAP should have the clang override.

jpuhlman, 2019-09-23:
DEBUG_PREFIX_MAP = ""

My thud builds which include the clang class, started failing to build apr and a couple other apps(still looking into it), but it appears the addition of the above caused the problem.

do_install_append_class-target() {
    sed -i -e 's,${DEBUG_PREFIX_MAP},,g' \
           -e 's,${STAGING_DIR_HOST},,g' ${D}${datadir}/build-1/apr_rules.mk
    sed -i -e 's,${STAGING_DIR_HOST},,g' \
           -e 's,APR_SOURCE_DIR=.*,APR_SOURCE_DIR=,g' \
           -e 's,APR_BUILD_DIR=.*,APR_BUILD_DIR=,g' ${D}${bindir}/apr-1-config
}

In apr, the first sed expression turns in to:
sed 's,,,g'
Which yeilds:
sed: -e expression #1, char 0: no previous regular expression

Adding the clang override won't fix the build for clang, but it will fix it for anything not using clang. It looks like the way the value is being used it is expecting something in the value. Maybe add the override and make it " " rather then "".

kraj, 2019-09-24:
Yes this should have toolchain-clang override

kraj, 2019-09-24:
I think let’s make both the changes you suggested since it’s only problematic in clang7 we won’t need this fix for newer branches

jpuhlman, 2019-09-24:
I have a validation build running right now, I'll createa pull request when it's done or feel free.

kraj, 2019-09-24:
https://github.com/kraj/meta-clang/pull/186 fixed it

closed 2019-09-24 by kraj