kraj/meta-clang #188: libc++ fails to link with: unrecognized option '--color-diagnostics'

MastaG, 2019-09-25:
Hi I'm building using the master branch of meta-clang, meta-openembedded, bitbake, openembedded.

I've only set: TOOLCHAIN = "clang" in my distro config.
Also I've added: ld-is-ldd to my DISTRO_FEATURES.

But clang++ git fails to compile with:
/home/mastag/src/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/libcxx/9.0.0-r0/recipe-sysroot-native/usr/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-ld: unrecognized option '--color-diagnostics'
| /home/mastag/src/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/libcxx/9.0.0-r0/recipe-sysroot-native/usr/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-ld: use the --help option for usage information 

Any ideas?

MastaG, 2019-09-25:
I think it's because the GNU linker gets staged in the sysroot for clang++ instead of llvm's lld.

/home/mastag/src/odroid-oe-core/build/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/libcxx/9.0.0-r0/recipe-sysroot-native/usr/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-ld -v
GNU ld (GNU Binutils) 2.32.0.20190204

And it does not support the --color-diagnostics flag.

Why isn't it using the lld linker instead?
I did set: ld-is-ldd in distro features.

MastaG, 2019-09-26:
@kraj 
Sorry I meant ld-is-lld for distro features and I also meant libc++ (instead of clang++).. sorry I didn't had much sleep when posting this.

This is my distro config:

TOOLCHAIN ?= "clang"
NO32LIBS ??= "1"
TCLIBCAPPEND = ""
DISTRO_FEATURES ?= "ld-is-lld opengl alsa bluetooth bluez5 argp ipv6 ext2 xattr smbfs pci wifi-extra largefile wifi nfs zeroconf pam pulseaudio wayland x11 mplt systemd ldconfig selinux polkit ${DISTRO_FEATURES_LIBC}"

When building libc++ the gnu gold linker is being used instead of llvm's lld linker.
Therefor I'm getting the: unrecognized option '--color-diagnostics'

Any recent changes that might have caused this?

kraj, 2019-09-26:
I am using gnu ld as well and it links fine for me. who adds `--color-diagnostics` I wonder, secondly add `LDFLAGS += "-fuse-ld=lld"` in libcxx recipe and see if that helps

MastaG, 2019-09-26:
Hi @kraj 

The --color-diagnostics is added by libc++ itself.
I'm building on Fedora 30 btw.

After manually removing it from the build file it completes successfully and the recipe is built and working.

However is there any reason why libc++ isn't being linked with clang's lld linker?

kraj, 2019-09-26:
> Hi @kraj
> 
> The --color-diagnostics is added by libc++ itself.
> I'm building on Fedora 30 btw.
> 
> After manually removing it from the build file it completes successfully and the recipe is built and working.
> 
> However is there any reason why libc++ isn't being linked with clang's lld linker?

No reason, except that lld is not tested to link it so we rely on default system linker

what change did you make to make it work ?

I see it asks linker first if it support it in  `llvm/cmake/modules/HandleLLVMOptions.cmake`
```
if (UNIX AND CMAKE_GENERATOR STREQUAL "Ninja")
  include(CheckLinkerFlag)
  check_linker_flag("-Wl,--color-diagnostics" LINKER_SUPPORTS_COLOR_DIAGNOSTICS)
  append_if(LINKER_SUPPORTS_COLOR_DIAGNOSTICS "-Wl,--color-diagnostics"
    CMAKE_EXE_LINKER_FLAGS CMAKE_MODULE_LINKER_FLAGS CMAKE_SHARED_LINKER_FLAGS)
endif()  

```

kraj, 2019-10-02:
should be fixed now

closed 2019-10-02 by kraj

MastaG, 2019-10-02:
yes it is!
confirmed with a build from scratch!

Thanks:)