kraj/meta-clang #190: Trying to build an SDK fails with master...

madscientist42, 2019-09-28:
ERROR: Multiple .bb files are due to be built which each provide libunwind:                             | ETA:  0:00:01
  /home/fearl/g3incar_build/build/../meta/recipes-support/libunwind/libunwind_1.3.1.bb
  /home/fearl/g3incar_build/build/../meta-clang/recipes-devtools/clang/llvm-libunwind.bb
A list of tasks depending on these providers is shown and may help explain where the dependency comes from.
/home/fearl/g3incar_build/build/../meta/recipes-support/libunwind/libunwind_1.3.1.bb has unique dependees:
  /home/fearl/g3incar_build/build/../meta-wgv/recipes-wgv/wgv/wg-components.bb:do_package_write_ipk
  /home/fearl/g3incar_build/build/../meta-wgv/recipes-wgv/wgv/wg-components.bb:do_prepare_recipe_sysroot
  /home/fearl/g3incar_build/build/../meta-wgv/recipes-wgv/wgv/wg-components.bb:do_package
/home/fearl/g3incar_build/build/../meta-clang/recipes-devtools/clang/llvm-libunwind.bb has unique dependees:
  /home/fearl/g3incar_build/build/../meta-clang/recipes-devtools/clang/libcxx_git.bb:do_prepare_recipe_sysroot
  /home/fearl/g3incar_build/build/../meta-clang/recipes-devtools/clang/libcxx_git.bb:do_package
It could be that one recipe provides something the other doesn't and should. The following provider and runtime provider differences may be helpful.
/home/fearl/g3incar_build/build/../meta/recipes-support/libunwind/libunwind_1.3.1.bb has unique provides:
  
/home/fearl/g3incar_build/build/../meta/recipes-support/libunwind/libunwind_1.3.1.bb has unique rprovides:
  
/home/fearl/g3incar_build/build/../meta-clang/recipes-devtools/clang/llvm-libunwind.bb has unique provides:
  llvm-libunwind
/home/fearl/g3incar_build/build/../meta-clang/recipes-devtools/clang/llvm-libunwind.bb has unique rprovides:
  llvm-libunwind-locale
  llvm-libunwind
  llvm-libunwind-staticdev
  llvm-libunwind-dbg
  llvm-libunwind-doc
  ^llvm-libunwind-locale-.*
  llvm-libunwind-src
  llvm-libunwind-dev

madscientist42, 2019-09-28:
We're using master right now instead of Warrior because we ended up needing to take on clang 9.0 for our stuff.  I'm **_trying_** to specify that I don't want libcxx OR llvm-libunwind for this recipe, but want libstdc++ and nongnu's libunwind.  Thought I had it sort of contained until you made the recent change (Which makes complete sense, mind...and probably will make life easier for everyone, ourselves included, once we sort out what this did to us...  X-D) and now this fun.

kraj, 2019-09-28:
in the concerning recipe set
```
LIBCPLUSPLUS = ""
COMPILER_RT = ""
TOOLCHAIN = "gcc"

```

kraj, 2019-09-28:
also try https://github.com/kraj/meta-clang/pull/191

kraj, 2019-09-28:
pull #191 should fix it. Please report back if it does not

closed 2019-09-28 by kraj

madscientist42, 2019-09-30:
Will do.  About to check that out here in a moment now that I've got the
morning meetings squared away

Thanks!