kraj/meta-clang #194: ninja: build stopped: subcommand failed.

adam-lee, 2019-10-01:
I'm on `thud` branch with top commit of `1d5b1c100fe4c82bcba0f005b6cae5810fde8243`. I am running into this error when I am building my image with `-c populate_sdk`. 

```
.
.
.
[2941/3033] Linking CXX executable bin/llvm-lto
[2942/3033] Linking CXX executable bin/clang-offload-bundler
[2943/3033] Linking CXX executable bin/clang-format
[2944/3033] Building CXX object utils/unittest/CMakeFiles/gtest.dir/googletest/src/gtest-all.cc.o
[2945/3033] Building CXX object tools/clang/tools/driver/CMakeFiles/clang.dir/driver.cpp.o
[2946/3033] Building CXX object tools/clang/tools/driver/CMakeFiles/clang.dir/cc1gen_reproducer_main.cpp.o
[2947/3033] Linking CXX executable bin/diagtool
[2948/3033] Building CXX object tools/clang/tools/driver/CMakeFiles/clang.dir/cc1_main.cpp.o
[2949/3033] Building CXX object tools/clang/tools/driver/CMakeFiles/clang.dir/cc1as_main.cpp.o
[2950/3033] Linking CXX executable bin/clang-diff
[2951/3033] Linking CXX executable bin/clang-import-test
[2952/3033] Linking CXX executable bin/clang-rename
[2953/3033] Linking CXX executable bin/arcmt-test
[2954/3033] Linking CXX executable bin/clang-apply-replacements
[2955/3033] Linking CXX executable bin/clang-refactor
[2956/3033] Linking CXX executable bin/clang-7
FAILED: bin/clang-7 
: && /home/adam/thud/build/build/tmp/work/armv7at2hf-neon-poky-linux-gnueabi/clang/7.1.0-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-g++  -march=armv7-a -mthumb -mfpu=neon -mfloat-abi=hard -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/adam/thud/build/build/tmp/work/armv7at2hf-neon-poky-linux-gnueabi/clang/7.1.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/adam/thud/build/build/tmp/work/armv7at2hf-neon-poky-linux-gnueabi/clang/7.1.0-r0=/usr/src/debug/clang/7.1.0-r0 -fdebug-prefix-map=/home/adam/thud/build/build/tmp/work/armv7at2hf-neon-poky-linux-gnueabi/clang/7.1.0-r0/recipe-sysroot= -fdebug-prefix-map=/home/adam/thud/build/build/tmp/work/armv7at2hf-neon-poky-linux-gnueabi/clang/7.1.0-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden  -march=armv7-a -mthumb -mfpu=neon -mfloat-abi=hard -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/adam/thud/build/build/tmp/work/armv7at2hf-neon-poky-linux-gnueabi/clang/7.1.0-r0/recipe-sysroot -fPIC -fvisibility-inlines-hidden -Werror=date-time -std=c++11 -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wno-missing-field-initializers -pedantic -Wno-long-long -Wno-maybe-uninitialized -Wno-class-memaccess -Wdelete-non-virtual-dtor -Wno-comment -fdiagnostics-color -ffunction-sections -fdata-sections -fno-common -Woverloaded-virtual -fno-strict-aliasing -DNDEBUG  -march=armv7-a -mthumb -mfpu=neon -mfloat-abi=hard -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/adam/thud/build/build/tmp/work/armv7at2hf-neon-poky-linux-gnueabi/clang/7.1.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/adam/thud/build/build/tmp/work/armv7at2hf-neon-poky-linux-gnueabi/clang/7.1.0-r0=/usr/src/debug/clang/7.1.0-r0 -fdebug-prefix-map=/home/adam/thud/build/build/tmp/work/armv7at2hf-neon-poky-linux-gnueabi/clang/7.1.0-r0/recipe-sysroot= -fdebug-prefix-map=/home/adam/thud/build/build/tmp/work/armv7at2hf-neon-poky-linux-gnueabi/clang/7.1.0-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden  -march=armv7-a -mthumb -mfpu=neon -mfloat-abi=hard -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/adam/thud/build/build/tmp/work/armv7at2hf-neon-poky-linux-gnueabi/clang/7.1.0-r0/recipe-sysroot -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong -Wl,-z,relro,-z,now -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong -Wl,-z,relro,-z,now -Wl,-allow-shlib-undefined   -Wl,--export-dynamic  -Wl,-rpath-link,/home/adam/thud/build/build/tmp/work/armv7at2hf-neon-poky-linux-gnueabi/clang/7.1.0-r0/build/./lib  -Wl,-O3 tools/clang/tools/driver/CMakeFiles/clang.dir/driver.cpp.o tools/clang/tools/driver/CMakeFiles/clang.dir/cc1_main.cpp.o tools/clang/tools/driver/CMakeFiles/clang.dir/cc1as_main.cpp.o tools/clang/tools/driver/CMakeFiles/clang.dir/cc1gen_reproducer_main.cpp.o  -o bin/clang-7  -Wl,-rpath,"\$ORIGIN/../lib" -lpthread lib/libclangBasic.a lib/libclangCodeGen.a lib/libclangDriver.a lib/libclangFrontend.a lib/libclangFrontendTool.a lib/libclangCodeGen.a lib/libclangRewriteFrontend.a lib/libclangARCMigrate.a lib/libclangStaticAnalyzerFrontend.a lib/libclangStaticAnalyzerCheckers.a lib/libclangStaticAnalyzerCore.a lib/libclangCrossTU.a lib/libclangIndex.a lib/libclangFrontend.a lib/libclangDriver.a lib/libclangParse.a lib/libclangSerialization.a lib/libclangSema.a lib/libclangAnalysis.a lib/libclangEdit.a lib/libclangFormat.a lib/libclangToolingInclusions.a lib/libclangToolingCore.a lib/libclangRewrite.a lib/libclangASTMatchers.a lib/libclangAST.a lib/libclangLex.a lib/libclangBasic.a lib/libLLVM-7.1.so && :
collect2: fatal error: ld terminated with signal 9 [Killed]
compilation terminated.
[2957/3033] Linking CXX executable bin/clang-func-mapping
[2958/3033] Linking CXX executable bin/modularize
[2959/3033] Linking CXX executable bin/clang-reorder-fields
[2960/3033] Linking CXX executable bin/clang-check
[2961/3033] Linking CXX executable bin/clang-tidy
ninja: build stopped: subcommand failed.
WARNING: exit code 1 from a shell command.
```

kraj, 2019-10-01:
mostly such issues mean your build machine has run out of memory, how much RAM does it have?

adam-lee, 2019-10-01:
It's allocated for 16GB. I will increase it to 24GB and try again. Thanks for a prompt response.

kraj, 2019-10-01:
other options is to use gold linker and see if that helps

`GN_ARGS += "use_lld=true use_gold=false"`

adam-lee, 2019-10-01:
Thanks. It's building ok now with extra RAM.

closed 2019-10-01 by adam-lee