kraj/meta-clang #195: x86_64-pokysdk-linux-llvm-ar: not found

adam-lee, 2019-10-01:
Running `bitbake clang` shows no issue, but when I build my image with `-c populate_sdk`, I am running into this issue:

```
| /bin/sh: 1: /home/adam/thud/build/build/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-compiler-rt/7.1.0-r0/recipe-sysroot-native/usr/bin/x86_64-pokysdk-linux/x86_64-pokysdk-linux-llvm-ar: not found
| [326/586] Building CXX object lib/interception/CMakeFiles/RTInterception.x86_64.dir/interception_type_test.cc.o
| [327/586] Building CXX object lib/stats/CMakeFiles/clang_rt.stats_client-x86_64.dir/stats_client.cc.o
| [328/586] Building CXX object lib/lsan/CMakeFiles/clang_rt.lsan-x86_64.dir/lsan.cc.o
| [329/586] Building CXX object lib/stats/CMakeFiles/clang_rt.stats-x86_64.dir/stats.cc.o
| [330/586] Building CXX object lib/lsan/CMakeFiles/clang_rt.lsan-x86_64.dir/lsan_allocator.cc.o
| ninja: build stopped: subcommand failed.
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/adam/thud/build/build/tmp/work/x86_64-nativesdk-pokysdk-linux/nativesdk-compiler-rt/7.1.0-r0/temp/log.do_compile.119453)
ERROR: Task (virtual:nativesdk:/home/adam/thud/poky/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2349 tasks of which 2343 didn't need to be rerun and 2 failed.
NOTE: Writing buildhistory

Summary: 2 tasks failed:
  virtual:nativesdk:/home/adam/thud/poky/meta-clang/recipes-devtools/clang/libcxx_git.bb:do_compile
  virtual:nativesdk:/home/adam/thud/poky/meta-clang/recipes-devtools/clang/compiler-rt_git.bb:do_compile
Summary: There were 3 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

```

Help will be appreciated!

kraj, 2019-10-01:
I guess you are on thud branch, in that case, can you try applying patch from

https://github.com/kraj/meta-clang/pull/196

adam-lee, 2019-10-01:
Thanks @kraj . I cherry-picked #196  and it's building `compiler-rt`, but not `libcxx`. I am getting the same error for `libcxx`.  Is `thud` still work in progress...? What is the latest stable that I should be using?

adam-lee, 2019-10-01:
I might have been wrong about `compiler-rt` building with the cherry-pick. Both `compiler-rt` and `libcxx` are still throwing the same error.

kraj, 2019-10-01:
OK I pushed another fix. thud is still supported until `zeus` releases this month. we usually support last two releases. I would suggest that you try master since that will become zeus release.

in anycase try the new patchset for thud

kraj, 2019-10-02:
Should be fixed on thud now, please reopen if not.

closed 2019-10-02 by kraj