kraj/meta-clang #197: Building with poky thud fails

russell-islam, 2019-10-02:
Loading cache: 100% |#########################################################################################| Time: 0:00:00
Loaded 3512 entries from dependency cache.
ERROR: No recipes available for:
  /home/xxx/sources/poky/../meta-clang/recipes-core/musl/musl_%.bbappend
Summary: There were 2 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.

kraj, 2019-10-02:
what branch of poky are you using?  it is `thud` too.

russell-islam, 2019-10-02:
yes

kraj, 2019-10-02:
Do you have musl recipe inside poky/meta/recipes-core/

russell-islam, 2019-10-02:
Yes. 
./sources/meta-clang/recipes-core/musl/musl_%.bbappend
./sources/poky/meta/recipes-core/musl/musl_git.bb
./sources/poky/meta/recipes-core/musl/musl-utils.bb

kraj, 2019-10-03:
interesting, Is there some other layer or config metadata which is blacklisting musl or `BBMASK`'ing it ?

kraj, 2021-03-25:
thud is EOL

closed 2021-03-25 by kraj