kraj/meta-clang #200: libunwind dependency get replaced with llvm-libunwind

saininav, 2019-10-08:
meta-clang layer is added in bblayer in order to compile opencl recipes from meta-intel.

While build igt-gpu-tools recipe from meta-intel, throw configuration error:

checking whether make supports nested variables... (cached) yes
checking for libdrm >= 2.4.82... yes
checking for pciaccess >= 0.10... yes
checking for libkmod... yes
checking for libprocps... yes
checking for libunwind... no
configure: error: Package requirements (libunwind) were not met:

No package 'libunwind' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables LIBUNWIND_CFLAGS
and LIBUNWIND_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.

While debugging with '$ bitbake igt-gpu-tool -e ' found that libunwind get replaced with llvm-libunwind.

saininav, 2019-10-08:
> if you dont want to use llvm-libunwind, in recipe please set
> 
> ```
> UNWINDLIB = ""
> ```

I created a patch, which solved my issue:
https://github.com/saininav/meta-clang/commit/8cd463bfbadfc07d692c33eeaff79984996cdee1

saininav, 2019-10-08:
Provide your feedback on https://github.com/saininav/meta-clang/commit/8cd463bfbadfc07d692c33eeaff79984996cdee1 patch

saininav, 2019-10-09:
Hi Khem,

On using UNWINDLIB = "" in my recipe. Throw below error:

ERROR: igt-gpu-tools-1.24-r0 do_prepare_recipe_sysroot: The file /usr/include/unwind.h is installed by both llvm-libunwind and libunwind, aborting

Do I need to set PREFERRED_PROVIDER ?

kraj, 2019-10-09:
I think it should be UNWINDLIB = “—unwindlib=libgcc”

saininav, 2019-10-09:
Still issue remains...

kraj, 2019-10-09:
please try with https://github.com/kraj/meta-clang/pull/204

saininav, 2019-10-09:
Build Pass. Thanks Khem. I will give a try on other opencl (clang) recipes too.

kraj, 2019-10-10:
Should be fixed now

closed 2019-10-10 by kraj