kraj/meta-clang #205: Issue when building agains Poky reference distro

schultzern, 2019-10-10:
First of all great work on this layer! 

I'm getting some issues when trying to build against Poky on the sumo release. Don't know if you support is as you list `openembedded-core` and `bitbake` as dependencies and not `Poky` but wanted to ask nonetheless.

I'm trying to build a SDK for a project with Qt and get this issue when executing `bitbake meta-toolchain-qt5 -c populate_sdk`:

```shell
ERROR: Multiple .bb files are due to be built which each provide libunwind:                                                   | ETA:  0:00:00
  /home/user/workdir/sources/poky/meta/recipes-support/libunwind/libunwind_1.2.1.bb
  /home/user/workdir/sources/meta-clang/recipes-devtools/clang/libcxx_git.bb

A list of tasks depending on these providers is shown and may help explain where the dependency comes from.
/home/user/workdir/sources/poky/meta/recipes-support/libunwind/libunwind_1.2.1.bb has unique dependees:
  /home/user/workdir/sources/meta-openembedded/meta-oe/recipes-support/glog/glog_0.3.4.bb:do_package
  /home/user/workdir/sources/meta-openembedded/meta-oe/recipes-support/glog/glog_0.3.4.bb:do_prepare_recipe_sysroot
/home/user/workdir/sources/meta-clang/recipes-devtools/clang/libcxx_git.bb has unique dependees:
 /home/user/workdir/sources/poky/meta/recipes-core/packagegroups/packagegroup-core-standalone-sdk-target.bb:do_package_write_ipk
It could be that one recipe provides something the other doesn't and should. The following provider and runtime provider differences may be helpful.
```

Any help at all is highly apreciated :) 

Thanks!

kraj, 2019-10-11:
@schultzern, I would suggest trying the master branch of meta-clang.  Poky is supported. poky = cor + bitbake + Ref-BSPs, I know in older releases we exported libunwind from libcxx and it would then conflict with libunwind recipe. Another solution you can do is remove `unwind` from PACKAGECONFIGs in libcxx recipe. 

Always paste your build configuration so we can get information about which layers are in play and which branches ar in use.

schultzern, 2019-10-16:
Thanks for quick feedback. I was able to solve the problem removing `unwind` from the PACKAGECONFIG in `libcxx` as you suggested. 

Noted. Will post my build configuration next time :)

closed 2019-10-16 by schultzern