kraj/meta-clang #206: fail using in poky warrior

yihect, 2019-10-17:
I'm using meta-clang in poky warrior, compiling is fine. The libLLVM-9.so is also generated in clang-libllvm rpm.
But there was a problem when using the LLVM library in another recipe.

That recipe uses some functions like LLVMInitializeBPFTarget(). When compiling, err msg reports they are undeclared, so I added:

Extern "C" void LLVMInitializeBPFTarget();
Extern "C" void LLVMInitializeBPFTargetMC();
Extern "C" void LLVMInitializeBPFTargetInfo();
Extern "C" void LLVMInitializeBPFAsmPrinter();
Extern "C" void LLVMInitializeBPFAsmParser();
Extern "C" void LLVMInitializeBPFDisassembler();

However, when linking, LLVMInitializeBPFTarget() is still reported as undefined reference. my llvm config is:

Root@cat: ~/bc/build# llvm-config --libs
-lLLVM-9

Using nm to check libLLVM-9.so, no *BPFTarget* symbol in it:
Nm ./tmp/usr/lib/libLLVM-9.so | grep BPF
0000000000671810 t _ZL12parseBPFArchN4llvm9StringRefE
0000000001a76aa0 T _ZN4llvm14RuntimeDyldELF20resolveBPFRelocationERKNS_12SectionEntryEmmjl
00000000006a8680 T _ZN4llvm3sys6detail20getHostCPUNameForBPFEv
00000000017ff540 t _ZN4llvm6objectL10resolveBPFENS0_13RelocationRefEmm
00000000017ff520 t _ZN4llvm6objectL11supportsBPFEm

I suspect that libLLVM-9.so does not contain related components, but 'llvm-config --components' shows that all-targets are already included.
aggressiveinstcombine all all-targets analysis asmparser asmprinter binaryformat bitreader bitstreamreader bitwriter codegen core coroutines coverage debuginfocodeview debuginfodwarf debuginfogsym debuginfomsf debuginfopdb demangle dlltooldriver engine executionengine fuzzmutate globalisel instcombine instrumentation interpreter ipo irreader jitlink libdriver lineeditor linker lto mc mca mcdisassembler mcjit mcparser mirparser native nativecodegen objcarcopts object objectyaml option orcjit Passes profiledata remarks runtimedyld scalaropts selectiondag support symbolize tablegen target textapi transformutils vectorize windowsmanifest x86 x86asmparser x86codegen x86desc x86disassembler x86info x86utils xray

I think it may still be the reason from the llvm library, but I don't know where it is causing the problem.

kraj, 2019-11-11:
it seems `BPF` is not enabled in `LLVM_TARGETS_TO_BUILD` for tests add it in`local.conf`
```
LLVM_TARGETS_TO_BUILD_append=";BPF"
```
does that help ?

kraj, 2019-11-14:
this should be fixed on master with https://github.com/kraj/meta-clang/pull/215

kraj, 2019-11-20:
@yihect please confirm if latest master is working for you.

yihect, 2019-11-22:
Yeah, it works. It has been a long time. Actually, I used another method to get around this problem, but this can work, thank you.

closed 2019-11-22 by yihect