kraj/meta-clang #207: Unable to build image SDK, missing llvm-ar?

wertig0n, 2019-10-25:
Hi!

I try to build the yocto SDK for my own custom image but get these errors, building the image regularly works fine:

`bitbake test-dev-image`

But when I try to extract an SDK image, not so much:

`bitbake test-dev-image -c populate_sdk`

The errors I get are (condensed, if you want me to provide full logs let me know):

```
ERROR: nativesdk-compiler-rt-7.1.0-r0 do_compile: Function failed: do_compile...

Log data follows:
| /bin/sh: build/tmp-glibc/work/x86_64-nativesdk-test-linux/nativesdk-compiler-rt/7.1.0-r0/recipe-sysroot-native/usr/bin/x86_64-test-linux/x86_64-test-linux-llvm-ar: No such file or directory
| /bin/sh: build/tmp-glibc/work/x86_64-nativesdk-test-linux/nativesdk-compiler-rt/7.1.0-r0/recipe-sysroot-native/usr/bin/x86_64-test-linux/x86_64-test-linux-llvm-ar: No such file or directory
| /bin/sh: build/tmp-glibc/work/x86_64-nativesdk-test-linux/nativesdk-compiler-rt/7.1.0-r0/recipe-sysroot-native/usr/bin/x86_64-test-linux/x86_64-test-linux-llvm-ar: No such file or directory
| /bin/sh: build/tmp-glibc/work/x86_64-nativesdk-test-linux/nativesdk-compiler-rt/7.1.0-r0/recipe-sysroot-native/usr/bin/x86_64-test-linux/x86_64-test-linux-llvm-ar: No such file or directory
| /bin/sh: build/tmp-glibc/work/x86_64-nativesdk-test-linux/nativesdk-compiler-rt/7.1.0-r0/recipe-sysroot-native/usr/bin/x86_64-test-linux/x86_64-test-linux-llvm-ar: No such file or directory
| /bin/sh: build/tmp-glibc/work/x86_64-nativesdk-test-linux/nativesdk-compiler-rt/7.1.0-r0/recipe-sysroot-native/usr/bin/x86_64-test-linux/x86_64-test-linux-llvm-ar: No such file or directory

ERROR: nativesdk-libcxx-7.1.0-r0 do_compile: Function failed: do_compile...

Log data follows:
| /bin/sh: build/tmp-glibc/work/x86_64-nativesdk-test-linux/nativesdk-libcxx/7.1.0-r0/recipe-sysroot-native/usr/bin/x86_64-test-linux/x86_64-test-linux-llvm-ar: No such file or directory
```

And if I list the files in both these directories I get the same output:

```
-rwxr-xr-x 4 abb abb 1235920 okt 22 13:53 x86_64-spark-linux-addr2line
-rwxr-xr-x 4 abb abb 1260312 okt 22 13:53 x86_64-spark-linux-ar
-rwxr-xr-x 4 abb abb 2224544 okt 22 13:53 x86_64-spark-linux-as
-rwxr-xr-x 8 abb abb 1076008 okt 22 14:12 x86_64-spark-linux-c++
-rwxr-xr-x 4 abb abb 1231440 okt 22 13:53 x86_64-spark-linux-c++filt
-rwxr-xr-x 4 abb abb 1076008 okt 22 14:12 x86_64-spark-linux-cpp
-rwxr-xr-x 4 abb abb   31016 okt 22 13:53 x86_64-spark-linux-elfedit
-rwxr-xr-x 8 abb abb 1076008 okt 22 14:12 x86_64-spark-linux-g++
-rwxr-xr-x 8 abb abb 1071912 okt 22 14:12 x86_64-spark-linux-gcc
-rwxr-xr-x 8 abb abb 1071912 okt 22 14:12 x86_64-spark-linux-gcc-8.2.0
-rwxr-xr-x 4 abb abb   26976 okt 22 14:12 x86_64-spark-linux-gcc-ar
-rwxr-xr-x 4 abb abb   26976 okt 22 14:12 x86_64-spark-linux-gcc-nm
-rwxr-xr-x 4 abb abb   26976 okt 22 14:12 x86_64-spark-linux-gcc-ranlib
-rwxr-xr-x 4 abb abb  659344 okt 22 14:12 x86_64-spark-linux-gcov
-rwxr-xr-x 4 abb abb  503560 okt 22 14:12 x86_64-spark-linux-gcov-dump
-rwxr-xr-x 4 abb abb  528200 okt 22 14:12 x86_64-spark-linux-gcov-tool
-rwxr-xr-x 4 abb abb 1302736 okt 22 13:53 x86_64-spark-linux-gprof
-rwxr-xr-x 8 abb abb 2987904 okt 22 13:53 x86_64-spark-linux-ld
-rwxr-xr-x 8 abb abb 2987904 okt 22 13:53 x86_64-spark-linux-ld.bfd
-rwxr-xr-x 4 abb abb 1248880 okt 22 13:53 x86_64-spark-linux-nm
-rwxr-xr-x 4 abb abb 1374896 okt 22 13:53 x86_64-spark-linux-objcopy
-rwxr-xr-x 4 abb abb 2625616 okt 22 13:53 x86_64-spark-linux-objdump
-rwxr-xr-x 4 abb abb 1260344 okt 22 13:53 x86_64-spark-linux-ranlib
-rwxr-xr-x 4 abb abb  579888 okt 22 13:53 x86_64-spark-linux-readelf
-rwxr-xr-x 4 abb abb 1235792 okt 22 13:53 x86_64-spark-linux-size
-rwxr-xr-x 4 abb abb 1235920 okt 22 13:53 x86_64-spark-linux-strings
-rwxr-xr-x 4 abb abb 1374896 okt 22 13:53 x86_64-spark-linux-strip
```

Disclaimer: This may or may not be an issue with meta-clang. I base my work on the [ti arago SDK](http://software-dl.ti.com/processor-sdk-linux/esd/docs/latest/linux/Overview_Building_the_SDK.html) which is based on thud and include meta-clang, and I have a [TI sitara based system](http://www.ti.com/processors/sitara-arm/overview.html) as a dev board. I heavily suspect the fix is very easy, probably a one-line in `conf/local.conf` or something, but will take a great deal of time to find on my own and with my current expertise in Yocto. Any help available would be greatly appreciated.

Thank you for reading!

kraj, 2019-10-25:
I hope you are using ‘thud’ branch of meta-clang as well

reid-p, 2019-10-25:
I've seen the same problem when cross compiling.
I also run into out of memory problems during the build (too many parallel things going)
while compile clang and sometimes still see the python mismatch issue.
They don't happen all the time...

kraj, 2019-10-25:
Does this 

https://github.com/kraj/meta-clang/commit/e46fd11991a04730f478c00bbf2fa5ee0dd5446e

Help

wertig0n, 2019-10-25:
Unfortunately not :(

`ERROR: ParseError at src/meta-clang/recipes-core/musl/musl_%bbappend:2: Could not inherit file classes/lto.bbclass`

Still same error with the other additions and that line removed

wertig0n, 2019-10-25:
Latest commit I can see in my branch is 1d5b1c100f... if that helps any.

kraj, 2019-10-25:
Remove the changes to musl bbappend they won’t work in thud but rest should

closed 2019-10-28 by wertig0n

kraj, 2019-10-28:
Glad you found your way out