kraj/meta-clang #208: aarch64-bst-linux-ld: cannot find -lc++ when build compiler-rt

AKADiNG, 2019-10-29:
do_configure failed when built compiler-rt, here is the log

-- Check for working CXX compiler: /home/ning.ding/osbuilder1/bstos/build/tmp/work/aarch64-halium-bst-linux/compiler-rt/9.0.0-r0/recipe-sysroot-native/usr/bin/aarch64-bst-linux/aarch64-bst-linux-clang++
-- Check for working CXX compiler: /home/ning.ding/osbuilder1/bstos/build/tmp/work/aarch64-halium-bst-linux/compiler-rt/9.0.0-r0/recipe-sysroot-native/usr/bin/aarch64-bst-linux/aarch64-bst-linux-clang++ -- broken
CMake Error at /home/ning.ding/osbuilder1/bstos/build/tmp/work/aarch64-halium-bst-linux/compiler-rt/9.0.0-r0/recipe-sysroot-native/usr/share/cmake-3.14/Modules/CMakeTestCXXCompiler.cmake:53 (message):
  The C++ compiler

    "/home/ning.ding/osbuilder1/bstos/build/tmp/work/aarch64-halium-bst-linux/compiler-rt/9.0.0-r0/recipe-sysroot-native/usr/bin/aarch64-bst-linux/aarch64-bst-linux-clang++"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/ning.ding/osbuilder1/bstos/build/tmp/work/aarch64-halium-bst-linux/compiler-rt/9.0.0-r0/build/CMakeFiles/CMakeTmp
    
    Run Build Command(s):ninja cmTC_e8021 
    [1/2] Building CXX object CMakeFiles/cmTC_e8021.dir/testCXXCompiler.cxx.o
    [2/2] Linking CXX executable cmTC_e8021
    FAILED: cmTC_e8021 
    : && /home/ning.ding/osbuilder1/bstos/build/tmp/work/aarch64-halium-bst-linux/compiler-rt/9.0.0-r0/recipe-sysroot-native/usr/bin/aarch64-bst-linux/aarch64-bst-linux-clang++  -mlittle-endian -Wno-error=unused-command-line-argument -Qunused-arguments  --sysroot=/home/ning.ding/osbuilder1/bstos/build/tmp/work/aarch64-halium-bst-linux/compiler-rt/9.0.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/ning.ding/osbuilder1/bstos/build/tmp/work/aarch64-halium-bst-linux/compiler-rt/9.0.0-r0=/usr/src/debug/compiler-rt/9.0.0-r0                      -fdebug-prefix-map=/home/ning.ding/osbuilder1/bstos/build/tmp/work/aarch64-halium-bst-linux/compiler-rt/9.0.0-r0=/usr/src/debug/compiler-rt/9.0.0-r0                      -fdebug-prefix-map=/home/ning.ding/osbuilder1/bstos/build/tmp/work/aarch64-halium-bst-linux/compiler-rt/9.0.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/ning.ding/osbuilder1/bstos/build/tmp/work/aarch64-halium-bst-linux/compiler-rt/9.0.0-r0/recipe-sysroot-native=  -stdlib=libc++   -fvisibility-inlines-hidden    -mlittle-endian -Wno-error=unused-command-line-argument -Qunused-arguments  --sysroot=/home/ning.ding/osbuilder1/bstos/build/tmp/work/aarch64-halium-bst-linux/compiler-rt/9.0.0-r0/recipe-sysroot    -mlittle-endian -Wno-error=unused-command-line-argument -Qunused-arguments  --sysroot=/home/ning.ding/osbuilder1/bstos/build/tmp/work/aarch64-halium-bst-linux/compiler-rt/9.0.0-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=/home/ning.ding/osbuilder1/bstos/build/tmp/work/aarch64-halium-bst-linux/compiler-rt/9.0.0-r0=/usr/src/debug/compiler-rt/9.0.0-r0                      -fdebug-prefix-map=/home/ning.ding/osbuilder1/bstos/build/tmp/work/aarch64-halium-bst-linux/compiler-rt/9.0.0-r0=/usr/src/debug/compiler-rt/9.0.0-r0                      -fdebug-prefix-map=/home/ning.ding/osbuilder1/bstos/build/tmp/work/aarch64-halium-bst-linux/compiler-rt/9.0.0-r0/recipe-sysroot=                      -fdebug-prefix-map=/home/ning.ding/osbuilder1/bstos/build/tmp/work/aarch64-halium-bst-linux/compiler-rt/9.0.0-r0/recipe-sysroot-native=  -stdlib=libc++   -fvisibility-inlines-hidden    -mlittle-endian -Wno-error=unused-command-line-argument -Qunused-arguments  --sysroot=/home/ning.ding/osbuilder1/bstos/build/tmp/work/aarch64-halium-bst-linux/compiler-rt/9.0.0-r0/recipe-sysroot -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed CMakeFiles/cmTC_e8021.dir/testCXXCompiler.cxx.o  -o cmTC_e8021   && :
    /home/ning.ding/osbuilder1/bstos/build/tmp/work/aarch64-halium-bst-linux/compiler-rt/9.0.0-r0/recipe-sysroot-native/usr/bin/aarch64-bst-linux/aarch64-bst-linux-ld: cannot find -lc++
    clang-9: error: linker command failed with exit code 1 (use -v to see invocation)
    ninja: build stopped: subcommand failed.

kraj, 2019-10-30:
I see `-stdlib=libc++` in the compiler cmdline, which should not have been the case since we pass `LIBCPLUSPLUS = ""` via the compiler-rt.bb recipe. So I wonder if its being injected via some local.conf change or some such  ?

AKADiNG, 2019-10-31:
> I see `-stdlib=libc++` in the compiler cmdline, which should not have been the case since we pass `LIBCPLUSPLUS = ""` via the compiler-rt.bb recipe. So I wonder if its being injected via some local.conf change or some such ?

yes , I find  
 ' TARGET_CXXFLAGS_append_toolchain-clang ?= " -stdlib=libc++ "  '  in my auto.conf，compiler-rt have been built successfully after I delete it .

But there still have other recipes build faild with clang, one example is perf ,even after I add 

TOOLCHAIN_pn-perf = "gcc"
CXX_remove_pn-perf_toolchain-clang = " -stdlib=libc++ "

in nonclangable.conf, it still not work 。Log is bellow

unable to execute 'aarch64-bst-linux-clang': No such file or directory
error: command 'aarch64-bst-linux-clang' failed with exit status 1
Makefile.perf:737: recipe for target 'install-python_ext' failed
make[1]: *** [install-python_ext] Error 1
Makefile:108: recipe for target 'install-python_ext' failed
make: *** [install-python_ext] Error 2
make: Leaving directory '/home/ning.ding/osbuilder1/bstos/build/tmp/work/a500-bst-linux/perf/1.0-r9/perf-1.0/tools/perf'
ERROR: oe_runmake failed

What confuse me is that why it seems still use clang to build perf after I set   TOOLCHAIN_pn-perf = "gcc"  ?

kraj, 2019-11-01:
@AKADiNG perf depends a lot on kernel, so I wonder if its related to kernel more than clang, but the error you cite here seems to tell me that it did not do a clean build after changing `TOOLCHAIN` variable. So maybe just delete `TMPDIR` and the do `bitbake perf`

AKADiNG, 2019-11-04:
> @AKADiNG perf depends a lot on kernel, so I wonder if its related to kernel more than clang, but the error you cite here seems to tell me that it did not do a clean build after changing `TOOLCHAIN` variable. So maybe just delete `TMPDIR` and the do `bitbake perf`

I solve this problem by switch to branch warrior , thank you very much

closed 2019-11-04 by AKADiNG