kraj/meta-clang #209: Unable to build x86 clang 9.0

mdavis777, 2019-11-01:
I am currently trying to build clang in 32bit core2 tune and I get a linker error.
This is building against thud and warrior.

[error.txt](https://github.com/kraj/meta-clang/files/3798085/error.txt)

mdavis777, 2019-11-01:
x86-64 build of the same configuration works fine.

kraj, 2019-11-01:
@mdavis777 perhaps adding `LDFLAGS_append_x86 = " -latomic" ` might help ...

mdavis777, 2019-11-01:
@kraj 
No change in the error.  I see -latomic in the linking line.

| FAILED: bin/clangd
| : && /build/mdavis/core/tmp/work/core2-32-poky-linux/clang/9.0.0-r0/recipe-sysroot-native/usr/bin/i686-poky-linux/i686-poky-linux-clang++  -m32 -march=core2 -mtune=core2 -msse3 -mfpmath=sse   -mlittle-endian -Wno-error=unused-command-line-argument -Qunused-arguments -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/build/mdavis/core/tmp/work/core2-32-poky-linux/clang/9.0.0-r0/recipe-sysroot  -O2 -pipe   -fvisibility-inlines-hidden  -m32 -march=core2 -mtune=core2 -msse3 -mfpmath=sse   -mlittle-endian -Wno-error=unused-command-line-argument -Qunused-arguments -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/build/mdavis/core/tmp/work/core2-32-poky-linux/clang/9.0.0-r0/recipe-sysroot -fPIC -fvisibility-inlines-hidden -Werror=date-time -Werror=unguarded-availability-new -std=c++11 -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wmissing-field-initializers -pedantic -Wno-long-long -Wimplicit-fallthrough -Wcovered-switch-default -Wno-noexcept-type -Wnon-virtual-dtor -Wdelete-non-virtual-dtor -Wstring-conversion -fdiagnostics-color -ffunction-sections -fdata-sections -fno-common -Woverloaded-virtual -Wno-nested-anon-types -DNDEBUG  -m32 -march=core2 -mtune=core2 -msse3 -mfpmath=sse   -mlittle-endian -Wno-error=unused-command-line-argument -Qunused-arguments -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/build/mdavis/core/tmp/work/core2-32-poky-linux/clang/9.0.0-r0/recipe-sysroot  -O2 -pipe   -fvisibility-inlines-hidden  -m32 -march=core2 -mtune=core2 -msse3 -mfpmath=sse   -mlittle-endian -Wno-error=unused-command-line-argument -Qunused-arguments -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/build/mdavis/core/tmp/work/core2-32-poky-linux/clang/9.0.0-r0/recipe-sysroot -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong -Wl,-z,relro,-z,now -latomic  -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong -Wl,-z,relro,-z,now -latomic -Wl,-allow-shlib-undefined    -Wl,-rpath-link,/build/mdavis/core/tmp/work/core2-32-poky-linux/clang/9.0.0-r0/build/./lib  -Wl,-O3 -Wl,--gc-sections tools/clang/tools/extra/clangd/refactor/tweaks/CMakeFiles/obj.clangDaemonTweaks.dir/AnnotateHighlightings.cpp.o tools/clang/tools/extra/clangd/refactor/tweaks/CMakeFiles/obj.clangDaemonTweaks.dir/DumpAST.cpp.o tools/clang/tools/extra/clangd/refactor/tweaks/CMakeFiles/obj.clangDaemonTweaks.dir/ExpandMacro.cpp.o tools/clang/tools/extra/clangd/refactor/tweaks/CMakeFiles/obj.clangDaemonTweaks.dir/RawStringLiteral.cpp.o tools/clang/tools/extra/clangd/refactor/tweaks/CMakeFiles/obj.clangDaemonTweaks.dir/SwapIfBranches.cpp.o tools/clang/tools/extra/clangd/refactor/tweaks/CMakeFiles/obj.clangDaemonTweaks.dir/ExtractVariable.cpp.o tools/clang/tools/extra/clangd/refactor/tweaks/CMakeFiles/obj.clangDaemonTweaks.dir/ExpandAutoType.cpp.o tools/clang/tools/extra/clangd/tool/CMakeFiles/clangd.dir/ClangdMain.cpp.o  -o bin/clangd  -Wl,-rpath,"\$ORIGIN/../lib" lib/libLLVMSupport.a lib/libLLVMX86Info.a -lpthread lib/libclangAST.a lib/libclangBasic.a lib/libclangTidy.a lib/libclangDaemon.a lib/libclangFormat.a lib/libclangFrontend.a lib/libclangSema.a lib/libclangTooling.a lib/libclangToolingCore.a lib/libclangToolingSyntax.a lib/libLLVMX86Info.a lib/libclangTidyAndroidModule.a lib/libclangTidyAbseilModule.a lib/libclangTidyBoostModule.a lib/libclangTidyCERTModule.a lib/libclangTidyFuchsiaModule.a lib/libclangTidyHICPPModule.a lib/libclangTidyBugproneModule.a lib/libclangTidyCppCoreGuidelinesModule.a lib/libclangTidyGoogleModule.a lib/libclangTidyLLVMModule.a lib/libclangTidyMiscModule.a lib/libclangTidyModernizeModule.a lib/libclangTidyObjCModule.a lib/libclangTidyPerformanceModule.a lib/libclangTidyPortabilityModule.a lib/libclangTidyReadabilityModule.a lib/libclangTidyZirconModule.a lib/libclangTidyUtils.a lib/libclangTidy.a lib/libclangTooling.a lib/libclangStaticAnalyzerFrontend.a lib/libclangStaticAnalyzerCheckers.a lib/libclangStaticAnalyzerCore.a lib/libclangCrossTU.a lib/libclangToolingRefactoring.a lib/libclangIndex.a lib/libclangFormat.a lib/libclangToolingInclusions.a lib/libclangFrontend.a lib/libclangDriver.a lib/libclangParse.a lib/libLLVMMCParser.a lib/libLLVMOption.a lib/libLLVMProfileData.a lib/libclangToolingCore.a lib/libclangRewrite.a lib/libclangSerialization.a lib/libclangSema.a lib/libclangEdit.a lib/libclangAnalysis.a lib/libclangASTMatchers.a lib/libclangAST.a lib/libclangLex.a lib/libclangBasic.a lib/libLLVMMC.a lib/libLLVMDebugInfoCodeView.a lib/libLLVMDebugInfoMSF.a lib/libLLVMBitReader.a lib/libLLVMCore.a lib/libLLVMBinaryFormat.a lib/libLLVMRemarks.a lib/libLLVMBitstreamReader.a lib/libLLVMSupport.a -lz -lrt -ldl -lpthread -lm lib/libLLVMDemangle.a && :
| /build/mdavis/core/tmp/work/core2-32-poky-linux/clang/9.0.0-r0/recipe-sysroot-native/usr/bin/i686-poky-linux/i686-poky-linux-ld: lib/libclangDaemon.a(Trace.cpp.o): in function `clang::clangd::trace::(anonymous namespace)::JSONTracer::beginSpan(llvm::StringRef, llvm::json::Object*)':
| Trace.cpp:(.text._ZN5clang6clangd5trace12_GLOBAL__N_110JSONTracer9beginSpanEN4llvm9StringRefEPNS4_4json6ObjectE+0x1fc): undefined reference to `__atomic_load'
[... cut ...]
| [2969/4269] Building CXX object tools/lldb/utils/TableGen/CMakeFiles/lldb-tblgen.dir/LLDBOptionDefEmitter.cpp.o
| [2970/4269] Building CXX object tools/lld/ELF/CMakeFiles/lldELF.dir/MapFile.cpp.o
| /build/mdavis/core/tmp/work-shared/llvm-project-source-9.0.0-r0/git/lld/ELF/MapFile.cpp:117:38: warning: comparison of integers of different signs: 'unsigned int' and 'const ssize_t' (aka 'const int') [-Wsign-compare]
|           last.outputOff + last.size == p.outputOff) {
|           ~~~~~~~~~~~~~~~~~~~~~~~~~~ ^  ~~~~~~~~~~~
| 1 warning generated.

kraj, 2019-11-01:
@mdavis777 I think the problem is ordering of libraries on linker cmdline, `-latomic` appears before `-lclangDaemon.a`, if you add `-latomic` at the end of linker cmdline then it should solve the issue

mdavis777, 2019-11-01:
@kraj Ya manually adding the -latomic to the end of the line did fix the issue.  Now I just need to find the correct place to patch to have CMake put it where it should.

kraj, 2019-11-01:
@mdavis777 it would be a patch to makefiles try something like below and see if it helps.

```diff
diff --git a/clang-tools-extra/clangd/tool/CMakeLists.txt b/clang-tools-extra/clangd/tool/CMakeLists.txt
index 0846b648db8..8fe7bfeb658 100644
--- a/clang-tools-extra/clangd/tool/CMakeLists.txt
+++ b/clang-tools-extra/clangd/tool/CMakeLists.txt
@@ -15,6 +15,11 @@ if(CLANGD_BUILD_XPC)
   list(APPEND CLANGD_XPC_LIBS "clangdXpcJsonConversions" "clangdXpcTransport")
 endif()

+set(CLANGD_ATOMIC_LIB "")
+if(NOT HAVE_CXX_ATOMICS64_WITHOUT_LIB)
+  list(APPEND CLANGD_ATOMIC_LIB "atomic")
+endif()
+
 target_link_libraries(clangd
   PRIVATE
   clangAST
@@ -28,4 +33,5 @@ target_link_libraries(clangd
   clangToolingCore
   clangToolingSyntax
   ${CLANGD_XPC_LIBS}
+  ${CLANGD_ATOMIC_LIB}
   )
```

mdavis777, 2019-11-11:
Sorry took me a little bit to get this tested.  Unfortunately the patch doesn't work.  HAVE_CXX_ATOMICS64_WITHOUT_LIB comes back set as true so it never appends the library.  I also tried the logic from the newest master of clang with and without the CLANG_BUILT_STANDALONE set and it failed in the same place.

```
if(CLANG_BUILT_STANDALONE)
  # needed to get HAVE_CXX_ATOMICS64_WITHOUT_LIB defined
  include(CheckAtomic)
endif()

set(CLANGD_ATOMIC_LIB "")
if(NOT HAVE_CXX_ATOMICS64_WITHOUT_LIB)
  list(APPEND CLANGD_ATOMIC_LIB "atomic")
endif()

```
This error only happens when the clang environment is off.  Otherwise it works fine.
```
LIBCPLUSPLUS = ""
COMPILER_RT = ""
UNWINDLIB = ""

```

kraj, 2019-11-11:
OK, it sounds like a deeper issue where clang is generating calls to `__atomic_store` or `__atomic_load` which is due to std::atomic template not finding an internal implementation for these functions. so the above patch where we are assuming it will add `-latomic` is not going to help.  so my suggestion is that add it manually, or come up with a test that explicitly emits these functions so we can test that in cmake before adding it to linker cmdline automatically

kraj, 2019-11-15:
@mdavis777 I am able to build `qemux86` image with clang successfully, also full world build is successful and it has built clang ( for target ) as well. Now I wonder what your build host is and what you distro config is and what layers are in play, Are you using latest master ?

kraj, 2019-11-18:
https://github.com/kraj/meta-clang/pull/219/commits/081f35343741fe91d618a37d8073aa37d6a90052 will fix it

kraj, 2019-11-20:
final fix is https://github.com/kraj/meta-clang/commit/ba6e545b1d0ee93b4ef698fae2d5a018a062b46a please re-open if still not fixed.

closed 2019-11-20 by kraj