kraj/meta-clang #210: Linker command failed when built my project with SDK

AKADiNG, 2019-11-05:
Thank you for your attention. When I built my project with sdk, there is a ld error ad below:

/home/ning.ding/proj/a500-sdk-clang/sysroots/x86_64-bstsdk-linux/usr/bin/aarch64-bst-linux/**aarch64-bst-linux-ld**: CMakeFiles/refapp.dir/demo.cpp.o: in function `BST::WebSocketHandler::loop()':
/home/ning.ding/proj/bst-os-refapp/src/../include/WebSocketServer.hpp:169: undefined reference to `cv::imencode(cv::String const&, cv::_InputArray const&, std::vector<unsigned char, std::allocator<unsigned char> >&, std::vector<int, std::allocator<int> > const&)'
/home/ning.ding/proj/a500-sdk-clang/sysroots/x86_64-bstsdk-linux/usr/bin/aarch64-bst-linux/aarch64-bst-linux-ld: /home/ning.ding/proj/bst-os-refapp/src/../include/WebSocketServer.hpp:181: undefined reference to `cv::imencode(cv::String const&, cv::_InputArray const&, std::vector<unsigned char, std::allocator<unsigned char> >&, std::vector<int, std::allocator<int> > const&)'
clang-8: error: linker command failed with exit code 1 (use -v to see invocation)
src/CMakeFiles/refapp.dir/build.make:174: recipe for target 'src/refapp' failed
make[2]: *** [src/refapp] Error 1
CMakeFiles/Makefile2:1333: recipe for target 'src/CMakeFiles/refapp.dir/all' failed
make[1]: *** [src/CMakeFiles/refapp.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

My opencv in sdk were built by clang， I think llvm-link should be used instead of aarch64-bst-linux-ld.

I have already set 

export CC=$CLANGCC
export CXX=$CLANGCXX
export CPP=$CLANGCPP
export LD=${OECORE_NATIVE_SYSROOT}/usr/bin/llvm-link
export AR=${OECORE_NATIVE_SYSROOT}/usr/bin/llvm-ar
export NM=${OECORE_NATIVE_SYSROOT}/usr/bin/llvm-nm

in my environment file， but it dose not work.  It still use aarch64-bst-linux-ld when do link.

kraj, 2019-11-05:
Can you try adding -stdlib=libc++ to LDFLAGS

AKADiNG, 2019-11-07:
> Can you try adding -stdlib=libc++ to LDFLAGS

I add 
set(CMAKE_CXX_FLAGS "-stdlib=libc++")

in my CmakeLists.txt， and it works.

It seems like libstdc++ is the default C++ standard library in SDK environment.

Thank you very much for your advices.

kraj, 2019-11-11:
np

closed 2019-11-11 by kraj